gtkwave/gtkwave #304: Support for multidimensional arrays

cousteaulecommandant, 2024-01-29:
What is the status of multidimensional array support?
As mentioned in #211, most (if not all) Verilog simulators "flatten" packed arrays such as `logic [1:0][3:0] data;` into a `data[7:0]` signal, so there isn't really much that GtkWave can do in order to convert this back into a multidimensional array.
However, I have noticed that I can edit the generated VCD file from `$var reg 32 ! data [7:0] $end` to `$var reg 32 ! data [1:0][3:0] $end`, and surprisingly GtkWave understands this to some degree: if I add the resulting signal to the trace viewer and unfold it, I get something like
```
data[1:0][3:0]
    data[1][3]
    data[1][3]
    data[1][3]

tbybell, 2024-01-29:
I don't believe VCS flattens like this.  I know for certain its FSDB dumper doesn't, and gtkwave can read those when linked against FsdbReader.

If the simulator emits separate dimensions, gtkwave can handle it.  VCD and FST should work properly provided the simulator doesn't flatten down to 1-D.  I use multidimensional arrays on my day job all the time, so the viewer does support it.

Icarus may have issues here because when I used it last as another simulator check, it's event sensitivity was on the whole flattened vector, not the individual elements.  That said, I don't know if it could emit elements separately.  This may have changed since then.  Others will have to comment.

-Tony

cousteaulecommandant, 2024-01-29:
That's good to hear.  Unfortunately, I don't have access to VCS to test this.
By "the simulator emits separate dimensions", do you mean emitting each array element as **separate signals** (as if data[1] and data[0] were ttwo different vars), or do these simulators explicitly "group" these array elements in a way that GtkWave understands, actually displaying them as a single **"bundle"** of bit vectors, with multiple nesting levels as in my example?  Because the latter would be very nice (and worth suggesting to the developers of simulation tools that don't support this yet).

In any case, I was asking because I was curious about this behavior of GtkWave I discovered accidentally (its ability to interpret multiple indexes in VCD files), and was wondering if there was something more to this I was missing.

Bottom line, if there is already a way to "unflatten" a generated VCD, I'd like to know (and if not, it may be worth considering as a new feature).  For example:
* Allow adding a slice of a signal to the Signals list.  For example, I may only want to display the waveform for `data[7:4]` instead of the whole thing.  This is not as nice as having all array elements displayed in a hierarchy but it would at least solve the main problem of extracting a piece of a very large vector.
* Allow re-interpreting a flattened signal as an array, through an option in the GtkWave GUI.
* Do some minor manual modifications to the VCD file so that GtkWave interprets a flattened array as a multidimensional array.  (Something that doesn't involve modifying every value in the dump; only the signal definition.)

PS: As I was writing this, I just found the "Combine Down" option, which allows me to group bits manually and have the multi-level nesting I was looking for (and I can save this structure as a gtkw file for future reruns).  But a way to do this automatically rather than having to group all signals manually would be nice.

tbybell, 2024-01-29:
Simulators such as VCS will dump out the elements individually
logic [2:0][7:0] a

would be dumped as separate $var entries:
a[2][7:0]
a[1][7:0]
a[0][7:0]

Note that the VCD grammar in the Verilog spec doesn't technically allow this, but the spec isn't an absolute and vendors have added their own unofficial extensions like this for years.

Instead of grouping signals manually, it's probably easiest to see what the format is, then write a program or script to generate the appropriate save file lines--especially if you have lots of these.

My guess why various simulators would dump this as a[23:0] is only because the packed array internally is processed as 23:0 under the covers in the simulator and the packed handing becomes more of a front end compiler issue.  It's been a while since I've looked at simulator implementations.

-Tony

cousteaulecommandant, 2024-01-29:
> My guess why various simulators would dump this as a[23:0] is only because the packed array internally is processed as 23:0 under the covers in the simulator 

Actually that makes sense because (as I understand it) conceptually a "packed array" is just a regular bit vector that happens to be "pre-sliced" into segments and uses multiple indexes for its bits, but otherwise behaves like a flat 1D bit vector (you can assign it to a 1D bit vector, do arithmetic with it, etc).  Unpacked arrays, on the other hand, are closer to the concept of "multiple separate signals that you group together under a single identifier" (instead of "a single signal that is divided into multiple segments"), so I understand why it might make more sense to dump them separately, at least conceptually.  (Now I'm suddenly seeing VCS as the one doing it wrong, and iverilog getting closer to reality, except that it uses the wrong bit indexes.  But that doesn't do any good to my simulation visualization.)

I'll look into the script solution because it sounds promising for my particular use case.