gtkwave/gtkwave #305: Crash when invoking gtkwave on Ubuntu 22.04.3

jordankrim, 2024-02-01:
On Ubuntu 22.04.3 (VM) when invoking gtkwave -g my.vcd::

Near byte 143, $VAR parse error encountered with 'TOP.abc_pkg.wire'
Near byte 2233, $VAR parse error encountered
Near byte 2393, Unknown VCD identifier: ''
Near byte 2406, $VAR parse error encountered
Near byte 2443, Unknown VCD identifier: ''
Near byte 2456, $VAR parse error encountered
Near byte 2491, Unknown VCD identifier: 'HA256_MOCK_XYZ_KEY_WORDS[2]'
Near byte 2506, $VAR parse error encountered
Near byte 2512, Unknown VCD identifier: '.&'
Near byte 2543, Unknown VCD identifier: ''
Near byte 2556, $VAR parse error encountered
Near byte 2562, Unknown VCD identifier: '.&'
Near byte 2593, Unknown VCD identifier: ''
Near byte 2606, $VAR parse error encountered
Near byte 2643, Unknown VCD identifier: ''
Near byte 2656, $VAR parse error encountered
Near byte 2693, Unknown VCD identifier: ''
Near byte 2706, $VAR parse error encountered
Near byte 2743, Unknown VCD identifier: ''
Near byte 2817, $VAR parse error encountered
Near byte 2864, $VAR parse error encountered
Near byte 2911, $VAR parse error encountered
Near byte 2958, $VAR parse error encountered
Near byte 3005, $VAR parse error encountered
Near byte 3052, $VAR parse error encountered
Near byte 3058, Unknown VCD identifier: '/&'
Near byte 3099, $VAR parse error encountered
Near byte 3166, Unknown VCD identifier: '/&'
Near byte 3194, Unknown VCD identifier: ''
Near byte 3513, $VAR parse error encountered
Near byte 3560, $VAR parse error encountered
Near byte 3607, $VAR parse error encountered
Near byte 3654, $VAR parse error encountered
Near byte 3761, $VAR parse error encountered with 'TOP.top_pkg.wire'
Near byte 4093, $VAR parse error encountered with 'TOP.top_pkg.tlul_pkg.1'
Near byte 4687, $VAR parse error encountered with 'TOP.top_pkg.tlul_pkg.TestHarness.wire'
Near byte 4814, $VAR parse error encountered with 'TOP.top_pkg.tlul_pkg.TestHarness.wire'
Near byte 5226, $VAR parse error encountered with 'TOP.top_pkg.tlul_pkg.TestHarness.wire'
Near byte 5791, $VAR parse error encountered
Segmentation fault (core dumped)

Any ideas?

tbybell, 2024-02-01:
It's not a normal VCD file...or if it is, the flattened signal names are killing the parser.  Try running vcd2fst to see what happens then try loading the FST file.  What did this come from?  Was it a homemade VCD writer?

jordankrim, 2024-02-01:
vcd2fst seems to have worked but when I loaded into gtkwave, I couldn't find my "DUT".  Hard to explain where this vcd comes from, but I run the same design on RHEL8 and it loads the vcd no problem and there does not seem to be any flattening as I can easily find my "DUT".  I don't think this is "homemade", as it works on RHEL8.

tbybell, 2024-02-02:
Ubuntu vs RHEL doesn't really mean anything.  Version numbers of gtkwave are more important.

Please post the VCD file if it's not something under NDA or that you cannot share.

rfuest, 2024-02-02:
Did you use the distribution package of GTKWave in both cases? If I remember correctly Ubuntu 22.04 does contain an older version of GTKWave while RHEL 8 has been updated to the latest release.

jordankrim, 2024-02-02:
RHEL8:  GTKWave Analyzer v3.3.118 (w)1999-2023 BSI
Ubuntu 22.04.3: GTKWave Analyzer v3.3.104 (w)1999-2020 BSI

So, can I update the Ubuntu version and if so, how?

rfuest, 2024-02-03:
> So, can I update the Ubuntu version and if so, how?
> 
> BTW, I downloaded v3.3118 and tried to compile it on Ubuntu, but I am getting hundreds of errors and it doesn't work.

What kind of errors? During configure, compilation or if you try to run the executable?

Another option for running an up to date version of GTKWave is the flatpak version: https://flathub.org/apps/io.github.gtkwave.GTKWave

> I also tried to attach the VCD but it is 321 MB (and the limit seems to be 25 MB), so please advise as to how to "post" it.

Is the file still too large after compressing it? If it is you will need to use an external service to upload the file and just post a link here.

Judging by your first post the error seems to be in very beginning of the file. If you can't send the entire file you could just send the first part of the file e.g, by using `head -n NUMBER_OF_LINES your_vcd_file.vcd > part.vcd` and change NUMBER_OF_LINES to something that produces a file that is small enough to send. But at least include all lines up to the `$enddefinitions $end` line, preferably also some more after that.

tbybell, 2024-02-03:
| Near byte 143, $VAR parse error encountered with 'TOP.abc_pkg.wire'

...says to me a cut and paste of some number of lines at the beginning of the file would indicate exactly what is wrong.  Byte #143 is about the first variable declaration coming out of a simulator.  There is probably something wrong in that symbol name "wire" is a reserved verilog keyword and the VCD writer (or some VCD postprocessor) is emitting garbage into the VCD file or similar issues are occurring.

Flatpak is a good idea, but I don't know what the state is with Canonical as I think I read somewhere that they recently removed flatpak in order to push their own Snap tool instead.

jordankrim, 2024-02-03:
(1) The errors I am getting are during the compile of gtkwave 3.3.118 that I downloaded as gtkwave-3.3.118.tar.gz file.  BTW, I don't know what the difference is between that one and the gtkwave-gtk3-3.3.118.tar.gz file is.

After a simple configure of the gtkwave-3.3.118 one with no extra configure parms:

./configure
make

Here is a sample of the first few errors that I get. (Like I said the log file is large--- about 96K+ lines long):

/usr/include/x86_64-linux-gnu/sys/cdefs.h:146:55: error: missing binary operator before token "("
  146 | #if __USE_FORTIFY_LEVEL == 3 && (__glibc_clang_prereq (9, 0)                  \
      |                                                       ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:634:49: error: missing binary operator before token "("
  634 | #if __GNUC_PREREQ (4,8) || __glibc_clang_prereq (3,5)
      |                                                 ^
In file included from /usr/include/glib-2.0/glib/gtypes.h:35:
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/sysroot/usr/include/time.h:57:22: error: expected ';' before 'typedef'
   57 | __BEGIN_NAMESPACE_STD
      |                      ^
      |                      ;
   58 | /* Returned by `clock'.  */
   59 | typedef __clock_t clock_t;
      | ~~~~~~~               
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/sysroot/usr/include/time.h:60:1: error: unknown type name '__END_NAMESPACE_STD'
   60 | __END_NAMESPACE_STD
      | ^~~~~~~~~~~~~~~~~~~
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/sysroot/usr/include/time.h: In function '__USING_NAMESPACE_STD':
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/sysroot/usr/include/time.h:73:1: error: expected declaration specifiers before '__BEGIN_NAMESPACE_STD'
   73 | __BEGIN_NAMESPACE_STD
      | ^~~~~~~~~~~~~~~~~~~~~

(2) I have attached a partial VCD file which contains the $enddefinitions $end and beyond.
[jordans_part.vcd.tar.gz](https://github.com/gtkwave/gtkwave/files/14154498/jordans_part.vcd.tar.gz)

tbybell, 2024-02-03:
That VCD file looks massively corrupted.  It fails on my gtkwave-1.3.119 as well.  Something is wrong with your verilator.  

504:/tmp> gtkwave jordans_part.vcd 

GTKWave Analyzer v3.3.119 (w)1999-2024 BSI

Near byte 143, $VAR parse error encountered with 'TOP.llki_pkg.wire'
$VAR encountered after $ENDDEFINITIONS near byte 2227.  VCD is malformed, exiting.

```
00000000: 24 76 65 72 73 69 6f 6e 20 47 65 6e 65 72 61 74 $version Generat
00000010: 65 64 20 62 79 20 56 65 72 69 6c 61 74 65 64 56 ed by VerilatedV
00000020: 63 64 20 24 65 6e 64 0a 24 74 69 6d 65 73 63 61 cd $end.$timesca
00000030: 6c 65 20 31 70 73 20 24 65 6e 64 0a 20 24 73 63 le 1ps $end. $sc
00000040: 6f 70 65 20 6d 6f 64 75 6c 65 20 54 4f 50 20 24 ope module TOP $
00000050: 65 6e 64 0a 20 20 24 73 63 6f 70 65 20 6d 6f 64 end.  $scope mod
00000060: 75 6c 65 20 6c 6c 6b 69 5f 70 6b 67 20 24 65 6e ule llki_pkg $en
00000070: 64 0a 20 20 20 24 76 61 72 20 77 69 72 65 20 06 d.   $var wire .
00000080: 20 20 20 24 76 61 72 20 77 69 72 65 20 06 20 20    $var wire .  
00000090: 20 24 76 61 72 20 77 69 72 65 20 06 20 20 20 24  $var wire .   $
000000a0: 76 61 72 20 77 69 72 65 20 06 20 20 20 24 76 61 var wire .   $va
000000b0: 72 20 77 69 72 65 20 06 20 20 20 24 76 61 72 20 r wire .   $var 
000000c0: 77 69 72 65 20 06 20 20 20 24 76 61 72 20 77 69 wire .   $var wi
000000d0: 72 65 20 06 20 20 20 24 76 61 72 20 77 69 72 65 re .   $var wire
000000e0: 20 06 20 20 20 24 76 61 72 20 77 69 72 65 20 06  .   $var wire .
000000f0: 20 20 20 24 76 61 72 20 77 69 72 65 20 06 20 20    $var wire .  
00000100: 20 24 76 61 72 20 77 69 72 65 20 06 20 20 20 24  $var wire .   $
00000110: 76 61 72 20 77 69 72 65 20 06 20 20 20 24 76 61 var wire .   $va
00000120: 72 20 77 69 72 65 20 06 20 20 20 24 76 61 72 20 r wire .   $var 
00000130: 77 69 72 65 20 06 20 20 20 24 76 61 72 20 77 69 wire .   $var wi
00000140: 72 65 20 06 20 20 20 24 76 61 72 20 77 69 72 65 re .   $var wire
00000150: 20 38 20 31 7c 25 20 4c 4c 4b 49 5f 4d 49 44 5f  8 1|% LLKI_MID_
```

rfuest, 2024-02-04:
As Tony mentioned your problems are clearly caused by a broken VCD file, but if you still want to compile a newer GTKWave version:

> The errors I am getting are during the compile of gtkwave 3.3.118 that I downloaded as gtkwave-3.3.118.tar.gz file. BTW, I don't know what the difference is between that one and the gtkwave-gtk3-3.3.118.tar.gz file is.

I would suggest to use the `gtk3` archive and add `--enable-gtk3` to your `./configure` call.

> Here is a sample of the first few errors that I get. (Like I said the log file is large--- about 96K+ lines long):

The compiler configuration on your system is broken. The log shows that GTKWave gets compiled with some include files from `~/CEP/.conda-env/x86_64-conda-linux-gnu/`, which is not correct. You must have something in your shell RC file or profile that changes the environment variables to break the build.

jordankrim, 2024-02-04:
Thanks for the info.  I was able to install the gtk3 version, but it didn't change the fact that it won't load the VCD.  I am working with the author of Verilator to try to uncover where the problem might be.

jordankrim, 2024-02-08:
To close this, the author says it is not a Verilator problem, but the problem is that the C compiler is not working correctly for std::to_string.   I was able to recode it and get it to work.  I do not know why a standard C library component would fail to work.

rfuest, 2024-02-08:
> I do not know why a standard C library component would fail to work.

My guess would be that the issue is related to your compiler setup, which also prevented GTKWave to compile. The compiler could, for example, use one set of include files during compilation but link to a different version of the C library.

closed 2024-02-08 by rfuest