gtkwave/gtkwave #309: FST file format, libfst and lisencing

keesj-riscure, 2024-02-21:
Hello

I see multiple project embracing the FST format and also do some efforts in documenting it
https://docs.rs/fstapi
https://blog.timhutt.co.uk/fst_spec/

As GTKWave is released as GPLV2  am wondering if libfst fals under the same regime. As this tool, and format is used by different parties using more permissive licenses it would make sense and it would be nice to clarify the license for the library and format.
 
For short .. what it the LISENCE on libfst? Is it what the header suggest MIT? 

https://github.com/gtkwave/gtkwave/blob/master/lib/libfst/fstapi.h#L22

tbybell, 2024-02-21:
License for libfst and all the format readers/writers are MIT, basically as permissible as possible as far as open source licenses go without simply releasing it as public domain.

umarcor, 2024-02-21:
Since gtkwave is a GitHub organisation now (see https://github.com/gtkwave/gtkwave/issues/3) it would be straightforward to move the waveform utils/libs to a sibling repo and have it submodules in this one.
It would make it more explicit not only the licensing but also the fact that those libs/utilities for dealing with multiple waveform formats can be useful besides gtkwave.