gtkwave/gtkwave #310: [Feature Request] Please add a command line flag to disable the splash screen.

zapta, 2024-02-22:
I am using gtkwave as part of the ``apio sim`` command and it's very useful, thanks!

I run ``apio sim`` tens and hundreds times a day and each time I need to wait a few seconds for the splash screen that obscures the signals (my simulation are small and load very fast).  Please add to gtkwave a command line flag that disables the splash screen, for example ``-H`` for Hide.

``gtkwave   -H   my_module_tb.vcd   my_module_tb.gtkw``

If it matters, I am using GTKWave Analyzer v3.4.0 (w)1999-2022 BSI on Mac OS Sonoma 14.2 which I installed using ``brew``.

rfuest, 2024-02-22:
You can disable the splash screen in your `gtkwaverc` by adding `splash_disable on`. If you need to specify this via the command line you can use `gtkwave --rcvar "splash_disable on" my_module_tb.vcd   my_module_tb.gtkw`.

zapta, 2024-02-22:
Beautiful!  Thanks Ralf.  Now it loads in no time.

zapta, 2024-02-22:
An equivalent functionality is already available. Closing.

closed 2024-02-22 by zapta