gtkwave/gtkwave #311: Please add the ability to zoom to max if .gtkw file is specified but not found (currently it zooms to min).

zapta, 2024-02-22:
TL;DR, requesting a behavior where gtkwave zooms to max if .gtkw file is specified but not found and to the zoom level from the .gtkw file if its found. Currently gtkwave zomes to min if the .gtkw file is not found.

When running an apio simulation using ``apio sim`` it generates the ``my_module_tb.vcd`` file and invoked gtkwave using the command``gtkwave   mymodule_tb.vcd  mymodule_tb.gtkw``. The user then configure the signals in the gtkwave GUI and saves the ``mymodule_tb.gtkw`` for future simulations of that module.

The problem is that on the first invocation of gtkwave, when the file ``mymodule_tb.gtkw`` still does not exist, gtkwave zooms by default min which leaves new users puzzled why they don't see signals they just added to the view.

tbybell, 2024-02-22:
See if this RC variable works for what you need:

       do_initial_zoom_fit <value>
              A nonzero value indicates that the trace should initially be crunched to fit the screen. A zero value indicates that the initial zoom should be zero (default).

zapta, 2024-02-22:
That works. Thanks!

zapta, 2026-10-26:
This one was addressed so closing.

I just opened https://github.com/gtkwave/gtkwave/issues/526 for do_initial_zoom_fit not working on gtkwave 4.0.0

closed 2026-10-26 by zapta