gtkwave/gtkwave #312: FST structures variables and enums

jjones-github, 2024-02-27:
There are a couple ventilator users, including myself and the project arch2code ([bitbucket link](https://bitbucket.org/arch2code/workspace/projects/A2C)) that use SystemVerilog structures. Currently verilator + gtkwave does not produce data that is very usable when it comes to structures. Especially structures on interfaces or any variable with an `string` based enum.

Does the `fst` format and gtkwave already support structures and decoding enum types?

This is the waveform created with verilator for a simple module that has interfaces with parameterized types that are passed in as structures. I attached a screen shot of my waveform. It is taken from this [branch](https://bitbucket.org/arch2code/arch2code/branch/feature/A2C-257-try-fst-instead-of-vcd) by running the command `make in-and-out` from repo root. I'm assuming some issues here are user error. I tried creating a `stem` file but that did not change my results.

I know that this style is properly supported by `vcs` and the `fsdb` format, but it would be awesome if it was supported in open source as well.

[image]

tbybell, 2024-02-28:
Structures and enums were modeled after what's done in the FsdbWriter.  They should be supported.  Enum support requires some special dumping considerations (enum tables) that I thought Verilator already did, because I had some back and forth with Wilson on how to get that working...

As far as visualizing structures, I've never used nWave to view them so I don't know what it specifically does.  If it helps, adding a level of hierarchy to the viewer sometimes is useful.  Menu option would be Edit->Set Trace Max Hier = 2.  I use that feature all the time if I have to drill down into individual component definitions such as d-flops.  In that way you'll see instname.signal instead of just signal.

jjones-github, 2024-02-29:
@tbybell , thank you the `Set Trace Max Hier` does help. I will work on getting an `fsdb` run with this same sim to do a comparison. 

There is a sister issue in verilator, [here](https://github.com/verilator/verilator/issues/4524). It is somewhat related but not identical to this question. Once I have a good comparison I'll ask questions on verilator too in order to get some cross project coordination.

Ideally it would be great to see a container per object. So one per interface, then inside the interface a container per structure (recursive in case of structures in structures). As well as the enumerated type conversion(s).

wsnyder, 2024-02-29:
Note Verilator does create enum dumps into FST files, the Verilator t_trace_complex has an example including an enum inside a struct.  (The struct currently gets exploded into separate members, due to the issue discussed here.)

jjones-github, 2024-02-29:
> Note Verilator does create enum dumps into FST files, the Verilator t_trace_complex has an example including an enum inside a struct. (The struct currently gets exploded into separate members, due to the issue discussed here.)

My example above does not have an enum in it so let me fix that for clarity.

jjones-github, 2024-02-29:
> My example above does not have an enum in it so let me fix that for clarity.

It turns out that this example did have an `enum` in it. I updated the original post with an updated gtkwave image and description of the enum inside a structure. I grouped the signals that are for one interface and gave them a name of that interface's instantiation. Still working on the `fsdb` vs `fst` comparison.