gtkwave/gtkwave #313: fstapi.c:1650:116: runtime error: left shift of negative value -2

wsnyder, 2024-03-03:
Running this verilator test, which uses the GCC address sanitizer on the GTK wave writer library

   test_regress/t/t_trace_cat_fst.pl --runtime-debug --vltmt -j

Gives the runtime warning

   fstapi.c:1650:116: runtime error: left shift of negative value -2
 
Perhaps this

                                       fpos += fstWriterSVarint(f, (((int64_t)((int32_t)(prev_alias = vm4ip[2]))) << 1) | 1);

Should be 

                                       fpos += fstWriterSVarint(f, (((uint64_t)((uint32_t)(prev_alias = vm4ip[2]))) << 1) | 1);

Not sure if that is correct intent, though that does fix the warning.  Also there might be other places too; I'd suggest running the GTKwave test suite with the compiler and link flags "-fsanitize=address,undefined " to see if others pop up.

Thanks

tbybell, 2024-03-03:
It's a weird trick that was used to shoehorn dynamic aliases into the file format, so the left shift is intentional...we're operating on a signed integer based on context (see the AND with 0x80000000 to detect the "escape" condition from what is normally unsigned data; no FST trace will exceed 2B signals and even remotely be interactively usable) where we stuff information into low-order signaling payload bits then we store as a variable length signed integer.  If you cast to uint, the sign extension from 32b to 64b then goes away and you should have functional problems because the sign part of the signaling information for dynamic aliases will have been destroyed.

```
int32_t  t_i32 = ((int32_t)(prev_alias = vm4ip[2])); /* vm4ip is generic unsigned data */
int64_t  t_i64 = (int64_t)t_i32; /* convert to signed */
uint64_t t_u64 = (uint64_t)t_i64; /* sign extend through 64b */

fpos += fstWriterSVarint(f, (int64_t)((t_u64 << 1) | 1)); /* all in this block was: fpos += fstWriterSVarint(f, (((int64_t)((int32_t)(prev_alias = vm4ip[2]))) << 1) | 1); */
```

This runtime warning is annoying because left shifting negative numbers is not a prohibited operation.  I suppose it's the whole arithmetic vs logical shift issue confuses people.  Regardless, it doesn't affect this code as we've sign extended way past any potential problem area.  I'll check in the fix, but it's definitely a well-intentioned but nuisance warning.

tbybell, 2024-03-03:
https://github.com/gtkwave/gtkwave/pull/314 should have your fix, thanks.

wsnyder, 2024-03-04:
Thanks, that fixes it.

Note in C/C++ it is undefined to shift a negative number - not just the sign bit but the whole result is undefined.  While x86 is always "sane", other architectures may do differently, hence the warning is technically correct.

closed 2024-03-04 by wsnyder