gtkwave/gtkwave #315: Compare VCD Files

Qin141209, 2024-03-04:
I would like to know if gtkwave can provide **waveform comparison** functions, such as in an interface to show the waveforms of two vcd files in different colors at the same time, and in the instantiation tree for waveforms inconsistent with the instantiation of the highlight.
  The current parameters do not seem to support these elements, there is no other tool can provide vcd file comparison and display the results, so far it seems to be parsing and comparison is not a difficult matter, the main question is how to compare the results of the remarkable presentation. Is it possible to implement this on the current branch?

tbybell, 2024-03-04:
There's no automatic waveform comparison function currently.  
Programs like https://github.com/veripool/vcddiff exist that perform some amount of that function offline.

The biggest pain related to this that I've encountered in the past are simulators that inconsistently split up bit vectors into individual bits for who knows what reason.  That complicates comparisons when one set of waves has a full vector for a signal, and another set of waves has individual, single bits dumped for each bit of the vector.  It's not typical, but it does sometimes happen.  I believe this was the case with simulators from Mentor and it was very annoying.  It's one reason why in some dumps you will see [ ] prefacing a signal name in the SST: the simulator "bit-blasted" the vector into individual bits, and gtkwave displays that to let the user know what is going on.

Excluding the most trivial of sim runs for real world models, your tree of signal differences is going to be practically every signal in the model with obvious exceptions such as free running clocks and constant values.  For special cases, or testing out a new module in isolation (for example, an ALU-like chunk of logic where you are looking at a results output bus versus iterated input stimulus), what you ask for works.  All the designs I work on for my day job have a couple of hundred thousand signals at the unit level, moving up to millions or tens of millions at the project level and those sim models rapidly jump off the tracks in behavior rendering a diff program practically useless.  For this reason, the first couple of changes reported might be the only useful ones.  To give you an idea of how rapidly RTL behavior can change, there have been times on the job where to test a logic fix on a failing test, I would have to mux in the logic changes at a clock cycle slightly before or at the time of fail.  If the logic existed since time zero, the model could follow different cycle-by-cycle behavior and as such would not encounter the fail condition that I wanted to ensure was fixed properly.

There are a couple of strategies.  One that I have done is dump a subset of signals to some comparison files.  For example, the input and output signals for the logic you are looking at.

You have a bunch of signals you want to view.  Bring them up in dump file A and export as VCD for those signals.  Do the same for dump file B and export those signals too.  At that point with a restricted set of signals, run both of those exported VCD files through a tool like vcddiff.  That works fine when you are checking for results versus a constant input stimulus or if you are looking for where the overall model behavior started measurably changing.

Alternatively, bring those signals up in twinwave for manual review across two parallel gtkwave sessions: you can skip the export to VCD for this.  Just use the same "save file" for each to ensure the waves that are brought up are the same.  Start at the highest time values and scroll your way backward in time to see where the two models changed in behavior, then investigate where, how, and why they did diverge.

What it boils down to is there is no one size fits all solution for this.

If anyone has any experience with how other tools handle this--and if what was implemented is useful--I'm open to hearing input.  I have the docs for nWave/Debussy/Verdi for my day job so I can always reference those.  Thanks.

Qin141209, 2024-03-05:
Thank you for your detailed response! Your insights on comparing VCD files have been very helpful. I appreciate the explanations, especially regarding the lack of automatic comparison functionality and the complexities of signal representation.

My initial idea is to parse the VCD and split it into many VCD files containing only a single signal. Complete the output of the signal to prevent b0 from being compared with b0000, and change the other formats and symbols to be consistent. Then use diff to directly compare the text, finally integrate all the comparison results to get a json result file. But I feel that this approach is still a relatively stupid method, I am also more interested in understanding how the commercial tools are handled.

If I encounter any challenges in practice or have further questions about other tools, I'll definitely reach out to you. Thanks again for your assistance!