gtkwave/gtkwave #316: When in the text field for filtering signals, I expect different select-all behaviour

ianfhunter, 2024-03-08:
I am using GTKWave v3.3.104 on Ubuntu 22.04 installed from apt.

When I'm in the text field for filtering signals and I press CTRL+A, it selects all the signals - not the filter text.
I would expect this behaviour while in the signals table, but not while in the textbox.

Current Behaviour (Highlights all signals up here. I unfortunately cannot include an example):
[image]

Expected Behaviour (Done by manually selecting all with mouse):
[image]

tbybell, 2024-03-08:
It's a widget focus issue.

```
538:/home/bybell/gtkwave/code/gtkwave3-gtk3/src> diff -cNb signalwindow.c.1 signalwindow.c
*** signalwindow.c.1	2024-03-08 10:23:03.360462668 -0500
--- signalwindow.c	2024-03-08 10:31:17.785706680 -0500
***************
*** 874,882 ****
--- 874,890 ----
  		  if(event->state & GDK_CONTROL_MASK)
  #endif
  		    {
+ 		      if(!gtk_widget_has_focus(GLOBALS->filter_entry))
+ 				{
  		      		treeview_select_all_callback();
  		      		rc = TRUE;
  				}
+ 		      else
+ 				{
+ 				gtk_editable_select_region (GTK_EDITABLE (GLOBALS->filter_entry),0, gtk_entry_get_text_length(GTK_ENTRY(GLOBALS->filter_entry)));
+ 				rc = TRUE;
+ 				}
+ 		    }
  		  break;
  
  		case GDK_KEY_A:
***************
*** 886,894 ****
--- 894,910 ----
  		  if(event->state & GDK_CONTROL_MASK)
  #endif
  		    {
+ 		      if(!gtk_widget_has_focus(GLOBALS->filter_entry))
+ 				{
  		      		treeview_unselect_all_callback();
  		      		rc = TRUE;
  				}
+ 			else
+ 				{
+ 				gtk_editable_select_region (GTK_EDITABLE (GLOBALS->filter_entry),0, 0);
+ 				rc = TRUE;
+ 				}
+ 		    }
  		default:
  		  break;
  		}

```

tbybell, 2024-03-08:
OK, I made those changes in git for LTS and svn on SF and it will be in 3.3.119.  I didn't touch the gtk4 version.