gtkwave/gtkwave #317: Reloading waveforms on GTKWave causes glitch

MahmoudKMaarouf, 2024-03-09:
I have installed GTKWave on terminal on my M silicon Mac via the discussion here: (brew command)
https://github.com/gtkwave/gtkwave/issues/250

It works great as shown here:
[image]

Then I alter RTL, rerun Icarus Verilog, rebuild, and regenerate the vcd.
iverilog -g2012 -Y.sv -y. -I. -o build/tb_data_packer.sv.out tb_data_packer.sv &&  vvp build/tb_data_packer.sv.out 

However, upon reload (File > Reload Waveform), all the signals glitch and go away (shown below)
[image]

Is there any way to see updated waveforms without reopening GTKWave and altering the signals ?

Thank you,
Mahmoud

tbybell, 2024-03-09:
That's not normal behavior.  Is there something weird going on like the waves are at psec/fsec/etc. timescale and 62 bits or so when a multiplier is involved is overflowing?  That clock period of 10sec could be insignificant or significant depending on the dumpfile's timescale.  What is on your shell/terminal output when the waves reload?

MahmoudKMaarouf, 2024-03-09:
> That's not normal behavior. Is there something weird going on like the waves are at psec/fsec/etc. timescale and 62 bits or so when a multiplier is involved is overflowing? That clock period of [...]

Thanks for the help tbybell.
I am still having the problem after adding a timescale.
` timescale 1ns/1ns
`

Here is the terminal output:
```
MacBook-Pro data_packer % GTKWAVE | Reloading waveform...
[0] start time.
[100] end time.
GTKWAVE | ...waveform reloaded
```

tbybell, 2024-03-10:
OK, you're using VCD.
Something sounds strange there, possibly directly related to Mac.  If you reload before regenerating the file, is there the same problem, or just on the new, regenerated file.  The timescale shouldn't end at *10s* if the end is marked as "To: 100s".
I wonder if it's related to screen updates.  Try resizing the window, minimize and unminimize it, cover part of the window with another window then click it to front, etc.

MahmoudKMaarouf, 2024-03-10:
You are right!

Resizing the window resolved the issue and showed the expected waveforms. I completely misunderstood the problem. The problem is not with recompiling/building. 

**The problem is that, upon clicking Reload Waveform, it will always glitch the waveforms. Moving the window fixes this issue.**

Now the question is if there is a way to fix this problem...

playduck, 2024-04-02:
This seems to be related to #278 -- same issue, re-sizing the window fixes it. But this should've been patched with [8d5874b](https://github.com/gtkwave/gtkwave/commit/8d5874b2e7b7e9c3732c37faabbb27bb985ec0e9). 

However it seems like the file was refactored in the meantime. The [snippet from the patch](https://github.com/gtkwave/gtkwave/blob/master/src/globals.c#L1753-L1755) is now moved up by around 300 lines.
Maybe #278 was re-introduced somehow? @rfuest 

As a workaround you can use an older version. I just built [7fe1d3e](https://github.com/gtkwave/gtkwave/commit/7fe1d3e35be8c0ce861935f190036293200c538e) (a commit before the change to globals.c) and it reloads perfectly fine.

rfuest, 2024-04-02:
> Edit: I just tried the current head and the reload works. The issue seems to only be when building via brew.
> Pulling from git and building via meson works and reloads the waveforms.

The brew by @randomplum seems to build GTKWave from their own GIT branch and not the upstream repo. This repo doesn't yet contain the commit https://github.com/gtkwave/gtkwave/commit/8d5874b2e7b7e9c3732c37faabbb27bb985ec0e9. @randomplum: Could you update your repo or use the upstream repo instead?

MahmoudKMaarouf, 2024-04-02:
If there is a way to use the brew command to always have the latest GTKWave version, that would be amazing. I have had a lot of trouble installing this software on my M1 Pro Mac, and only that brew command from @randomplum worked. 

Thank you all for the help.

randomplum, 2024-04-02:
> > Edit: I just tried the current head and the reload works. The issue seems to only be when building via brew.
> > Pulling from git and building via meson works and reloads the waveforms.
> 
> The brew by @randomplum seems to build GTKWave from their own GIT branch and not the upstream repo. This repo doesn't yet contain the commit [8d5874b](https://github.com/gtkwave/gtkwave/commit/8d5874b2e7b7e9c3732c37faabbb27bb985ec0e9). @randomplum: Could you update your repo or use the upstream repo instead?

I'll update it, I froze it on a commit since upstream changes were breaking building scripts back then.

playduck, 2024-04-02:
Just tried it, installing and building via `brew install --HEAD gtkwave` works and reloads waveforms as expected.

Maybe worth mentioning:  installing via `brew install --cask gtkwave` uses the sourceforge version from the [homebrew repo](https://github.com/Homebrew/homebrew-cask/blob/4127906698f973320a041e64934a8b3c2c953b4f/Casks/g/gtkwave.rb). This doesn't seem to work, as it can't find the Switch module (even if explicitly installed and re-downloaded, but maybe I'm doing something wrong on my end)

MahmoudKMaarouf, 2024-04-03:
Upon running  **brew install --HEAD gtkwave** , I get the error:
Warning: Formula gtkwave was renamed to homebrew/cask/gtkwave.
Warning: Treating gtkwave as a formula. For the cask, use homebrew/cask/gtkwave or specify the `--cask` flag.
Warning: randomplum/gtkwave/gtkwave HEAD-d793aa1_2 is already installed and up-to-date.
To reinstall HEAD_2, run:
  brew reinstall gtkwave

Upon running **brew reinstall gtkwave**, I get the error:
[image]

playduck, 2024-04-03:
@MahmoudKMaarouf That seems odd. It works fine on my M1 Max.
Maybe try:
```bash
brew uninstall gtkwave
brew cleanup -s -n # dry run, check if the output is sane
brew cleanup -s
brew update
brew install --verbose --HEAD gtkwave
```

rfuest, 2024-04-03:
I did just try to update the CI runner to `macos-14`, which also uses an M1 processor. I hadn't realized that `macos-latest` was still using x86, because the [table](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners/about-github-hosted-runners#standard-github-hosted-runners-for-public-repositories) in the GitHub docs is incorrect.

The CI build with the newer macOS version does now fail with this error:
[image]

A quick search on the internet revealed that this and the massive amount of nullability warning might be caused by using the clang compiler from brew together with the include files from the macOS SDK. But without direct access to a macOS machine this would be hard to debug.

rfuest, 2024-04-03:
> The [github actions macOs docs](https://github.com/actions/runner-images/blob/main/images/macos/macos-14-arm64-Readme.md) however explicitly state that the homebrew version is accessible via `$(brew --prefix llvm@15)/bin/clang`. Maybe specifying this compiler magically fixes things?

Thanks for the link. Unfortunately setting the compiler to the homebrew version didn't fix the CI build. But at least I managed to figure out that CI was actually using the Apple version of clang.

rfuest, 2024-04-03:
I did manage to get the CI working by setting the Xcode SDK version to `15.3.0` (#327). I'm not sure what the default version is, because the documentation didn't exactly match the version numbers in the CI log. But at least `14.3.1` didn't work.

MahmoudKMaarouf, 2024-04-03:
Beautiful - it's working now! Are there any other steps to take? 
Thank you all so much.