gtkwave/gtkwave #319: Feature request: add a way to see which block/device a signal belongs to, after it's been added to the list

DeflateAwning, 2024-03-23:
I wish I could easily see which device/block a signal is from after it's been added to the graph.

Proposed option 1: add an option to prefix each signal name with the device it's from:
[image]

Proposed option 2: show the added signals in a tree structure, like they are in the top-left SST view. 

Proposed option 3 (least good): when you hover over an added signal, it could show you what device it's from.

tbybell, 2024-03-23:
There are several existing options:

View->Show Mouseover will bring up a floating bubble that shows the full hierarchy for the signal name in the Signals pane.
Right clicking on the signal name then selecting Open Scope will open the appropriate hierarchy in the SST.
Pressing "h" in the signals pane will toggle full hierarchy and back.
Edit->Set Trace Max Hier to a number other than 1 will allow more levels of hierarchy to be visible in the signal window.

-Tony

DeflateAwning, 2024-03-23:
Very cool, thanks! Perhaps a better default would be to enable the "View > Show Mouseover" by default? And perhaps same with the "Edit > Set Trace Max Heir"

tbybell, 2024-03-23:
You can try to do something like this either directly, in the .gtkwaverc file, or through a command line alias:

334:/home/bybell/gtkwave/code/gtkwave3/examples> gtkwave des.gtkw --rcvar 'disable_mouseover no' --rcvar 'hier_max_level 2'

DeflateAwning, 2024-03-23:
Makes sense, I guess this is just a suggestion that I think this should be global default behavior. I'm totally a fan of configurability, but with lots of options comes a desire for sensible defaults.

tbybell, 2024-03-23:
It's largely a matter of personal preference.  Those sensible defaults will vary from person to person.  It's not really too much different from vi vs emacs vs whatever editors and how their init files are set up.  That's part of the reason the viewer supports a .gtkwaverc file, or specifying it at the command line, or whatever: what's worked well for me quite literally for design and debug of half a dozen commercial processor cores isn't going to be the same for everyone.  Even for my usage, I'll use the command line overrides for special cases depending on what I'm debugging as one size doesn't fit all.

Whatever I select will make someone out there unhappy, which I suppose is good because it means the user base is large enough at this point where different, personal choices become evident.  :-)

DeflateAwning, 2024-03-24:
Sure, I've just never seen a hover tip _disabled_ by default. Any reason why you picked that as the default?

tbybell, 2024-03-24:
Largely it was to keep the floating window from obscuring signal names and/or wave data.  Having that pop up when it wasn't wanted got annoying so I kept it disabled by default.  Later on, I want to say that there were difficulties with either the accurate x/y positioning of the floating window with some window managers (e.g., because the underlying gfx mechanism on the machine lacked the concept of a root window 0,0 position) or there were issues regarding flickering when the popup window moved dynamically in the Waves pane such as when the left mouse button is held and the cursor is moved around the screen.

DeflateAwning, 2024-03-25:
Okay, I mean if you're confident it's best this way, then leave it as-is. It's well-known that most people don't adjust the defaults though, so it's too bad that so many (almost including myself) are missing out on this feature.

rfuest, 2024-03-25:
Here are my two cents in regards to the default settings: The LTS readme does suggest to copy the `.gtkwaverc` to the home or VCD file directories (https://github.com/gtkwave/gtkwave/tree/lts/gtkwave3-gtk3#installation), which I assume means that they should copy `examples/gtkwaverc` to their home directory. I never liked the idea of de-facto having to kinds of default settings, one for most users and another one for users who have bothered to read the installation instructions.

Regarding the mouse over: There already have been changes made in the master branch that did improve the flickering (#263) and this should further improve in the future. My suggestion would be to at least enable the signal list mouseover by default.

tbybell, 2024-03-25:
If you solved the flickering problem in master, sure, change the default there.  I'm going to keep it as it is in LTS because...it's LTS and eventually will be on the way out.