gtkwave/gtkwave #320: Request: add a comment about the Sourceforge repo vs. this repo to the README

DeflateAwning, 2024-03-23:
What's the relationship between this version of Gtkwave and the one that resides [at Sourceforge](https://sourceforge.net/projects/gtkwave/)? Are they the same project? Are they both actively independently maintained?

Sorry to ask such an insignificant question; am just curious, because I almost added my bug reports/feature requests over there.

tbybell, 2024-03-23:
Sourceforge repo is the original one.  It exists mostly for historical reasons.  I haven't been able to update the main webpage because they changed something on the site with respect where the ht_docs html directory is stored and I have to look into that.

For the other webpages for gtkwave on SF, several linux distros have scripts that grab the tarballs from the releases webpage, so I still release there as I don't want to have to hunt everyone down.

Any updates I make on SVN on SF eventually find their way to the LTS branch here on github.

DeflateAwning, 2024-03-23:
Makes sense; where's the best place to open bug reports/feature requests? Perhaps that could be added to the README and to the About page?

tbybell, 2024-03-23:
The best place to open bug reports, feature requests, etc. is on github.  If they are opened on SF, I'll still see them and either clone them over to github or advise the user to open them on github.  Nothing will get lost.

I'll look to see if I can figure out how to update the SF main page.  There's obviously *some* way to do it, but I haven't made any time to look into it.