gtkwave/gtkwave #322: Enabling gconf with meson

eehusky, 2024-03-26:
I was trying to get the auto reload gconf trick/hack/backdoor to work but it looks like the HAVE_GCONF flag its not enabled anywhere in the meson configuration.  I got that part taken care of but my couple attempts to add gconf-2.0 to the meson configuration but I must be missing something obvious (having never used meson before) as the library links and header files arent getting added to the compiler flags and the build is failing.

I did see an older PR about removing the gconf functionality but it looks like all the gconf parts and pieces are still in the code base so I was sort of hoping it might still work if i can sort out the build side of things.

Any chance someone with a bit more meson experience could help me out?

For the purposes of just making it work...this is what i was adding to the config.

```
gconf_dep = dependency('gconf-2.0')
gconf = cc.find_library('gconf-2')
config.set('WAVE_HAVE_GCONF', 1)
```

rfuest, 2024-04-02:
I didn't want to spend any time on fixing the GConf hack and decided to implement a very basic D-Bus based solution instead. Let me know if this works for you: #324

eehusky, 2024-04-03:
Works for me!  Both figuratively and literally, just pulled and rebuilt.  Ill add this to the end of my test scripts and keep an eye out for any problems but...so far so good.  Thanks!

eehusky, 2024-04-03:
This is probably unrelated but do you have any idea what would cause the "window snap" to stop working when snapping to the left or right side of the screen?  (I dont know what this is called but on windows/linux mint w/ cinnamon you can drag the window to the left/right/top to take up half the screen hamburger or hotdog style).  

Somewhere between 3.3.104 and the latest master branch gtkwave stopped snapping to the left and right side but snapping to the top of the screen still works (at least on linux, i dont have a windows box handy to try)

rfuest, 2024-04-03:
> This is probably unrelated but do you have any idea what would cause the "window snap" to stop working when snapping to the left or right side of the screen? (I dont know what this is called but on [...]

Interesting, it does still work in GNOME Shell. Do you use Cinnamon with Wayland or X11? I'll try to reproduce the bug inside Cinnamon when I have some time.

eehusky, 2024-04-03:
This is with X11, mint 21.3.  

I think there might be a wider spread problem and its got me wondering if there might be something off version of the gtk libraries i built against.  

i also noticed that the wave selection/highlighting is a little wonky.
  
 * ctrl+a doesnt work which is a pain.
 * Also, less importantly, selecting some waves and then deselecting some with ctrl pressed doesnt update the wave highlighting correctly (once i noticed ctrl+a not working...i went exploring).

[image]

Which flavor of linux do you normally typically use?  I might try building this in VM and see if i get lucky. 

Heres a grab of the settings the build picked up
```
username@nebula:~/github/gtkwave$ meson setup --wipe build
The Meson build system
Version: 1.4.0
Source dir: /home/username/github/gtkwave
Build dir: /home/username/github/gtkwave/build
Build type: native build
Project name: gtkwave
Project version: 3.4.0
C compiler for the host machine: ccache cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
C++ compiler for the host machine: ccache c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C++ linker for the host machine: c++ ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
Run-time dependency glib-2.0 found: YES 2.72.4
Run-time dependency gobject-2.0 found: YES 2.72.4
Run-time dependency gtk+-3.0 found: YES 3.24.33
Run-time dependency gtk+-unix-print-3.0 found: YES 3.24.33
Found CMake: /usr/bin/cmake (3.25.1)
Run-time dependency gtk-mac-integration-gtk3 found: NO (tried pkgconfig and cmake)
Run-time dependency zlib found: YES 1.2.11
Run-time dependency tcl found: YES 8.6.12
Run-time dependency tk found: NO (tried pkgconfig and cmake)
Library m found: YES
Has header "Judy.h" : YES 
Library Judy found: YES
Has header "regex.h" : YES 
Library regex found: NO
Run-time dependency threads found: YES
Run-time dependency bzip2 found: NO (tried pkgconfig and cmake)
Has header "bzlib.h" : YES 
Library bz2 found: YES
Program gperf found: YES (/usr/bin/gperf)
Program flex found: YES (/usr/bin/flex)
Has header "inttypes.h" : YES 
Has header "sys/stat.h" : YES 
Has header "alloca.h" : YES 
Has header "getopt.h" : YES 
Has header "fcntl.h" : YES 
Has header "unistd.h" : YES 
Checking for function "fseeko" : YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Werror=address: YES 
Compiler for C supports arguments -Werror=array-bounds: YES 
Compiler for C supports arguments -Werror=empty-body: YES 
Compiler for C supports arguments -Werror=implicit: YES 
Compiler for C supports arguments -Werror=implicit-function-declaration: YES 
Compiler for C supports arguments -Werror=incompatible-pointer-types: YES 
Compiler for C supports arguments -Werror=init-self: YES 
Compiler for C supports arguments -Werror=int-conversion: YES 
Compiler for C supports arguments -Werror=int-to-pointer-cast: YES 
Compiler for C supports arguments -Werror=main: YES 
Compiler for C supports arguments -Werror=misleading-indentation: YES 
Compiler for C supports arguments -Werror=missing-braces: YES 
Compiler for C supports arguments -Werror=missing-include-dirs: YES 
Compiler for C supports arguments -Werror=nonnull: YES 
Compiler for C supports arguments -Werror=overflow: YES 
Compiler for C supports arguments -Werror=parenthesis: NO 
Compiler for C supports arguments -Werror=pointer-arith: YES 
Compiler for C supports arguments -Werror=pointer-to-int-cast: YES 
Compiler for C supports arguments -Werror=return-type: YES 
Compiler for C supports arguments -Werror=sequence-point: YES 
Compiler for C supports arguments -Werror=shadow: YES 
Compiler for C supports arguments -Werror=strict-prototypes: YES 
Compiler for C supports arguments -Werror=trigraphs: YES 
Compiler for C supports arguments -Werror=undef: YES 
Compiler for C supports arguments -Werror=write-strings: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Werror=format-security -Werror=format=2: YES 
Compiler for C supports arguments -Wignored-qualifiers: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Wlogical-op: YES 
Compiler for C supports arguments -Wmissing-format-attribute: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wnested-externs: YES 
Compiler for C supports arguments -Wno-cast-function-type: YES 
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -Wno-sign-compare: YES 
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Wold-style-definition: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wswitch-default: YES 
Compiler for C supports arguments -Wundef: YES 
Compiler for C supports arguments -Wuninitialized: YES 
Compiler for C supports arguments -Wunused: YES 
Configuring config.h using configuration
Run-time dependency gobject-introspection-1.0 found: YES 1.72.0
Dependency gobject-introspection-1.0 found: YES 1.72.0 (cached)
Program /usr/bin/g-ir-scanner found: YES (/usr/bin/g-ir-scanner)
Dependency gobject-introspection-1.0 found: YES 1.72.0 (cached)
Program /usr/bin/g-ir-compiler found: YES (/usr/bin/g-ir-compiler)
Build-time dependency gio-2.0 found: YES 2.72.4
Program /usr/bin/glib-compile-resources found: YES (/usr/bin/glib-compile-resources)
Program gtk4-update-icon-cache found: NO
Program gtk-update-icon-cache found: YES (/usr/bin/gtk-update-icon-cache)
Program update-desktop-database found: YES (/usr/bin/update-desktop-database)
Program update-mime-database found: YES (/usr/bin/update-mime-database)
Build targets in project: 35
```

rfuest, 2024-04-12:
> This is with X11, mint 21.3.
> 
> I think there might be a wider spread problem and its got me wondering if there might be something off version of the gtk libraries i built against.
> 
> i also noticed that the wave selection/highlighting is a little wonky.
> 
> * ctrl+a doesnt work which is a pain.

I don't think there is anything wrong with your setup and this are real issues in GTKWave. Improving the keyboard controls is on my TODO list for GTKWave 4, which will use less custom code to handle keyboard inputs and should behave closer to the GTK defaults. This should make the code easier to maintain and the controls will be more familiar for most users.

> * Also, less importantly, selecting some waves and then deselecting some with ctrl pressed doesnt update the wave highlighting correctly (once i noticed ctrl+a not working...i went exploring).

Thanks for reporting this issue, it should be fixed by #330. At the moment each part of the UI needs to be redrawn manually if the state changes, which can easily cause different parts of the UI to become desynced. The new drawing code in GTKWave 4 will make this less likely.