gtkwave/gtkwave #326: Issue while trying to install gtkwave on Sonoma 14.4.1

jcjones15, 2024-04-03:
Hi, 

 I am trying to follow the steps highlighted in other threads:
brew update
brew outdated
brew upgrade
brew cleanup
brew uninstall gtkwave
brew untap randomplum/gtkwave
brew install --HEAD randomplum/gtkwave/gtkwave

However it is still failing, I am pasting the command output for the last command here:
`brew install --HEAD randomplum/gtkwave/gtkwave
==> Fetching randomplum/gtkwave/gtkwave
==> Cloning https://github.com/randomplum/gtkwave.git
Updating /Users/mainuser/Library/Caches/Homebrew/gtkwave--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 0394b61 Add a basic D-Bus interface
==> Installing gtkwave from randomplum/gtkwave
==> meson setup build
==> meson compile -C build --verbose
Last 15 lines from /Users/mainuser/Library/Logs/Homebrew/gtkwave/02.meson:
[67/165] rm -f lib/libfst/libfst.a && ar csr lib/libfst/libfst.a lib/libfst/libfst.a.p/fastlz.c.o lib/libfst/libfst.a.p/fstapi.c.o lib/libfst/libfst.a.p/lz4.c.o && ranlib -c lib/libfst/libfst.a
[68/165] clang -Isrc/gtkwave.p -Isrc -I../src -I. -I.. -I../src/cocoa -Ilib/libfst -I../lib/libfst -Ilib/libghw -I../lib/libghw -Ilib/libjrb -I../lib/libjrb -Ilib/libgtkwave/src -I../lib/libgtkwave/src -I/opt/homebrew/Cellar/pcre2/10.43/include -I/opt/homebrew/Cellar/glib/2.80.0_2/include/glib-2.0 -I/opt/homebrew/Cellar/glib/2.80.0_2/lib/glib-2.0/include -I/opt/homebrew/opt/gettext/include -I/opt/homebrew/Cellar/xz/5.4.6/include -I/opt/homebrew/Cellar/glib/2.80.0_2/include/gio-unix-2.0 -I/opt/homebrew/Cellar/glib/2.80.0_2/include -I/opt/homebrew/Cellar/libxau/1.0.11/include -I/opt/homebrew/Cellar/libxdmcp/1.1.5/include -I/opt/homebrew/Cellar/libxcb/1.16.1/include -I/opt/homebrew/Cellar/libx11/1.8.8/include -I/opt/homebrew/Cellar/libxext/1.3.6/include -I/opt/homebrew/Cellar/at-spi2-core/2.52.0/include/atk-1.0 -I/opt/homebrew/Cellar/jpeg-turbo/3.0.2/include -I/opt/homebrew/opt/libpng/include/libpng16 -I/opt/homebrew/opt/zstd/include -I/opt/homebrew/Cellar/libtiff/4.6.0/include -I/opt/homebrew/Cellar/gdk-pixbuf/2.42.10_1/include/gdk-pixbuf-2.0 -I/opt/homebrew/Cellar/libepoxy/1.5.10/include -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/Cellar/fontconfig/2.15.0/include -I/opt/homebrew/Cellar/libxrender/0.9.11/include -I/opt/homebrew/Cellar/pixman/0.42.2/include/pixman-1 -I/opt/homebrew/Cellar/cairo/1.18.0/include -I/opt/homebrew/Cellar/cairo/1.18.0/include/cairo -I/opt/homebrew/Cellar/fribidi/1.0.13/include/fribidi -I/opt/homebrew/Cellar/graphite2/1.3.14/include -I/opt/homebrew/Cellar/harfbuzz/8.4.0/include/harfbuzz -I/opt/homebrew/Cellar/pango/1.52.2/include/pango-1.0 -I/opt/homebrew/Cellar/gtk+3/3.24.41/include/gtk-3.0 -I/opt/homebrew/Cellar/xorgproto/2024.1/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/ffi -I/opt/homebrew/Cellar/gtk-mac-integration/3.0.1_1/include/gtkmacintegration -I/opt/homebrew/Cellar/gtk+3/3.24.41/include/gtk-3.0/unix-print -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -std=c99 -O3 -D_GNU_SOURCE -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_64 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_64 -xobjective-c -DWAVE_COCOA_GTK -DMAC_INTEGRATION -Wcast-align -Werror=address -Werror=array-bounds -Werror=empty-body -Werror=implicit -Werror=implicit-function-declaration -Werror=incompatible-pointer-types -Werror=init-self -Werror=int-conversion -Werror=int-to-pointer-cast -Werror=main -Werror=misleading-indentation -Werror=missing-braces -Werror=missing-include-dirs -Werror=nonnull -Werror=overflow -Werror=pointer-arith -Werror=pointer-to-int-cast -Werror=return-type -Werror=sequence-point -Werror=shadow -Werror=strict-prototypes -Werror=trigraphs -Werror=undef -Werror=write-strings -Wformat-nonliteral -Werror=format-security -Werror=format=2 -Wignored-qualifiers -Wimplicit-function-declaration -Wmissing-format-attribute -Wmissing-include-dirs -Wnested-externs -Wno-cast-function-type -Wno-missing-field-initializers -Wno-sign-compare -Wno-unused-parameter -Wold-style-definition -Wpointer-arith -Wstrict-prototypes -Wswitch-default -Wundef -Wuninitialized -Wunused -MD -MQ src/gtkwave.p/analyzer.c.o -MF src/gtkwave.p/analyzer.c.o.d -o src/gtkwave.p/analyzer.c.o -c ../src/analyzer.c
[69/165] clang -Isrc/gtkwave.p -Isrc -I../src -I. -I.. -I../src/cocoa -Ilib/libfst -I../lib/libfst -Ilib/libghw -I../lib/libghw -Ilib/libjrb -I../lib/libjrb -Ilib/libgtkwave/src -I../lib/libgtkwave/src -I/opt/homebrew/Cellar/pcre2/10.43/include -I/opt/homebrew/Cellar/glib/2.80.0_2/include/glib-2.0 -I/opt/homebrew/Cellar/glib/2.80.0_2/lib/glib-2.0/include -I/opt/homebrew/opt/gettext/include -I/opt/homebrew/Cellar/xz/5.4.6/include -I/opt/homebrew/Cellar/glib/2.80.0_2/include/gio-unix-2.0 -I/opt/homebrew/Cellar/glib/2.80.0_2/include -I/opt/homebrew/Cellar/libxau/1.0.11/include -I/opt/homebrew/Cellar/libxdmcp/1.1.5/include -I/opt/homebrew/Cellar/libxcb/1.16.1/include -I/opt/homebrew/Cellar/libx11/1.8.8/include -I/opt/homebrew/Cellar/libxext/1.3.6/include -I/opt/homebrew/Cellar/at-spi2-core/2.52.0/include/atk-1.0 -I/opt/homebrew/Cellar/jpeg-turbo/3.0.2/include -I/opt/homebrew/opt/libpng/include/libpng16 -I/opt/homebrew/opt/zstd/include -I/opt/homebrew/Cellar/libtiff/4.6.0/include -I/opt/homebrew/Cellar/gdk-pixbuf/2.42.10_1/include/gdk-pixbuf-2.0 -I/opt/homebrew/Cellar/libepoxy/1.5.10/include -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/Cellar/fontconfig/2.15.0/include -I/opt/homebrew/Cellar/libxrender/0.9.11/include -I/opt/homebrew/Cellar/pixman/0.42.2/include/pixman-1 -I/opt/homebrew/Cellar/cairo/1.18.0/include -I/opt/homebrew/Cellar/cairo/1.18.0/include/cairo -I/opt/homebrew/Cellar/fribidi/1.0.13/include/fribidi -I/opt/homebrew/Cellar/graphite2/1.3.14/include -I/opt/homebrew/Cellar/harfbuzz/8.4.0/include/harfbuzz -I/opt/homebrew/Cellar/pango/1.52.2/include/pango-1.0 -I/opt/homebrew/Cellar/gtk+3/3.24.41/include/gtk-3.0 -I/opt/homebrew/Cellar/xorgproto/2024.1/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/ffi -I/opt/homebrew/Cellar/gtk-mac-integration/3.0.1_1/include/gtkmacintegration -I/opt/homebrew/Cellar/gtk+3/3.24.41/include/gtk-3.0/unix-print -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -std=c99 -O3 -D_GNU_SOURCE -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_64 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_64 -xobjective-c -DWAVE_COCOA_GTK -DMAC_INTEGRATION -Wcast-align -Werror=address -Werror=array-bounds -Werror=empty-body -Werror=implicit -Werror=implicit-function-declaration -Werror=incompatible-pointer-types -Werror=init-self -Werror=int-conversion -Werror=int-to-pointer-cast -Werror=main -Werror=misleading-indentation -Werror=missing-braces -Werror=missing-include-dirs -Werror=nonnull -Werror=overflow -Werror=pointer-arith -Werror=pointer-to-int-cast -Werror=return-type -Werror=sequence-point -Werror=shadow -Werror=strict-prototypes -Werror=trigraphs -Werror=undef -Werror=write-strings -Wformat-nonliteral -Werror=format-security -Werror=format=2 -Wignored-qualifiers -Wimplicit-function-declaration -Wmissing-format-attribute -Wmissing-include-dirs -Wnested-externs -Wno-cast-function-type -Wno-missing-field-initializers -Wno-sign-compare -Wno-unused-parameter -Wold-style-definition -Wpointer-arith -Wstrict-prototypes -Wswitch-default -Wundef -Wuninitialized -Wunused -MD -MQ src/gtkwave.p/currenttime.c.o -MF src/gtkwave.p/currenttime.c.o.d -o src/gtkwave.p/currenttime.c.o -c ../src/currenttime.c
[70/165] clang  -o lib/libgtkwave/test/gw-color-theme lib/libgtkwave/test/gw-color-theme.p/gw-color-theme.c.o -Wl,-dead_strip_dylibs -Wl,-headerpad_max_install_names -Wl,-rpath,@loader_path/../src -Wl,-rpath,/opt/homebrew/Cellar/glib/2.80.0_2/lib -Wl,-rpath,/opt/homebrew/opt/gettext/lib lib/libgtkwave/src/libgtkwave.dylib /opt/homebrew/Cellar/glib/2.80.0_2/lib/libglib-2.0.dylib /opt/homebrew/opt/gettext/lib/libintl.dylib /opt/homebrew/Cellar/glib/2.80.0_2/lib/libgobject-2.0.dylib
[71/165] clang  -o lib/libgtkwave/test/gw-named-markers lib/libgtkwave/test/gw-named-markers.p/gw-named-markers.c.o -Wl,-dead_strip_dylibs -Wl,-headerpad_max_install_names -Wl,-rpath,@loader_path/../src -Wl,-rpath,/opt/homebrew/Cellar/glib/2.80.0_2/lib -Wl,-rpath,/opt/homebrew/opt/gettext/lib lib/libgtkwave/src/libgtkwave.dylib /opt/homebrew/Cellar/glib/2.80.0_2/lib/libglib-2.0.dylib /opt/homebrew/opt/gettext/lib/libintl.dylib /opt/homebrew/Cellar/glib/2.80.0_2/lib/libgobject-2.0.dylib
[... cut ...]
[meson-log.txt](https://github.com/gtkwave/gtkwave/files/14855291/meson-log.txt)
[02.meson.cc.txt](https://github.com/gtkwave/gtkwave/files/14855293/02.meson.cc.txt)
[02.meson.txt](https://github.com/gtkwave/gtkwave/files/14855295/02.meson.txt)
[01.meson.cc.txt](https://github.com/gtkwave/gtkwave/files/14855296/01.meson.cc.txt)
[01.meson.txt](https://github.com/gtkwave/gtkwave/files/14855298/01.meson.txt)
[00.options.out.txt](https://github.com/gtkwave/gtkwave/files/14855300/00.options.out.txt)

jcjones15, 2024-04-03:
@randomplum

rfuest, 2024-04-03:
The build log show that the file `GObject-2.0.gir` isn't found. This file is part of the GObject introspection system and it should be installed as a dependency by the brew formula. I'm not sure why this happens, but you could try to build again with a modified brew formula that disables introspection:
1. Download the brew formula from https://github.com/randomplum/homebrew-gtkwave
2. change head to: `head "https://github.com/gtkwave/gtkwave.git", branch: "master"`
3. change the first meson call to: `system "meson", "setup", "build", "-Dintrospection=false", *std_meson_args`
4. Install the modified version using. `brew install ./gtkwave.rb`

@randomplum: Did you encounter the same issue or do you have any idea what the cause could be?

jcjones15, 2024-04-03:
Hi @rfuest, thanks for replying. I am getting the error:
`Error: gtkwave is a head-only formula.` So it is not using the formula

randomplum, 2024-04-03:
@rfuest yes same issue, the GObject-2.0.gir is in homebrew/Cellar/glib/2.80.0_2/share/gir-1.0/GObject-2.0.gir and not sure why yet. Build started failing after I synced repo recently.  I'll investigate Tomorrow, for now I've synced the repo and added "-Dintrospection=false" to the formula repo. This way it build while I investigate

rfuest, 2024-04-04:
Maybe this is an issue with glib 2.80, which now includes some of the introspection system, that was a separate package before. There still seems to be some work done on the brew formula for glib 2.80: https://github.com/Homebrew/homebrew-core/commits/56f0ef8c63692deb94313c2fa7a6892679d6ff75/Formula/g/glib.rb

Downgrading glib to 2.78.4 to see if this fixes the build might be a good troubleshooting step.