gtkwave/gtkwave #329: Tiny/difficult-to-use UI in Hi-DPI on Linux.

NickOveracker, 2024-04-11:
This is related to but distinct from [Issue #43](https://github.com/gtkwave/gtkwave/issues/43).

The UI is small and perhaps unusable in some places on my high-DPI display in Mint.

OS: Linux Mint 21.2
Desktop environment: Cinnamon 64-bit
Hardware: Dell XPS-15

[image]

rfuest, 2024-04-11:
Which version of GTKWave are you using? It seems like you are still using a build of GTKWave that used GTK2. If this is the version that is included in your distro you could try the Flatpak version instead: https://flathub.org/apps/io.github.gtkwave.GTKWave

NickOveracker, 2024-04-11:
I was running the latest version (3.3.119), obtained from the [SourceForge page](https://gtkwave.sourceforge.net/). But the Flatpak did the trick for me!

I'll leave it up to you whether to close the issue based on that. I might just suggest adding a Flatpak link to the SourceForge page for anyone else who runs into the same problem.

rfuest, 2024-04-11:
> I'll leave it up to you whether to close the issue based on that. I might just suggest adding a Flatpak link to the SourceForge page for anyone else who runs into the same problem.

I'll leave it open for @tbybell, who maintains the GTKWave LTS version on SF. But I agree that it is hard for a new user to find the correct version of GTKWave. My short term suggestion would be to:
- Add a link to GitHub from gtkwave.sourceforge.net
  With a note that issues should be reported on GH and that all development will move GH in the future
- Remove the macOS links from the page
  The binaries don't work on recent versions of macOS anymore. As a temporary solution we could link to the homebrew version discussed in #250.
- Make the compilation instructions more visible on the SF page
  I don't think there is any documentation about the difference between `gtkwave3` and `gtkwave3-gtk3`. And the `--enable-gtk3` configure flag is only mentioned in the `README.md` file, which is hard to find on SF: https://sourceforge.net/p/gtkwave/code/HEAD/tree/gtkwave3-gtk3/ Perhaps it would be better to make GTK3 the default for `gtkwave3-gtk3`.
- Add a note to the GH readme that the master branch contains a development version of GTKWave and add a link where to find the LTS releases

In the longer term someone could pick up #144 again and we could move the GTKWave homepage and user guide to GitHub pages.

tbybell, 2024-04-11:
Various info added to the SF gtkwave page.  If there is more information needing to be changed, please let me know.

rfuest, 2024-04-11:
I think most of the relevant information is there, but I did find it hard to parse the information on the page. I've tried to break the text up into different sections to make it easier to read:
https://gist.github.com/rfuest/b0a5ed8327326c8fb60bbe8a5f14307d

[image]

Did you intentionally only link to the non `gtk3` source archive?

tbybell, 2024-04-11:
Thanks, updated with your modified index.html file.  The page not pointing to the gtk3 source was not intentional, so thanks for fixing that.  Originally, I didn't point to it as there were stability issues with the gtk3 version, but that hasn't been the case for a very long time.

closed 2024-04-12 by rfuest