gtkwave/gtkwave #331: Special handling of Icarus VCD files

rfuest, 2024-04-16:
I'm currently refactoring the VCD loader and came across this check for VCD files that were generated by Icarus: https://github.com/gtkwave/gtkwave/blob/ab54185373dd77b7f498d7bd35f5c310d2435a24/src/vcd_recoder.c#L957-L961

@tbybell: Do you remember why this was added? I found the commit in which the check was added, but there is no explanation why it was added: https://github.com/rfuest/gtkwave-archive/commit/6b608ee7ea733f926fbe5833cb73876614dacd14

tbybell, 2024-04-16:
```
From bybell@nc.rr.com Tue Jun  1 21:02:13 2010 -0400
Return-Path: <bybell@nc.rr.com>
Received: from cdptpa-web23-z01 ([10.127.132.114])
          by cdptpa-smta01.mail.rr.com with ESMTP
          id <20100602010213687.TJGJ20074@cdptpa-smta01.mail.rr.com>;
          Wed, 2 Jun 2010 01:02:13 +0000
Message-ID: <20100602010213.ULWD0.187656.root@cdptpa-web23-z01>
Date: Tue, 1 Jun 2010 21:02:13 -0400
From:  <bybell@nc.rr.com>
To: "Cary R." <xxxxxxx@yahoo.com>
Subject: Re: fst and array words
In-Reply-To: <20100602003319.9T0JQ.187192.root@cdptpa-web23-z01>
MIME-Version: 1.0
Content-Type: text/plain; charset=utf-8
Content-Transfer-Encoding: 7bit
X-Priority: 3 (Normal)
Sensitivity: Normal
X-Originating-IP: 
Status: O
X-Status: 
X-Keywords:                  
X-UID: 80

Cary,

OK, I dumped the VCD, converted to FST, and converted back.  No data loss and it's more-or-less the same as the original VCD.

I do see what you're talking about: the [] thing you see is "autocoalesce" in action in gtkwave.  This has nothing to do with the fstapi.c file, but would be in gtkwave's fst.c interface code.  I have no idea if you're dumping parts of a vector or array indices as they all look alike.

Autocoalesce handles simulators that bitblast vectors into x[7], x[6], ..., x[0].  Icarus doesn't, but other simulators sure do.

I suppose I need to look why the VCD loader doesn't combine the two nets yet FST does.  

If you try a one one .gtkwaverc file that has this:
autocoalesce off

...you'll see that there are 2 signals in the viewer rather than one.  Internally there always are two but I don't make them visible.

-Tony

---- bybell@nc.rr.com wrote: 
> Cary,
> 
> Sure, throw me the patch--I just grabbed the latest iverilog source from git.  Note that running on NC only dumps the escaped integer variable.
> 
> When you say "If the array has multiple bits I get this", I take it that you mean in gtkwave?  I'd definitely believe that as it always constructs indices for multi-bit vars.
> 
> Try fst2vcd on that FST file: for the variable name you should get exactly what you wrote into the file as there is no construction of [msb:lsb] string for the variable names--names should simply pass through.
> 
> -Tony
> 
> 
> ---- "Cary R." <xxxxxxx@yahoo.com> wrote: 
> > Hi Tony,
> > 
> > Based on a cppcheck warning I knew the sys_fst.c file was not doing
> > the right thing for escaped identifiers. I modified the code to add
> > a '\\' before the name and got this to mostly work. The one problem
> > case is an array of single bits. This is not working correctly. It
> > is giving the following when I add the [0] and [1] words.
> > 
> > reg []\arr[1:0]
> > 
> > I would expect this to be
> > 
> > reg \arr[0]
> > reg \arr[1]
> > 
> > If the array has multiple bits I get this.
> > 
> > reg \varr[0][1:0]
> > reg \varr[1][1:0]
> > 
> > A VCD dump does work as I expect. I'm attaching a Verilog file
> > that can be used to test this, but you'll need the latest code
> > from git and a patch I'm about to submit to the patch tracker.
> > Let me know what I can do to help.
> > 
> > Cary
> > 
> > 
> >       
> 

```

tbybell, 2024-04-16:
The long and short of it all is the array of single bits as dumped by Icarus.  FST sees this as well.

rfuest, 2024-04-16:
Thanks for the explanation.

How important is the autocoalesce feature in your experience? Are there many tools that output vectors split into single bits or would it be possible to disable autocoalesce by default? I would prefer to have one default behaviour instead of the special case for one type of VCD file. But I know that the VCD format specification is vague and all tools seem to interpret the specification differently, which might make using one default impossible.

tbybell, 2024-04-16:
It's been a while since I've had licenses for NC Verilog and ModelSim/Questa, but I know they will split up vectors depending on how the model builds in memory.  Back in the old days of comp.lang.verilog, I had one engineer (*) from Cadence who used to tell me that the spec is not really to be treated as absolute as implementations are the real spec.  I know that's not the way things "should" go, but if they write the tools people pay money for...  I don't think they ever addressed the bitblasting issue in the spec.

Related to that, I don't recall VCS splitting up vectors.

(*) https://www.linkedin.com/in/steven-sharp-b7a31a7/ ...I almost always deferred to Steve's judgment on anything involving seeming violations of the Verilog LRM because he was the most knowledgeable person by far on that newsgroup.

rfuest, 2024-05-01:
I did some more testing and if my understanding of the VCD specificaiton is correct the `Icarus` check isn't necessary. At the same time you added the code to check for the version string you also added some special cases for escaped identifiers. iverilog always uses a escaped identifier for single bit arrays and these are never coalesced by GTKWave.

I've created this test VCD file, which documents how I think the VCD loader should work (and does):
```
$timescale 1ns $end

$comment expected: v[0:2] $end
$var wire 1 a v [0] $end
$var wire 1 b v [1] $end
$var wire 1 c v [2] $end

$comment expected: w[2:0] $end
$var wire 1 c w [2] $end
$var wire 1 b w [1] $end
$var wire 1 a w [0] $end

$comment expected: \x[0] and \x[1] ([0] and [1] are part of the identifier) $end
$var wire 1 a \x[0] $end
$var wire 1 b \x[1] $end

$comment expected: \y[0][1:0] (The first [0] is part of the identifier and the [0] and [1] after the space are bit indices) $end
$var wire 1 a \y[0] [1] $end
$var wire 1 b \y[0] [0] $end

$comment expected: \z[0][1] and \z[0][0] (All brackets are part of the identifiers) $end
$var wire 1 a \z[0][1] $end
$var wire 1 b \z[0][0] $end

$enddefinitions $end

#0
0a
1b
zc
```

But there are some differences between the FST and VCD loaders, if I convert this file with `vcd2fst`:
- The FST loader doesn't handle the escaped identifiers the same way.
  The `\y[0]` signal is no longer coalesced and instead treated the same way as the `\z[0]...` signals. The space which separates the escaped identifier from the bit index in the VCD file is ignored.
- The FST loader sets `msi` and `lsi` to `1` for `\x[1]`.
  If the brackets are part of the identifier the `msi` and `lsi` fields should be `0` (like in the VCD loader).

tbybell, 2024-05-01:
The VCD reader in vcd2fst is simplified for the common case of what I've seen come out of VCS and some other simulators.  So, for example, random newlines as substitutes for whitespace inside a $ directive won't work as one expects.  Does the spec allow it?  Probably.  Do commercial tools generate that?  No.  I can always tell when someone decided to write his own VCD writer because the output files generated will break the VCD loaders and my response usually is "fix your writer: nobody else has this problem."  :-)  I know that sounds flippant, but I've had people who work on real commercial tools tell me the spec isn't representative of the real world and if the behavior isn't seen in VCS or NC to go get lost.

Escapes are a pain and provide an annoying set of corner cases because one could easily generate hostile "attack" sequences with random "[" and "]" brackets, etc.  I did very little testing in this area.  Generally, I'd believe such escaped identifiers come from synthesized netlists and the like and would encode signals in a somewhat behaved manner like NET$8421 or similar and not some random mishmash of bracketed indices.  Incompletely specified brackets like "\x[4" are even acceptable according to the spec.  Does it confuse the loaders?  I'm sure it does.

Note that I have no idea what tools that process one language (e.g., Chisel or whatever) into Verilog generate.  There's probably an escaped mess in there somewhere in order to encode filenames or who knows what.

Anyway, I wouldn't doubt the FST loader in gtkwave gets confused on bit indices and such for escaped identifiers.  Your test VCD file appears correct to me.  I'll attempt to look at what's going on in the C code this coming weekend then compare against VCS and nWave/Debussy/Verdi.

tbybell, 2024-05-20:
See if this works for you.  This is from the LTS version which I haven't checked in yet, but the edits for any reindented code in the development version should be quite obvious.

```
573:/home/bybell/gtkwave/code/gtkwave3-gtk3/src> diff -cNb fst.c.1 fst.c
*** fst.c.1	2024-05-19 22:08:35.859763499 -0400
--- fst.c	2024-05-19 22:23:24.371144121 -0400
***************
*** 213,230 ****
  			s = *nam;
  			pnts = h->u.var.name;
  			pntd = s;
  			while(*pnts)
  				{
  				if(*pnts != ' ')
  					{
! 					if(*pnts == '[') { lb_last = pntd; col_last = NULL; rb_last = NULL; }
! 					else if(*pnts == ':' && lb_last != NULL && col_last == NULL && rb_last == NULL) { col_last = pntd; }
! 					else if(*pnts == ']' && lb_last != NULL && rb_last == NULL) { rb_last = pntd; }
! 					else if(lb_last != NULL && rb_last == NULL && (isdigit((int)(unsigned char)*pnts) || (*pnts == '-'))) { }
  					else { lb_last = NULL; col_last = NULL; rb_last = NULL; }
  
  					*(pntd++) = *pnts;
  					}
  				pnts++;
  				}
  			*pntd = 0; *namlen = pntd - s;
--- 213,237 ----
  			s = *nam;
  			pnts = h->u.var.name;
  			pntd = s;
+ 
+ 			int esc_sm = 0;
  			while(*pnts)
  				{
+ 				if(*pnts == '\\') esc_sm = 1;
  				if(*pnts != ' ')
  					{
! 					if(*pnts == '[' && !esc_sm) { lb_last = pntd; col_last = NULL; rb_last = NULL; }
! 					else if(*pnts == ':' && lb_last != NULL && col_last == NULL && rb_last == NULL && !esc_sm) { col_last = pntd; }
! 					else if(*pnts == ']' && lb_last != NULL && rb_last == NULL && !esc_sm) { rb_last = pntd; }
! 					else if(lb_last != NULL && rb_last == NULL && (isdigit((int)(unsigned char)*pnts) || (*pnts == '-')) && !esc_sm) { }
  					else { lb_last = NULL; col_last = NULL; rb_last = NULL; }
  
  					*(pntd++) = *pnts;
  					}
+ 				else
+ 					{
+ 					esc_sm = 0;
+ 					}
  				pnts++;
  				}
  			*pntd = 0; *namlen = pntd - s;
```

Technically, there should really only be one backslash possible, but making sure the escape state machine doesn't fire twice probably is overkill here.  I don't have the spec handy to know if internally that space is supposed to be a proper part of the signal name (i.e., "\" ... "[whitespace]" is a form of bracketing), but if it is, it seems like it would confuse users a lot, not to mention really wreak havoc on things like save files, or regex searches, or whatever.

rfuest, 2024-05-28:
The `\y[0]` signal still doesn't get autocoalesced, because of this check for a `\`: https://github.com/gtkwave/gtkwave/blob/e791e5619e37791f7b6a0e45e5f7c6113abf7081/gtkwave3-gtk3/src/fst.c#L978

> Technically, there should really only be one backslash...

This gets kind of messy if we also consider that VHDL extended identifiers are working differently. In contrast to escaped Verilog identifiers they can contain spaces and even backslashes (escaped as double backslashes). `\extended \\ identifier\` is a valid VHDL identifier. 

I did a test how VHDL extended identifiers are handled in a GHDL simulation with different output formats:
- GHW: GTKWave shows it correctly as `\extended \\ identifier\`.
- FST: GTKWave strips the internal spaces `\extended\\identifier\`.
- VCD: GTKWave crashes on import, because GHDL uses this format to define the variable: `$var reg 1 ' \extended \\ identifier\ $end`, which isn't a valid VCD file in my opinion. We might need to check for `GHDL` in the version string if we want to handle this correctly.

I think that I'm going to extract the parsing of the variable names into a separate module in the development version. This way I can add a bunch of unit tests to make sure the edge cases are handled correctly.

tbybell, 2024-05-28:
This should be the fix in fst.c for the escaped "y" variable.  I missed that when doing a side by side comparison with the VCD.

`if((allowed_to_autocoalesce)&&(prevsym)&&(revcmp)) /* allow chaining for search functions.. NOTE: removed &&(!strchr(f_name[(i)&F_NAME_MODULUS], '\\')) because of esc_sm fix */`

For the stripping of spaces in FST, look at what happens for GHDL if in the vicinity of the esc_sm code adds that you fix `if(*pnts != ' ')` so it's always true (such as `if(1)`) as that's the code that strips spaces.  However, I don't know how other chunks of code such as save files or the Tcl interface or whatever in their current implementation will cope with them OK and if they'd have to be replaced with something like a proxy character such as ASCII code 160 just to get around isspace() checks in some places.  The GUI might be confusing with an "add visible spaces" at some point necessary for what is a weird corner case.  I don't know what VHDL allows.

For the most part, those escaped identifiers are only really used to interface with external tools such as synthesis.  I know the language supports them and one could create identifiers that look like brainf--k code, but things can get ridiculous after a point.  Regex expressions can be rendered useless by such symbols.

I do agree though that looking for "GHDL" (or whatever the identifier the simulator emits is) in the version string is the way to enable such behavior as what it needs is at odds with Verilog.