gtkwave/gtkwave #335: Wave View reacting sluggish under Wayland

95lux, 2024-04-26:
When i use gtkwave under Wayland, the cursor is not reacting to mouseclicks. It does not move to the new position until i move the mouse out of the wave view window to the left, in the signals window. 
Also stepping through a selected signal is a bit sluggish. 
Additionaly i can't use the Trackpad to scroll/zoom.
When i run the app with xWayland (`env -u WAYLAND_DISPLAY gtkwave`), cursor behavior and scrolling with touchpad are working again.
I could use this workaround, but i would rather use native Wayland. Anyone has experienced a similair behaviour?

rfuest, 2024-04-26:
Yes, this is a known issue. I've never been able to run GTKWave 3.3 without redraw issues under Wayland. Your options are to either use XWayland or try the GIT version of GTKWave, which should redraw the waveform display correctly and also contains some performance optimizations.

I haven't tested to use GTKWave with a trackpad for a long time and I'm not sure if the development version will work better than GTKWave 3.3. Improved trackpad support is on my TODO list for the GTK 4 port.

95lux, 2024-04-27:
Thanks so much for answering. I am running under xWayland currently. Works okay! I will try out the git version, thanks!
Looking so much forward to GTK4 Port :)