gtkwave/gtkwave #336: Import error if VCD file has no newline at the end

rfuest, 2024-04-29:
While I created some tests for the refactored VCD loader I discovered an issue that also affects GTKWave LTS:
If there is no newline after the last value change in the VCD file the loader continues to read from the buffer and sometimes emits warnings about invalid identifiers:
```
GTKWave Analyzer v3.3.118 (w)1999-2023 BSI

Near byte 32779, Unknown VCD identifier: '*'
[0] start time.
[7] end time.
WM Destroy
```
Note that the VCD file is only 264 bytes long. This is my test file https://github.com/rfuest/gtkwave/blob/loaders-refactoring/lib/libgtkwave/test/files/autocoalesce.vcd. You might need to run GTKWave multiple times to reproduce the error message.

I changed `getch` to return `-1` if `ch` is 0, which seems to have fixed the problem: https://github.com/rfuest/gtkwave/blob/112abaaeab6d21c7ca5d764398f7f61b586241c3/lib/libgtkwave/src/gw-vcd-loader.c#L684-L697

tbybell, 2024-04-29:
Interesting bug.

I think there also needs to be a memset() in there prior to the read.  You got lucky on a short, single read into a calloc'd buffer.  If this happened at file position > VCD_BSIZ bytes, you would've gotten the previous buffer's contents when the fread() failed.  What's below is what I'm going to put in LTS.

```
555:/home/bybell/gtkwave/code/gtkwave3-gtk3/src> diff -cNb vcd_recoder.c.1 vcd_recoder.c
*** vcd_recoder.c.1	2024-04-29 10:36:13.199633612 -0400
--- vcd_recoder.c	2024-04-29 10:37:13.464889791 -0400
***************
*** 848,853 ****
--- 848,854 ----
  if(feof(GLOBALS->vcd_handle_vcd_recoder_c_2)) return(-1);
  
  GLOBALS->vcdbyteno_vcd_recoder_c_3+=(GLOBALS->vend_vcd_recoder_c_3-GLOBALS->vcdbuf_vcd_recoder_c_3);
+ memset(GLOBALS->vcdbuf_vcd_recoder_c_3, 0, VCD_BSIZ);
  rd=fread(GLOBALS->vcdbuf_vcd_recoder_c_3, sizeof(char), VCD_BSIZ, GLOBALS->vcd_handle_vcd_recoder_c_2);
  GLOBALS->vend_vcd_recoder_c_3=(GLOBALS->vst_vcd_recoder_c_3=GLOBALS->vcdbuf_vcd_recoder_c_3)+rd;
  
***************
*** 864,870 ****
  static inline signed char getch(void) {
    signed char ch = (GLOBALS->vst_vcd_recoder_c_3!=GLOBALS->vend_vcd_recoder_c_3)?((int)(*GLOBALS->vst_vcd_recoder_c_3)):(getch_fetch());
    GLOBALS->vst_vcd_recoder_c_3++;
!   return(ch);
  }
  
  static inline signed char getch_peek(void) {
--- 865,871 ----
  static inline signed char getch(void) {
    signed char ch = (GLOBALS->vst_vcd_recoder_c_3!=GLOBALS->vend_vcd_recoder_c_3)?((int)(*GLOBALS->vst_vcd_recoder_c_3)):(getch_fetch());
    GLOBALS->vst_vcd_recoder_c_3++;
!   return(ch ? ch : -1);
  }
  
  static inline signed char getch_peek(void) {

```

rfuest, 2024-04-29:
> I think there also needs to be a memset() in there prior to the read.

You are right, I've added the `memset` to the refactored code: https://github.com/gtkwave/gtkwave/pull/302/commits/a775374654edd3c7b20b94f496963408e024dcaa

> EDIT: probably needs to be in getch_peek() as well for robustness.

I had already added the same code to `getch_peek`, but forgot to mention it in the description of the issue.

closed 2024-04-29 by rfuest