gtkwave/gtkwave #341: Feature request: vertical edge marker for clock signals

plomari, 2024-05-03:
I would like gtkwave to render a vertical marker across the whole signal view for each rising edge of a clock signal. Of course it may also be useful for falling edges or both edges in some cases, and there could be multiple signals for which such a marker would be interesting.

This should be a fairly mainstream feature, so maybe it already exists and I just couldn't find it...

rfuest, 2024-05-03:
This feature does already exits:
- Select the clock trace
- Menu: Search -> Pattern Search 1
- Select the edge type in the dropdown list below the signal name
- Click on the "Mark" button

This will replace the default grid with vertical lines at the selected condition. You can use "Pattern Search 2" to show a second type of events with dashed grid lines.

plomari, 2024-05-03:
Thank you. This is what I was looking for.

closed 2024-05-03 by plomari