gtkwave/gtkwave #344: Heap buffer overflow issue in function vcdsymbsearchcompare of file src/helpers/vcd2vzt.c

xiaoxiaoafeifei, 2024-05-11:
Describe the bug
Hi, I found a heap buffer overflow issue in function vcdsymbsearchcompare of file src/helpers/vcd2vzt.c

To reproduce
Steps to reproduce the behavior:

export CC="afl-clang-fast -fsanitize=address" CXX="afl-clang-fast++ -fsanitize=address"
meson build
ninja
ninja install

/usr/local/bin/vcd2vzt poc_file res.vzt
poc_file: 
[poc_file.zip](https://github.com/gtkwave/gtkwave/files/15281938/poc_file.zip)

Evidence
=================================================================
==1662832==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x63000000f770 at pc 0x56542b38ad73 bp 0x7ffc6fd12180 sp 0x7ffc6fd12178
READ of size 8 at 0x63000000f770 thread T0
    #0 0x56542b38ad72 in vcdsymbsearchcompare /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:247:10
    #1 0x56542b38ad72 in bsearch /usr/include/x86_64-linux-gnu/bits/stdlib-bsearch.h:33:22
    #2 0x56542b38ad72 in bsearch_vcd /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:268:9
    #3 0x56542b380cb2 in parse_valuechange /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:800:21
    #4 0x56542b380cb2 in vcd_parse /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:1428:25
    #5 0x56542b380cb2 in vcd_main /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:1625:5
    #6 0x56542b387759 in main /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:2011:5
    #7 0x7f9116d2bd8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #8 0x7f9116d2be3f in __libc_start_main csu/../csu/libc-start.c:392:3
    #9 0x56542b2a94f4 in _start (/usr/local/bin/vcd2vzt+0x224f4) (BuildId: 81c2d544470b4e9927d37c0d4a3a0062228876ee)

0x63000000f770 is located 0 bytes after 62320-byte region [0x630000000400,0x63000000f770)
allocated by thread T0 here:
    #0 0x56542b343508 in __interceptor_calloc (/usr/local/bin/vcd2vzt+0xbc508) (BuildId: 81c2d544470b4e9927d37c0d4a3a0062228876ee)
    #1 0x56542b38bb33 in calloc_2 /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/v2l_debug.c:125:5
    #2 0x56542b387759 in main /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:2011:5
    #3 0x7f9116d2bd8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16

SUMMARY: AddressSanitizer: heap-buffer-overflow /root/fuzz/fuzz_gtkwave/tes/gtkwave/build/../src/helpers/vcd2vzt.c:247:10 in vcdsymbsearchcompare
Shadow bytes around the buggy address:
  0x63000000f480: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x63000000f500: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x63000000f580: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x63000000f600: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x63000000f680: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x63000000f700: 00 00 00 00 00 00 00 00 00 00 00 00 00 00[fa]fa
  0x63000000f780: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x63000000f800: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x63000000f880: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x63000000f900: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x63000000f980: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1662832==ABORTING

System:
Unbound version: 1.19.3
OS: Ubuntu 22.04.3 LTS

tbybell, 2024-05-11:
This looks like your verilator is built wrong or there is some library issue because the VCD file is quite malformed.  This hit another user on this mailing list, but I remember the specifics.  I do remember Wilson debugged it for the user.

```
00000000: 24 76 65 72 73 69 6f 6e 20 47 65 6e 65 72 61 74 $version Generat
00000010: 65 64 20 62 79 20 56 65 72 69 6c 61 74 65 64 56 ed by VerilatedV
00000020: 63 64 20 24 65 6e 64 0a 24 74 69 6d 65 73 63 61 cd $end.$timesca
00000030: 6c 65 20 31 70 73 20 24 65 6e 64 0a 20 24 73 63 le 1ps $end. $sc
00000040: 6f 70 65 20 6d 6f 64 75 6c 65 20 54 4f 50 20 24 ope module TOP $
00000050: 65 6e 64 0a 20 20 24 73 63 6f 70 65 20 6d 6f 64 end.  $scope mod
00000060: 75 6c 65 20 6c 6c 6b 69 5f 70 6b 67 20 24 65 6e ule llki_pkg $en
00000070: 64 0a 20 20 20 24 76 61 72 20 77 69 72 65 20 06 d.   $var wire .
00000080: 20 20 20 24 76 61 72 20 77 69 72 65 20 06 20 20    $var wire .  
00000090: 20 24 76 61 72 20 77 69 72 65 20 06 20 20 20 24  $var wire .   $
000000a0: 76 61 72 20 77 69 72 65 20 06 20 20 20 24 76 61 var wire .   $va
000000b0: 72 20 77 69 72 65 20 06 20 20 20 24 76 61 72 20 r wire .   $var 
000000c0: 77 69 72 65 20 06 20 20 20 24 76 61 72 20 77 69 wire .   $var wi
000000d0: 72 65 20 06 20 20 20 24 76 61 72 20 77 69 72 65 re .   $var wire
000000e0: 20 06 20 20 20 24 76 61 72 20 77 69 72 65 20 06  .   $var wire .
000000f0: 20 20 20 24 76 61 72 20 77 69 72 65 20 06 20 20    $var wire .  
00000100: 20 24 76 61 72 20 77 69 72 65 20 06 20 20 20 24  $var wire .   $
00000110: 76 61 72 20 77 69 72 65 20 06 20 20 20 24 76 61 var wire .   $va
00000120: 72 20 77 69 72 65 20 06 20 20 20 24 76 61 72 20 r wire .   $var 
00000130: 77 69 72 65 20 06 20 20 20 24 76 61 72 20 77 69 wire .   $var wi
00000140: 72 65 20 06 20 20 20 24 76 61 72 20 77 69 72 65 re .   $var wire
00000150: 20 38 20 31 7c 25 20 4c 4c 4b 49 5f 4d 49 44 5f  8 1|% LLKI_MID_
```

On my run (LTS):
```
515:/tmp/q> vcd2vzt poc_file z.vzt

Converting VCD File 'poc_file' to VZT file 'z.vzt'...

Near line 13376, Unknown VCD identifier: ':'
Near line 16460, Unknown VCD identifier: '`J'
Near line 20914, Unknown VCD identifier: '^'#1239'
Triggered TALOS-2023-1805: $var after $enddefinitions security check, exiting.
Abort (core dumped)
```

Any potential fixes would likely _not_ be in vcdsymbsearchcompare() as that's fixing the symptom, not the problem.
Is the qsorted list malformed?
I don't don't see that in my run if I sanity check versus the sorted array:

```
605:/tmp/q/gtkwave3-gtk3/src/helpers> diff -cNb vcd2vzt.c.1 vcd2vzt.c
*** vcd2vzt.c.1	2024-05-11 12:20:57.219561061 -0400
--- vcd2vzt.c	2024-05-11 12:54:39.107422425 -0400
***************
*** 357,363 ****
  	{
  	vcd_distance = vcd_maxid - vcd_minid + 1;
  
! 	if(vcd_distance <= 8 * 1024 * 1024)
  		{
  		indexed = (struct vcdsymbol **)calloc_2(vcd_distance, sizeof(struct vcdsymbol *));
  
--- 357,363 ----
  	{
  	vcd_distance = vcd_maxid - vcd_minid + 1;
  
! 	if((vcd_distance <= 8 * 1024 * 1024) && 0) // force down qsort path to test that code
  		{
  		indexed = (struct vcdsymbol **)calloc_2(vcd_distance, sizeof(struct vcdsymbol *));
  
***************
*** 390,395 ****
--- 390,406 ----
  		root_v = NULL;
  		for(i=0;i<numsyms;i++)
  			{
+ 			if((i<(numsyms-1)) && (numsyms>1))
+ 				{
+ 				const void *s1 = sorted[i]->id;
+ 				const void *s2 = &sorted[i+1];
+ 
+ 				if(vcdsymbsearchcompare(s1,s2)>0)
+ 					{
+ 					chk_report_abort("Sorted VCD identifier list malformed");
+ 					}
+ 				}
+ 
  			if(sorted[i] != root_v) root_v = NULL;
  			alias_vs_normal_symadd(sorted[i], root_v);
  			}
```

I don't see how bsearch() would go into outer space unless either numsyms was updated (e.g., missing some of the TALOS fixes that would cause an update such as the 2023-1805 abort seen above) or there was memory corruption elsewhere.  Again, symptom vs problem.

Removing that 2023-1805 abort appears to duplicate your problem.

```
==27591== Invalid read of size 8
==27591==    at 0x4019A0: vcdsymbsearchcompare (vcd2vzt.c:232)
==27591==    by 0x5DF6C97: bsearch (in /usr/lib64/libc-2.17.so)
==27591==    by 0x401A4C: bsearch_vcd (vcd2vzt.c:255)
==27591==    by 0x402BE7: parse_valuechange (vcd2vzt.c:832)
==27591==    by 0x40440A: vcd_parse (vcd2vzt.c:1502)
==27591==    by 0x404A87: vcd_main (vcd2vzt.c:1719)
==27591==    by 0x4054CD: main (vcd2vzt.c:2111)
==27591==  Address 0x18 is not stack'd, malloc'd or (recently) free'd
```

Did some of the security fixes not get into the development version?

rfuest, 2024-05-11:
> Did some of the security fixes not get into the development version?
> EDIT: yep, that's what it looks like when I grep on TALOS.

They weren't merged yet, because I wanted to finish #302 first to reduce possible merge conflicts.

tbybell, 2024-05-12:
OK, good to hear that.  I'm glad this wasn't a case of, "here's yet another corner case for buffer overflows."