gtkwave/gtkwave #352: Handling of hierarchy delimiters in module and variable names

rfuest, 2024-06-02:
At the moment names that include the hierarchy delimiter are handled differently in the VCD and FST loaders.

This is my test VCD file:
```vcd
$scope module module.with.dot $end
$var wire 1 a wire.with.dot $end
$enddefinitions $end
#0
0a
```

If I load the VCD file in GTKWave 3.3.118 I get this hierarchy:
[image]

When I first convert it to FST with `vcd2fst` this hierarchy is displayed:
[image]

Also note that the VCD hierarchy no longer uses the correct `module` tree kind, because the hierarchy has been built by `build_tree_from_name` and not `allocate_and_decorate_module_tree_node`. Another difference is that the FST loader doesn't escape the name.

For the development version I would like to change the VCD loader to no longer automatically create submodules when a scope name contains the hierarchy delimiter. I haven't seen this feature used in any real world VCD files and it is also not part of the VCD spec. If you prefer to keep this feature we might at least want to change the behaviour for scope commands with correctly escaped names, like `$scope module \module.with.dot $end`, which also generate submodules.

In the long term I'm planning to change the internal storage of names to no longer rely on the hierarchy delimiter to separate the different levels of a name. One possible implementation would be to store all names internally with an escape character (like `VCDNAM_ESCAPE`) to split the names and only use the actual hierarchy delimiters in user facing strings. This will be a bigger change, but I think it would be a better way to handle any crazy names that include spaces and/or the hierarchy delimiter. At the moment there are, for example, special cases for escaped identifiers in `hier_extract` which don't work as expected if a variable is inside a scope with an escaped name.

tbybell, 2024-06-02:
Identifiers with dots in them *only* came to be in the code base because the IBM simulators I used 12-25 years ago only produced flattened signal names when you're iterating signal names out though their reader APIs.  If there was a concept of hierarchy, you had to reconstruct it in your visualization tool.  That's the only real reason the code that handles dots is in there.  If it's beneficial to remove it all and handle things in a manner somewhat similar to FST's trees, I'm in agreement with such a change.  Trees sort way faster than flattened signal names and require much less memory on large designs.  There is ifdef'd code to handle flattened name sorts and such for all formats, but that is only because I was too lazy to remove it once I proved it worked and it didn't crash.

For the FST vs VCD, one thing I just noticed tangential to this is handling of escapes at both hierarchy and signal levels causes weird behavior what with Set Max Hier getting confused as it's counting dots.  As the exit escape sequence character is missing for names,  there's no real way with plain vanilla dot characters to accurately determine hierarchy when only looking at a flattened name.  FST does look like it stores the hierarchy and signal name parts properly given how it builds trees.

There gets to be a point where the dots in names allow some really bad corner cases due to legacy reasons:
a.b.c
a.b.c.d
are *both* valid signal names with the IBM simulators because during model build, a user could create arbitrary names however he wanted despite what the front end compiler language supported.  I want to say the dots in signal names were added from the DSLFLAT client code during model build.  Users could write their own client code for model builds, so practically anything goes with those old IBM tools.

Another fun one, but with parentheses for illustration:
a.b.c ["a" hier plus "b" hier" plus "c" signal]
(a.b).c  ["a.b" hier plus "c" signal]
a.(b.c) ["a" hier plus "b.c" signal]

What is meant one one sees a.b.c?  At this point, again, who cares, but I could imagine duplicate flattened signal names from (a.b).c and a.(b.c) concurrently existing causing chaos or in the least, $DUP prefixed signals.

Explicit trees or impossible to generate escape characters to represent hierarchy are two ways I can see cleanly handling escaped names.  The Set Max Hier menu command definitely would have issues in chopping down signal names with dots in escaped names as you've discovered.

As far as escaped names go, I dislike them greatly because unless one is looking at the output of gate level sim, one probably won't ever have to deal with them.  Those escapes from synthesis generated code can get extremely ugly because a lot of those tools used characters like "|" or "/" as separators which is the main reason why the alt_hier_delimeter exists:

```
       alt_hier_delimeter <value>
              This allows another character in addition to the hier_delimeter to be used to delimit levels in the hierarchy for VCD. 
              Only the first character in the value is significant. Note that this is normally off. The intended use  is  to  resolve  the
              hierarchies of netlist based models that often contain slashes to delimit hierarchy inside of $var statements.
```

So, for example, if you instantiated some PCI IP, you'd see all the PCI IP's signal name (including its hierarchy in the IP) represented as a flattened signal name.  It makes viewing such signals very annoying due to their long name length:

a.b.c.d|e|f|g|h|i|j
where the name in the IP really is d.e.f.g.h.i.j rooted at a.b.c, but the tools replaced the IP's separator with "|" characters, probably due to processing limitations somewhere in the toolchain.

Those dots in hierarchy or signal names are probably more of a curiosity for most users and nobody would miss the removal of alt_hier_delimeter.  I say that because I can't recall ever fielding any support questions for misbehaving dots.

Whatever you do, do what seems sane for Verilog and VHDL based on each of their LRM's definition of hierarchies and signals.  All the other, old behavior is arbitrary and obsolete.  If any stray, random users need old, buggy behavior for who knows what reason, they can be pointed to the LTS version.

rfuest, 2024-06-03:
Thanks for the great explanation. I'll remove support for splitting names with dots and the `alt_hier_delimeter` option. If someone needs these features later on it should be possible to reimplement them as a post processing step after the dump file has been loaded. This could even be implemented as a python plugin, which would make it very flexible and easy to adjust for different use cases.