gtkwave/gtkwave #358: (Temporarily) removing Tcl support from GTKWave 4

rfuest, 2024-06-10:
Keeping the Tcl support working is becoming more challenging the further along the refactoring progresses. I don't think it makes a lot of sense to continue to keep this version of the Tcl API working at this point in time. Because the Tcl API relies on menu item names and trace indices, which both will change in GTKWave 4, there is unfortunately no good way of keeping compatibility with existing Tcl scripts anyway.

My plan is to remove support for Tcl entirely for the first stable release of GTKWave 4 and instead only provide support for Python scripting. If there is enough interest in adding Tcl support back to GTKWave we can reimplement an improved Tcl API in a future release of GTKWave or as a separate plugin.

suzizecat, 2024-06-19:
Hello there, 

I would like to chime in to mention a use case and ask for some clarification.
I am working (low temperature those days) on a language server for System Verilog and would have liked to interface with GTKWave in order to have an environment where I can select a signal in VSCode and right-click to add it to an open GTKWave in the same fashion you have in professional environments.

As I don't want to integrate the GTKWave codebase in my project, my idea was to launch `gtkwave -w` and interface through STDIO as this approach is fairly straightforward to implement.

Will any kind of STDIO interface to GTKWave be available after this update ?

rfuest, 2024-06-19:
Thanks for reporting back with your use case. My primary intention for this issue was to get feedback from users to get an overview of how they use the Tcl interface and what alternatives GTKWave 4 can provide to work in these use cases.

A STDIO scripting interface isn't planned at the moment. But it will be possible to add a signal from an external program via D-Bus. In my opinion this is a better interface than having to rely on a text based way of communication with GTKWave. Do you think that this will work for your application?

suzizecat, 2024-06-19:
As far as I understand D-Bus is not portable to Windows therefore it would be less than ideal for my situation (as I am often working with WSL on my office computer and an odd mix of windows/linux apps). Otherwise, for full Linux applications, that may be good enough as nodeJS provides a module to access DBus. 

As far as I know, using STDIO (to be perfectly clear, I mean by this using stdin and stdout through pipes or whatever) is a fairly common IPC method, and simply having some kind of prompt would allow users to automate some tasks (it is not my usecase though). 
Finally, IMHO, it is by far the easiest kind of interface to implement as there is almost nothing to develop for the data transport (in contrast to any kind of TCP communication, in example). However, the development of a "server" to asynchronously read those commands and send back data is a pain in C/C++. 

Perhaps the plugin capability or the python scripting may be able to provide such interfaces if the necessary internals of GTKWave are exposed ?
In this case, I could make a plugin that expose the proper commands through STDIO and do my stuff without impacting GTKWave development. 

Would you happen to know how RTLBrowse communicate with GTKWave as of today ? (as this is the kind of application I would like to reproduce)

tbybell, 2024-06-19:
> Would you happen to know how RTLBrowse communicate with GTKWave as of today ? (as this is the kind of application I would like to reproduce)

It's a region of shared memory pointing to a struct that both sides agree to be using.

suzizecat, 2024-06-20:
> It's a region of shared memory pointing to a struct that both sides agree to be using.

Which I believe is not a safe way to publicly interface with other apps, so not well suited for my usecase (beside, I don't even want to start thinking about interfacing a memory region between C/C++ and TypeScript 😬 ) 

I will try to develop a PoC to see exactly what are my requirements, but I believe that the legacy Tcl console will do just fine for the time being.

Another interesting point for my usecase is that commercial solutions (at least Cadence SimVision) provides a console that you may use in a terminal. I am pretty sure that the others CAD vendor do this too. This would allow for an easier integration in my solution as I would only have to change the commands sent to the viewer to support different tools.

rfuest, 2024-06-20:
> As far as I understand D-Bus is not portable to Windows therefore it would be less than ideal for my situation (as I am often working with WSL on my office computer and an odd mix of windows/linux [...]

I haven't tried it myself yet, but I was under the impression that D-Bus is also available on Windows: https://www.freedesktop.org/wiki/Software/dbus/#windowsport

> As far as I know, using STDIO (to be perfectly clear, I mean by this using stdin and stdout through pipes or whatever) is a fairly common IPC method, and simply having some kind of prompt would allow users to automate some tasks (it is not my usecase though).

Do you have any examples of programs that use IPC via STDIO? I cannot think of an example at the moment, but I'm sure they exist.

For automation I'm planning to add a helper application that will communicate with GTKWave via D-Bus and allow a user to easily control GTKWave in a script. For example: `gtkwavctl reload` to reload the dump file.

> Finally, IMHO, it is by far the easiest kind of interface to implement as there is almost nothing to develop for the data transport (in contrast to any kind of TCP communication, in example). [...]

It is easy, but also has drawbacks, like having to remember to escape strings correctly. That's why I like D-Bus, which is a typed interface and we don't need to manually serialize and deserialize the data to a string.

> Perhaps the plugin capability or the python scripting may be able to provide such interfaces if the necessary internals of GTKWave are exposed ?
> In this case, I could make a plugin that expose the proper commands through STDIO and do my stuff without impacting GTKWave development.

Yes, that would be an option. When the plugin support is finished, a plugin will have full control over the open project. It should then be possible to create a pipe for STDIO communication or use to any other method of communication.

> Which I believe is not a safe way to publicly interface with other apps, so not well suited for my usecase (beside, I don't even want to start thinking about interfacing a memory region between [...]

I have recently experienced the fun problems this can cause if both sides don't agree, because I've accidentally run `rtlbrowse` form GTKWave 3.3 from a GIT build of GTKWave. And doing that with TypeScript in the mix does sound scary :wink:. That's why we want to remove this IPC mechanism from GTKWave 4.

> I will try to develop a PoC to see exactly what are my requirements, but I believe that the legacy Tcl console will do just fine for the time being.

That's probably the best target at the moment, because GTKWave 4 might still be a long way in the future. But when you have determined the requirements I would like to discuss a solution how we could make these functions available for GTKWave 4.

tbybell, 2024-06-20:
> That's probably the best target at the moment, because GTKWave 4 might still be a long way in the future. But when you have determined the requirements I would like to discuss a solution how we [...]

The legacy Tcl console support was implemented by Don Baltus to interface fairly tightly with BlueSpec or BlueWave or whatever their relevant product was called, so other developers have done IPC through that console in the past.

The Tcl "repscripts" were something from a contractor named Yiftach Tzori for his own personal tool flow.  I want to say he integrated that with vi, but i don't remember.  That was quite long ago.  I'm not a fan of what are more-or-less timer interrupt driven scripts.

Some of this cruft that Ralf is stripping out during refactoring built up over time due to "hey, that's a useful feature!" feature creep.  

Anyway, the 3.3 series will not be removing features because at this point, I don't know who is using what and I don't want to introduce churn.  It's the stable, (hopefully) reliable, LTS branch of the code.

suzizecat, 2024-06-20:
> I haven't tried it myself yet, but I was under the impression that D-Bus is also available on Windows: https://www.freedesktop.org/wiki/Software/dbus/#windowsport

What I mean is that D-Bus might be available on Windows but not "out of the box" and I would rather avoid such "unusual" (AFAIK) dependency. Now, I must say that such interface is indeed interesting.

> Do you have any examples of programs that use IPC via STDIO? I cannot think of an example at the moment, but I'm sure they exist.

The example that comes to my mind first would be (most of ?) languages servers implementing the [Language Server Protocol (LSP)](https://microsoft.github.io/language-server-protocol/). While the actual IPC transport method is irrelevant, I often found either STDIO and TCP options.

See the [vscode-languageserver](https://github.com/microsoft/vscode-languageserver-node/blob/33cbe5c87032cb53cf30c2b90f29d22a45ee6800/client/src/node/main.ts#L28) library/package/whatever used to implement either LSP client or server with nodeJS which implements this:
```typescript
export enum TransportKind {
	stdio,
	ipc,
	pipe,
	socket
}
```

Or the [Pygls (for python LSP server implementation)](https://pygls.readthedocs.io/en/latest/reference/servers.html#pygls.server.Server) that implements TCP, WebSocket and STDIO.

Otherwise, the STDIO is often used to read from a process, and less usually to write to it, but it often makes an easy way to implement an exchange between two program (especially if you already have a prompt).

The main advantage, IMO, with the usage of STDIO, is that you don't have to establish a TCP connection (find a port and so on). The main issue is that you must pay extra attention to the use of `printf` or `cout`. You want all your logging message to go to either a file or STDERR. 

There might be other well known applications that uses this mechanism, but none that immediately comes to my mind.

rfuest, 2024-06-25:
> What I mean is that D-Bus might be available on Windows but not "out of the box" and I would rather avoid such "unusual" (AFAIK) dependency. Now, I must say that such interface is indeed interesting.

I think a D-Bus daemon is a hard dependency for GTK and must be installed as part of GTKWave's GTK dependency. I'm not sure how GTKWave 4 will be packaged for Windows, but if the GTK dependency is installed in GTKWave's application directory the included D-Bus might not be usable by external dependencies. This will need to be looked at as part of testing the GTK 4 port on Windows.

D-Bus will be the first RPC mechanism that is added to GTKWave, because GTK automatically exposes all actions that are defined in the application via D-Bus. This makes all menu entries and toolbar buttons executable via RPC without any additional code.

> Another software that use this mechanism would be... GTKWave itself ! For the "Translate Filter Process" mechanism (see [page 58 of the reference manual](https://gtkwave.sourceforge.net/gtkwave.pdf) 😉 )

My comment about the text based RPC was a bit congusing, I guess. I could only come up with examples like this, where STDIO is used to call another program with some data to process it, but not to keep open a long term remote control interface.

Thanks for the information about this is handled for the LSP protocol, I had always assumed that they used a socket connection. 

JSON-RPC could also work as a format if we decide to add a second RPC mechanism. JSON will be used by the new savefile format, so that this wouldn't add any additional dependencies and there are probably a lot of JSON-RPC libraries for different languages that would make it easy to remote control GTKWave. While I'm not a big fan of the JSON format it cannot be denied that it's ubiquity has advantages and it doesn't require manual parsing.

> The main advantage, IMO, with the usage of STDIO, is that you don't have to establish a TCP connection (find a port and so on). The main issue is that you must pay extra attention to the use of `printf` or `cout`. You want all your logging message to go to either a file or STDERR.

That was also one concern I had. I don't think we need to use STDOUT for anything else, but I didn't want to dedicate it to RPC right away. TCP has the advantage that an external program could also control an instance of GTKWave that was started in another way, without the need for the controlling application to capture the STDIO of the GTKWave process.

suzizecat, 2024-06-26:
> My comment about the text based RPC was a bit confusing, I guess. I could only come up with examples like this, where STDIO is used to call another program with some data to process it, but not to keep open a long term remote control interface.

In this particular case, it is interesting because I believe that GTKWave streams data, in continuous, to the underlying process. It is not just a single call with the data and "just capturing" the output of the process.

> That was also one concern I had. I don't think we need to use STDOUT for anything else, but I didn't want to dedicate it to RPC right away. TCP has the advantage that an external program could also [...]

My recommendation in this aspect, would be to always use a dedicated function/library when printing stuff to the console, in order to easily redirect all traffic if you need to keep stdout clean.
For logging purpose (and if you can afford the modern C++ requirements), I cannot recommend enough [spdlog](https://github.com/gabime/spdlog). But I'm not sure at all that it comply with the GTKWave stack (C/C++ and versions)

rfuest, 2024-06-27:
> My recommendation in this aspect, would be to always use a dedicated function/library when printing stuff to the console, in order to easily redirect all traffic if you need to keep stdout clean. [...]

I agree that logging could to be improved. If there is important information in the log it should also be available in the UI (see https://github.com/gtkwave/gtkwave/issues/231). A logging framework, like the one you suggested, would make this easier and cleaner to implement. We don't use any C++ in the main GTKWave application and libgtkwave and I'm planning to the the logging support that is integrated into GLib (https://docs.gtk.org/glib/logging.html).

suzizecat, 2024-07-31:
To add two (updated) cents to this discussion with regards to the usefulness of this Tcl interface,
I managed to implement in VSCode the same kind of feature as you have with RTLBrowse, without relying on stem files.

[image]

You can see the result [here](https://suzizecat.github.io/diplomat-lsp/vscode/sim_debug.html#inline-value-display) and the used part of the Tcl API [here](https://github.com/suzizecat/diplomat-vscode/blob/master/resources/gtkwave_setup.tcl)

(I must precise that as of today, the annotation is placed using a text search in the file, rather than an actual symbol lookup, but I was excited to have a result. In a short term, I will fix that using a proper symbol lookup).

rfuest, 2024-07-31:
That looks neat. Could you write a short summary of all RPC functions you use at the moment or plan to use in the future, so that we can make sure the new RPC mechanism will work for your use case. In some cases the new RPC could offer an easier alternative, like a function to get the value of a signal at a given time instead of the `signalChangeList` function.

Why are you decrementing the time by 1 in this code?
https://github.com/suzizecat/diplomat-vscode/blob/16e456291db26b5bf7de0a853a1ed1b7fbad5734/resources/gtkwave_setup.tcl#L58-L60
Doesn't this make the displayed values inconsistent with the ones displayed in GTKWave?

suzizecat, 2024-07-31:
> Why are you decrementing the time by 1 in this code?
> https://github.com/suzizecat/diplomat-vscode/blob/16e456291db26b5bf7de0a853a1ed1b7fbad5734/resources/gtkwave_setup.tcl#L58-L60
> Doesn't this make the displayed values inconsistent with the ones displayed in GTKWave?

Yes it does.

My usecase for this whole feature is to easily find why a signal got its current value and evaluate the conditions that leads to its value. Therefore, I am often inspecting an edge and I am almost always interested in the value producing this edge, hence the value of signals right before the edge.

Ultimately, I will probably add an option in the extension to enable this decrement and let the user choose to have either
 - The value after the edge
 - The value before the edge
 - The transition, shown as `0 -> 1` (or equivalent) 

> Could you write a short summary of all RPC functions you use at the moment or plan to use in the future, so that we can make sure the new RPC mechanism will work for your use case.

As of today, the features that I requires are:
 - Knowing that the user moved the marker (`tell_time_updated` [which is invoked every 500ms](https://github.com/suzizecat/diplomat-vscode/blob/16e456291db26b5bf7de0a853a1ed1b7fbad5734/src/waveform_viewer.ts#L321-L331)) 
 - Getting the value of any (list of) signals, at a given time, referenced by their absolute path and somehow associated to their path (or at a bare minimum, to their name, but the path is simpler) : `get_signals_values` . When getting the value, also getting the kind of representation used will be useful to convert to other bases if the user wish so (hence getting the flag) 
 - Getting the currently selected time.

I will also, at some point, add, if possible:
 - a `add signal to waves` function (selection in VS Code, right click, wave added to the current GTKWave) 
 - a mirroring of the design hierarchy that I show in VS Code to GTKWave SST, so that you can easily be in the same place in VS Code and GTKWave 

The features that are not used here (because not available) but would be nice are:
 - Getting the list of values for all signals in a scope at once (e.g. I provides `u_dut.u_ip` and gtkwave return data for all signals in `u_dut.u_ip`
 - Depending on the RPC protocol, it may be of interest to have a map `short ID, e.g. a number <=> full signal name` so I could retrieve this map when opening the wave file and use the IDs as fully qualified path later on in my interactions (again, when I need to manipulate big set of files)

The overall idea is to minimize the amount of data transmitted to issue a command because that may impact performances for very big files. I have, however, not yet encountered a situation where such optimization is required for a smooth operation.

electronicsguy, 2025-10-04:
"I'm not sure I understand your feature request. Do you want to export some kind of interactive web based viewer from GTKWave that can be displayed in a browser or inside an Electron app"

Yes. Something like what waveform/VCDROM does.