gtkwave/gtkwave #366: App name contains part of GTK trademark

aklsh, 2024-06-18:
https://foundation.gnome.org/legal-and-trademarks/

Please rename it to not contain "GTK" and something distinct. I think 2 months is a reasonable time to come to a solution here.

_Originally posted by @bbhtt in https://github.com/flathub/io.github.gtkwave.GTKWave/issues/19_

rfuest, 2024-06-19:
@tbybell What should we do about this? I find it strange to get such an important message with an ultimatum that the application will be removed from a GitHub account that has no real name and isn't publicly listed as a member of the Flathub GitHub organization. From their GitHub activity they seem to be involved in the project, but we have no way to verify this. This is not what I would have expected for a big project like Flathub. There are also other projects with GTK in their name on Flathub, which haven't received a similar warning (yet?). Some of these projects use "Gtk", but I'm not sure if trademarks are case sensitive.

We obviously don't want to rename the application and in the worst case we have to stop distribution via Flathub. But that wouldn't be great, because the desktop Linux world seems to slowly migrate to Flatpaks as the preferred distribution method for GUI applications. Perhaps you could contact the GNOME foundation as the project owner to ask for permission to use their trademark? (The licensing email address is mentioned in the document that is linked above). I hope that they understand our situation, because GTKWave has existed almost existed as long as GTK and long before they registered the GTK trademark, which was registered in the end of 2019.

tbybell, 2024-06-19:
At the moment, I do not plan to take any action until directly contacted by a legal staff member who is a representative of the GNOME Foundation or one who is acting provably and directly on their behalf.  As trademark holder, it is their responsibility to do their own policing for enforcement and compliance.  If they indicate to me that there is an issue of trademark dilution or infringement that requires action upon our part, we will then proceed to remediate the situation in a mutually agreed upon manner.  

I don't know what the case is with Flathub, but project repackagers and maintainers such as Mr. Bhattacharya do not speak for the file hosting organization Flathub.  However, if he so chooses to exercise his discretion to delist gtkwave, it is his option as maintainer of that specific package on that site.

umarcor, 2024-06-19:
@rfuest, moreover, there is no reference to why that's a problem for flathub; it be just the opinion of a contributor.

rfuest, 2024-06-19:
> I don't know what the case is with Flathub, but project repackagers and maintainers such as Mr. Bhattacharya do not speak for the file hosting organization Flathub. However, if he so chooses to [...]

The person who has reported the alleged trademark issue is not the maintainer of GTKWave on Flathub. The package is maintained by @aklsh.

> @rfuest, moreover, there is no reference to why that's a problem for flathub; it might be just the opinion of a contributor.

They have some guidelines about trademarks in their docs: https://docs.flathub.org/docs/for-app-authors/metainfo-guidelines/quality-guidelines#no-trademark-violations. But that paragraph is very vague.

tbybell, 2024-06-19:
> The person who has reported the alleged trademark issue is not the maintainer of GTKWave on Flathub. The package is maintained by @aklsh.

Thanks for the clarification.  OK, the maintainer is Akilesh, who opened this issue, and the person who raised a complaint against the package name on him on Flathub is bbhtt.  I don't know who this Mr. Bhattacharya (bbhtt) is: Ph.D students in mathematics are not attorneys.

> They have some guidelines about trademarks in their docs: https://docs.flathub.org/docs/for-app-authors/metainfo-guidelines/quality-guidelines#no-trademark-violations. But that paragraph is very [...]

One could probably argue that at its release in 1998 (gtkwave was an early adopter that year some number of months later), the GIMP ToolKit may have received common law trademark status, but I'm not an IP attorney: https://www.bitlaw.com/trademark/common.html

They certainly did not claim trademark status back in the old days before the Foundation.

```
532:/tmp/w/gtk+-1.3.10> fgrep -ir trademark .
./gdk/gdkkeyuni.c:  { 0x0ac9, 0x2122 }, /*                   trademark ™ TRADE MARK SIGN */
./gdk/gdkkeyuni.c:/*  0x0acb                         trademarkincircle ? ??? */
./gdk/gdkkeyuni.c:  { 0x0ac9, 0x2122 }, /*                   trademark ™ TRADE MARK SIGN */
./gdk/gdkkeysyms.h:#define GDK_trademark 0xac9
./gdk/gdkkeysyms.h:#define GDK_trademarkincircle 0xacb
./gdk/win32/gdkkeys-win32.c:  { 0x000ac9, "trademark" },
./gdk/win32/gdkkeys-win32.c:  { 0x000acb, "trademarkincircle" },
./docs/reference/gdk-pixbuf/gdk-pixbuf.sgml:	products and services are claimed as trademarks. Where those
./docs/reference/gdk-pixbuf/gdk-pixbuf.sgml:	names appear in any GNOME documentation, and those trademarks
./docs/reference/gdk-pixbuf/html/index.html:	products and services are claimed as trademarks. Where those
./docs/reference/gdk-pixbuf/html/index.html:	names appear in any GNOME documentation, and those trademarks
```

Again, as I said, _if there are issues_, I would like to hear directly from someone from GNOME Foundation.  I'm not going to take a layperson's read on IP Law.  I worked too long at IBM to know such things are rarely black and white.  For all I know, statements providing trademark attribution as well as something stating that gtkwave is not affiliated with GNOME Foundation may suffice.  However, I also would not be surprised to be told "[they] must vigorously protect [their] valuable assets."  I can't guess.  I will not, however, do their enforcement work for them.

If this were a new product, sure, I would say we should change the name conservatively to avoid potential issues and I wouldn't put it up for debate.  I've been there before back in 1991, ironically versus another (albeit unrelated) acronym owned by HP.  However, this is a long established product with literally hundreds or thousands of references in academic papers, there are runscripts in sim farms depending on the name, etc. This isn't some popup chat client.  That's the source of my resistance.

If you wondered why some products seem to use this method below for naming, all of the above is why.  :-)
```
[bybell@bybell-ThinkPad-T420s dict]$ cat words | head -`echo 1$RANDOM` | tail -1
eschara

```

rfuest, 2024-06-20:
This has just been resolved: https://github.com/flathub/io.github.gtkwave.GTKWave/issues/19#issuecomment-2180756978

closed 2024-06-20 by rfuest

tbybell, 2024-06-20:
Excellent, thanks!