gtkwave/gtkwave #371: [Feature] Programmatically retrieve the highlighted nets in the viewer

suzizecat, 2024-07-10:
Hello there !

In the context of the development of a VSCode extension to work on SystemVerilog projects, I would like to implement a way to go back and forth between GTKWave and VSCode.

To go from VSCode to GTKWave, I will probably be able to use some Tcl command to add a signal, based upon its absolute path, to the viewer. However I can't find a way to retrieve a list of highlighted signals through Tcl, my objective being showing the right file (in VSCode) each time you select a signal (in GTKWave).

I found that I may use the selection on the SST part, through a callback such as `gtkwave::cbTreeSigSelect` but I would rather trigger on the signal selection in the waveform part of GTKWave.

I haven't found either a Tcl command that I may call periodically that would return this signal list.

Would there be a way that I am not aware of, or could this feature be implemented ?

I am using GTKWave Analyzer v3.3.104 and am aware of #358 

Best regards,
Julien FAUCHER

tbybell, 2024-07-10:
Use getDisplayedSignals and getTraceFlagsFromName?

There's not a way to do this based on selection currently.  It should be relatively easy because those callbacks are invoked simply by reaching into the Tcl interpreter to set a variable like this:

`gtkwavetcl_setvar(WAVE_TCLCB_TREE_SIG_SELECT, name, WAVE_TCLCB_TREE_SIG_SELECT_FLAGS);`

A similar function call would be inserted into the if(t) set of highlight selection code in signalwindow.c if I'm reading correctly what you're asking for.

```
else if( (!GLOBALS->use_standard_trace_select) || (GLOBALS->standard_trace_dnd_degate) || ((t)&&(!(t->flags & TR_HIGHLIGHT))) )
                {
                GLOBALS->starting_unshifted_trace = t;

                t2=GLOBALS->traces.first;
                while(t2)
                        {
                        t2->flags &= ~TR_HIGHLIGHT;
                        t2 = t2->t_next;
                        }

                if(t) { t->flags |= TR_HIGHLIGHT; MarkTraceCursor(t); } /* scan-build */
                }
```

suzizecat, 2024-07-10:
Your first solution (getDisplayedSignals and getTraceFlagsFromName) would be correct if the "selected" state is in the flags.

Could you point me toward the definition of the flags values? I couldn't find any.

tbybell, 2024-07-10:
analyzer.h, search for "enum TraceEntFlagBits".  TR_HIGHLIGHT tells you what is highlighted/selected.

suzizecat, 2024-07-10:
> analyzer.h, search for "enum TraceEntFlagBits". TR_HIGHLIGHT tells you what is highlighted/selected.

Perfect !

So my solution as of now is to repeatedly call the following procedure that works like a charm:

```tcl
set last_selected ""

proc tell_selected { } {
    global last_selected

    foreach name [gtkwave::getDisplayedSignals] {
        if {$name eq $last_selected} {continue}
        set value [gtkwave::getTraceValueAtMarkerFromName $name]
        set flag [gtkwave::getTraceFlagsFromName $name]

        if {$flag%2 == 1} {
            puts "Selected $name have value $value"
            set last_selected $name
            break
        }
    }
}
```

A callback would indeed be more efficient, but as a proof of concept, this will do !
Feel free to close the issue as answered and thank you for your quick feedbacks ! 🙂

closed 2024-07-11 by tbybell