gtkwave/gtkwave #372: Reloading wavefrom cause crash.

FrankOu2001, 2024-07-17:
Git Commit: e9b5cce38c5c5272d60ff9baf71df027b3607ebd
OS: Debian GNU/Linux trixie trixie/sid x86_64
Kernel: Linux 6.9.8-amd64
GCC: gcc (Debian 13.3.0-1) 13.3.0
GTK version: 3.24.43-1

When I add a signal and then click Reload to try to reload the waveform, GTKWave crashes and exits!

I tried to run gtkwave in gdb and got the output:

```
(gdb) set args xxx.fst
(gdb) run
Starting program: /usr/local/bin/gtkwave xxx.fst
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff58006c0 (LWP 1033992)]
[New Thread 0x7ffff4e006c0 (LWP 1033993)]
[New Thread 0x7fffefe006c0 (LWP 1033994)]
[New Thread 0x7fffef4006c0 (LWP 1033995)]
[New Thread 0x7fffeea006c0 (LWP 1033996)]

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

FSTLOAD | Processing 1872 facs.
FSTLOAD | Built 1083 signals and 789 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
[New Thread 0x7fffee0006c0 (LWP 1033997)]
GTKWAVE | Touch screen detected, enabling gestures.
[New Thread 0x7fffe36006c0 (LWP 1033998)]
[Thread 0x7fffe36006c0 (LWP 1033998) exited]
GTKWAVE | Reloading waveform...

** (gtkwave:1033989): CRITICAL **: 15:32:59.275: gw_dump_file_get_global_time_offset: assertion 'GW_IS_DUMP_FILE(self)' failed

** (gtkwave:1033989): CRITICAL **: 15:32:59.275: gw_dump_file_get_time_dimension: assertion 'GW_IS_DUMP_FILE(self)' failed

** (gtkwave:1033989): CRITICAL **: 15:32:59.275: gw_dump_file_get_time_range: assertion 'GW_IS_DUMP_FILE(self)' failed

** (gtkwave:1033989): CRITICAL **: 15:32:59.276: gw_dump_file_get_blackout_regions: assertion 'GW_IS_DUMP_FILE(self)' failed

** (gtkwave:1033989): CRITICAL **: 15:32:59.276: gw_blackout_regions_contains: assertion 'GW_IS_BLACKOUT_REGIONS(self)' failed
FSTLOAD | Processing 1872 facs.
FSTLOAD | Built 1083 signals and 789 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.

Thread 1 "gtkwave" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f80699 in GW_IS_TIME_RANGE (ptr=ptr@entry=0x5555557aaec0)
    at ../lib/libgtkwave/src/gw-time-range.h:9
9       G_DECLARE_FINAL_TYPE(GwTimeRange, gw_time_range, GW, TIME_RANGE, GObject)
```

rfuest, 2024-07-18:
The way reload is implemented at the moment it is very easy to accidentally break and it cannot be tested in CI. At this point in the refactoring process I'm not paying much attention to the UI so that broken features are more likely.

I hope the changes in #373 have fixed this issue, but reload has been broken before and I cannot guarantee that it won't break again in the future. If you need a more stable version of GTKWave I would suggest to stick with the LTS releases at the moment.

FrankOu2001, 2024-07-19:
It works now. Thank you!

closed 2024-07-19 by FrankOu2001