gtkwave/gtkwave #378: Cannot load dumpfile in new window

yikkrubz, 2024-08-10:
When I try to load a dumpfile via File -> Open New Window, gtkwave prints in the terminal: Unable to find application named 'gtkwave' (see below). Loading the dumpfile via File -> Open New Tab or as a command line argument works as expected.

gtkwave git version updated today
macos 16.6.1 (intel)
MacPorts base version 2.10.0

$ which gtkwave
/usr/local/bin/gtkwave

$ gtkwave
(gtkwave:85994): GLib-GObject-CRITICAL **: 19:53:42.397: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'marker-baseline'

(gtkwave:85994): GLib-GObject-CRITICAL **: 19:53:42.397: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'waveform-value'

(gtkwave:85994): GLib-GObject-CRITICAL **: 19:53:42.397: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'vector-stroke'

(gtkwave:85994): GLib-GObject-CRITICAL **: 19:53:42.397: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'marker-unnamed'

(gtkwave:85994): GLib-GObject-CRITICAL **: 19:53:42.397: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'marker-named'

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

GTKWAVE | Use the -h, --help command line flags to display help.
Unable to find application named 'gtkwave'

```
git diff
diff --git a/lib/libgtkwave/src/gw-fst-file.c b/lib/libgtkwave/src/gw-fst-file.c
diff --git a/lib/libgtkwave/src/gw-fst-file.c b/lib/libgtkwave/src/gw-fst-file.c
index 9256fa4..c5e359c 100644
--- a/lib/libgtkwave/src/gw-fst-file.c
+++ b/lib/libgtkwave/src/gw-fst-file.c
@@ -262,7 +262,7 @@ static void fst_callback2(void *user_callback_data_pointer,
 }
 
 static void fst_callback(void *user_callback_data_pointer,
-                         guint64 tim,
+                         uint64_t tim,
                          fstHandle txidx,
                          const unsigned char *value)
 {
diff --git a/meson.build b/meson.build
index d1edbe4..8e41f25 100644
--- a/meson.build
+++ b/meson.build
@@ -228,6 +228,5 @@ subdir('contrib')
 
 gnome.post_install(
     update_mime_database: get_option('update_mime_database'),
-    update_desktop_database: true,
     gtk_update_icon_cache: true,

```

rfuest, 2024-08-11:
Try to change this line: https://github.com/gtkwave/gtkwave/blob/d2005675801b89ef0201c6968cee8419847dc5af/src/menu.c#L2337
into
```c
#if 1
```
to see if this fixes the issue.

yikkrubz, 2024-08-11:
Thank you very much for your reply.

While running meson setup build, meson detects gtk-mac-integration-gtk3, gtk3 and gtk4.
The gtkwave binary is linked with libgtk3 and libgtkmacintegration-gtk3.4 (see below).

When I change the #ifdef MAC_INTEGRATION to a #if 0 in line 2337 of src/menu.c, the issue is resolved.

gtkwave git version updated today
macos 14.6.1 (intel)
MacPorts base version 2.10.0

$ which gtkwave
/usr/local/bin/gtkwave

$ meson setup build
The Meson build system
Version: 1.5.1
Source dir: /Users/Peter/Downloads/tmp/gtkwave
Build dir: /Users/Peter/Downloads/tmp/gtkwave/build
Build type: native build
Project name: gtkwave
Project version: 3.4.0
C compiler for the host machine: cc (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.3.9.4)")
C linker for the host machine: cc ld64 1053.12
C++ compiler for the host machine: c++ (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.3.9.4)")
C++ linker for the host machine: c++ ld64 1053.12
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/opt/local/bin/pkg-config) 0.29.2
Run-time dependency glib-2.0 found: YES 2.78.4
Run-time dependency gobject-2.0 found: YES 2.78.4
Run-time dependency gtk+-3.0 found: YES 3.24.42
Run-time dependency gtk4 found: YES 4.14.3
Run-time dependency gtk+-unix-print-3.0 found: YES 3.24.42
Run-time dependency gtk-mac-integration-gtk3 found: YES 3.0.1
Run-time dependency zlib found: YES 1.3.1
Library m found: YES
Has header "Judy.h" : NO 
Has header "regex.h" : YES 
Library regex found: NO
Run-time dependency threads found: YES
Found CMake: /opt/local/bin/cmake (3.29.5)
Run-time dependency bzip2 found: YES 1.0.8
Program gperf found: YES (/usr/bin/gperf)
Program flex found: YES (/opt/local/bin/flex)
Has header "inttypes.h" : YES 
Has header "sys/stat.h" : YES 
Has header "alloca.h" : YES 
Has header "getopt.h" : YES 
Has header "fcntl.h" : YES 
Has header "unistd.h" : YES 
Checking for function "fseeko" : YES 
Checking for function "realpath" : YES 
Checking for function "setenv" : YES 
Checking for function "unsetenv" : YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Werror=address: YES 
Compiler for C supports arguments -Werror=array-bounds: YES 
Compiler for C supports arguments -Werror=empty-body: YES 
Compiler for C supports arguments -Werror=implicit: YES 
Compiler for C supports arguments -Werror=implicit-function-declaration: YES 
Compiler for C supports arguments -Werror=incompatible-pointer-types: YES 
Compiler for C supports arguments -Werror=init-self: YES 
Compiler for C supports arguments -Werror=int-conversion: YES 
Compiler for C supports arguments -Werror=int-to-pointer-cast: YES 
Compiler for C supports arguments -Werror=main: YES 
Compiler for C supports arguments -Werror=misleading-indentation: YES 
Compiler for C supports arguments -Werror=missing-braces: YES 
Compiler for C supports arguments -Werror=missing-include-dirs: YES 
Compiler for C supports arguments -Werror=nonnull: YES 
[... 83 lines cut ...]
diff --git a/meson.build b/meson.build
index d1edbe4..8e41f25 100644
--- a/meson.build
+++ b/meson.build
@@ -228,6 +228,5 @@ subdir('contrib')
 
 gnome.post_install(
     update_mime_database: get_option('update_mime_database'),
-    update_desktop_database: true,
     gtk_update_icon_cache: true,
 )
diff --git a/src/menu.c b/src/menu.c
index 5a476e2..18ae716 100644
--- a/src/menu.c
+++ b/src/menu.c
@@ -2334,7 +2334,7 @@ void menu_new_viewer_cleanup(GtkWidget *widget, gpointer data)
             return;
         }
 
-#ifdef MAC_INTEGRATION
+#if 0
         /* from : @pfx = split(' ', "open -n -W -a gtkwave --args --chdir dummy"); */
         if (GLOBALS->optimize_vcd) {
             execlp("open",
```

rfuest, 2024-08-12:
A related issue just came up in #250. The code that is used when `MAC_INTEGRATION` is defined seems to only work when GTKWave is installed as a macOS `.app`. I'll temporarily disable this code in the development version, because Homebrew is the only semi supported way of installing the development version of GTKWave on macOS at the moment.

rfuest, 2024-08-12:
I've committed fixes for the type issue and the open in new window feature the the repo. (#379)

yikkrubz, 2024-08-12:
Thank you very much for committing the fixes.

closed 2024-08-12 by yikkrubz