gtkwave/gtkwave #382: Simulation time greater 9 seconds Warning: max_time bits > DBL_MANT_DIG (53) is printed

yikkrubz, 2024-09-14:
When I simulate top_tb with ghdl or nvc for more than 9 seconds, gtkwave does not accept the dumpfile generated. 
The message 
GTKWAVE | Warning: max_time bits > DBL_MANT_DIG (53), GUI may malfunction!
GTKWAVE | Exiting, use dbl_mant_dig_override rc var set to 1 to disable.
is printed (see below).

Older versions of gtkwave were able to display the wave form.

I assume the maximum simulation time is approximately 2.5 hours ((2^63 - 1) * 1 fs) at 1 fs time resolution for VHDL.

Am I missing something?

top_tb.vhd

```
library` ieee;
use ieee.std_logic_1164.all;

entity top_tb is
end top_tb;

architecture sim of top_tb is

  constant clk_hz : integer := 100;
  constant clk_period : time := 1 sec / clk_hz;

  signal clk : std_logic := '1';
 
begin

  clk <= not clk after clk_period / 2;

end architecture;
```

ghdl -a  top_tb.vhd
ghdl -e  top_tb
ghdl -r top_tb --wave=top_tb.ghw --stop-time=9010ms
./top_tb:info: simulation stopped by --stop-time @9010ms

gtkwave top_tb.ghw top_tb.gtkw

(gtkwave:91075): GLib-GObject-CRITICAL **: 00:27:51.302: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'marker-baseline'

(gtkwave:91075): GLib-GObject-CRITICAL **: 00:27:51.302: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'waveform-value'

(gtkwave:91075): GLib-GObject-CRITICAL **: 00:27:51.302: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'vector-stroke'

(gtkwave:91075): GLib-GObject-CRITICAL **: 00:27:51.302: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'marker-unnamed'

(gtkwave:91075): GLib-GObject-CRITICAL **: 00:27:51.302: g_object_set_is_valid_property: object class 'GwColorTheme' has no property named 'marker-named'

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

[0] start time.
[9010000000000000] end time.
GTKWAVE | Warning: max_time bits > DBL_MANT_DIG (53), GUI may malfunction!
GTKWAVE | Exiting, use dbl_mant_dig_override rc var set to 1 to disable.

macos 14.6.1
macports

rfuest, 2024-09-15:
While GTKWave usually uses a signed 64 bit integer to represent time values, it does use floating point calculations for some display related operations like zooming into a waveform. If the time range exceeds the 53 bit precision of a double some amount of rounding will be applied. This doesn't cause huge problems if you just slightly overflow the range, but it will make GTKWave unusable if the dump file uses the entire range of a 64 bit integer.

If you didn't have any problems before you can just set the rc variable mentioned in the warning to restore the previous behaviour.

yikkrubz, 2024-09-19:
Thank you very much for your reply.

closed 2024-09-19 by yikkrubz