gtkwave/gtkwave #386: I have tried all possible solutions to rectify the problem and it is showing update required for "gtkwave-bin"

shashankteli, 2024-10-05:
[image]

rfuest, 2024-10-05:
The screenshot isn't linked correctly and isn't displayed. Without additional information we cannot help you any further. What do you mean by "all possible solutions"? Did you try the homebrew version from #250?

shashankteli, 2024-10-06:
[image]
I have tried homebrew version from #250 but it is not working and also confusing,can you say exact steps.

rfuest, 2024-10-07:
> I have tried homebrew version from https://github.com/gtkwave/gtkwave/issues/250 but it is not working and also confusing,can you say exact steps.

No I can't, because I don't have access to a Mac. If you have homebrew installed you should be able to install GTKWave with the command I gave you in #384. As I suggested before, there might be other Mac users that can help you if you explain your problem in #250.

But without additional information nobody will be able to help you fix this issue. In which way did the homebrew version of GTKWave "not work"? Did you have issues installing it or did you encounter problems when you tried to run it? Where there any error messages?