gtkwave/gtkwave #390: Provide updated prebuilt binary for macOS

yanjiew1, 2024-10-27:
Currently, the provided binary cannot be run on macOS 14 and later. #250 

Since #389 is merged, are there any plans to provide updated prebuilt binary for macOS?

I am trying to contribute a formula for gtkwave to the homebrew/core tap.  (See <https://github.com/Homebrew/homebrew-core/pull/195646>) However, Homebrew maintainers thought that it will be better if the gtkwave cask can be fixed.

tbybell, 2024-10-27:
I don't own a Mac, so I can't do this.  Until Apple cut off backward compatibility, I was doing my binary builds on Snow Leopard (2009) under emulation/virtualization.

yanjiew1, 2024-10-28:
I see.  I will try to setup a GitHub workflow and use the free GitHub Action Runner to build the macOS binary.

juliusbaxter, 2024-11-11:
I managed to get it built today on OS X 14.7 Sonoma using brew.

```
brew install desktop-file-utils shared-mime-info gobject-introspection gtk-mac-integration meson ninja pkg-config gtk+3 gtk4 tcl-tk xz gcc
```

With the meson build, I had issue with the `xml2stems.cc` file not getting compiled properly. Seems it tries to use the system's clang which, as hard as I tried, I couldn't get it to build c++ and include the c++ standard libraries (kept barfing on `include <string>`). So I told it to use the brew-installed `g++-14` for the CPP compiler and it appears to have worked.

```
CXX=g++-14 meson setup build
cd build
meson install
```

shamsmm, 2024-11-14:
> I managed to get it built today on OS X 14.7 Sonoma using brew.
>
> ```
> brew install desktop-file-utils shared-mime-info gobject-introspection gtk-mac-integration meson ninja pkg-config gtk+3 gtk4 [...]

I have a linker issue what should I pass to `meson`?
`ld: multiple errors: archive member '/' not a mach-o file in '/Users/shams/T/gtkwave/gtkwave/build/lib/libghw/libghw.a'; archive member '/' not a mach-o file in '/Users/shams/T/gtkwave/gtkwave/build/lib/libjrb/libjrb.a'; archive member '/' not a mach-o file in '/Users/shams/T/gtkwave/gtkwave/build/lib/libfst/libfst.a'`