gtkwave/gtkwave #391: GTKwave UI is looking poor in my setup

rudro101, 2024-11-17:
After checking the GTKWave website, I noticed that the interface showcased there appears significantly more refined than what I’m seeing on my setup. Wondering if there are settings or tweaks I might be missing to improve the UI quality?
I'm using GTKWave on WSL2. Distro is Ubuntu.

[image]
This was in the gtkwave site

[image]
This is in my setup.

tbybell, 2024-11-17:
> ![Screenshot 2024-11-17
> 195717](https://github.com/user-attachments/assets/dd9686d1-11dc-4439-96f4-fc0f6e3cf800)

It's probably default theme-related.

-Tony

rudro101, 2024-11-18:
Can you please tell me how I can enhance the UI looking?