gtkwave/gtkwave #392: On Apple M3, meson setup incorrectly identify as x86_64

alansuphd, 2024-11-30:
I am using a MacBook Pro with Apple M3 chip. Following the compiling instructions in README.md, after all needed tools and packages are installed, then download from git the gtkwave, then at meson setup, it finds:

Host machine cpu family: x86_64
Host machine cpu: x86_64

Nonetheless, I tried to proceed with meson compile. At step [253/257] Linking target contrib/rtlbrowse/rtlbrowse, error messages started with

FAILED: contrib/rtlbrowse/rtlbrowse

then after hundreds of lines of error messages, the compilation stops with below messages and appears to be the problem by mistaking Apple M3 as x86_64:

ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[256/257] Compiling C++ object contrib/xml2stems/xml2stems.p/xml2stems.cc.o
../contrib/xml2stems/xml2stems.cc:169:37: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
  169 |                                     sprintf(genname, "%s:G%d", nam, geninst++);
      |                                     ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/_stdio.h:274:1: note: 'sprintf' has been explicitly marked deprecated here
  274 | __deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
      | ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:218:48: note: expanded from macro '__deprecated_msg'
  218 |         #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
      |                                                       ^
1 warning generated.
ninja: build stopped: subcommand failed.

tbybell, 2024-11-30:
There are about 250+ sprintf code declarations in the codebase which roughly matches your warnings count.  For the time being, turn off that warning.  The code has gone through a security audit for these kinds of buffer overflows and should know what it is doing.

alansuphd, 2025-01-09:
I have the same problem. Any suggestions?

rfuest, 2025-01-09:
My guess would be that meson is running using Rosetta 2 in x86_64 mode. Perhaps you did change the config of your terminal in the past to run your shell this way? I don't own a Mac and can't give you detailed advice, but there seems to be an `arch` command to show the current architecture the shell is using or to start a new shell with a different architecture.

If you are sure that meson is invoked in ARM mode it would be helpful if you upload the full meson log (`build/meson-logs/meson-log.txt`) to debug the issue further.

umazhar, 2025-01-27:
Any suggestions? I have the same issue.

rfuest, 2025-01-27:
Did you try the suggestions in my previous comment?