gtkwave/gtkwave #394: Allow copying values selected trace values and horizontal scroll

B83C, 2024-12-06:
The title is self evident. Maybe make the delimiter configurable for copied trace values.

tbybell, 2024-12-06:
It's not as self-evident as you think.  :-)

Please state exactly what you want.  If it's that you desire the value at the primary marker, see View->Show Mouseover and View->Mouseover Copies to Clipboard.

I don't know specifically what you mean by horizontal scroll.  There is plenty of horizontal scrolling function for various usage cases in there, but you don't say what yours is, thanks.

B83C, 2024-12-07:
Sorry for being unclear. Yeah, what I mean was copying a bunch of selected values to clipboard instead. 
[image]
For instance, I would like to copy values of the highlighted entries with ease, giving me smth like this: 01 02 01 01 C2 01... Would be nice to have the element delimiter configurable (i.e. comma separated, or smth of the user's interest). 
Also, regarding horizontal scroll, I meant scrolling with mouse and key modifiers (i.e. shift + scrll up/down to scroll horizontally) cuz I find it rather annoying to reach out for the scrollbars on a small screen. Would be better if keyboard navigation is added.

tbybell, 2024-12-08:
A mouse with a scroll wheel has worked for me for forever and it's how I use gtkwave on my day job.  The mouse is context sensitive to the vertical vs horizontal direction of scroll for signals vs waves panes.  What system (specifically) are you running gtkwave on?  It works for me even through a couple of levels of indirection: my work laptop is a Windows 10 machine that runs Citrix into another Windows machine which is running an x server connected to a Linux development machine.  :-)

From the keyboard, menu options Time->Shift/Page are keys Alt-5/6 and Alt-7/8 which a left/right shift and page respectively.  Those are programmable using the gtkwaverc file seen in examples/ in the source distribution.  (Rename the file to .gtkwaverc and copy it to your home directory is easiest.)

For the clipboard copy, I'll look at it and some other versions of it.  Any of those GUI data harvesting operations are pretty easy to do: it's more effort to figure out what to do compared to sitting down and coding them.  It's the kind of thing that could probably even be done with a straightforward Tcl script, but gtkwave doesn't yet have the ability to have hotkeys bindings to kicking off individual Tcl scripts.

What is your current usage case for these values?  Generating documentation or something more involved?

B83C, 2024-12-29:
This is weird, scroll wheel only works for me when I hover my cursor over the scroll bar. And I am on Arch linux (sway) using version 3.4.0. I was coding up some compression algorithm and had to test the compression result. I know extracting the values directly from gtkwave would be cumbersome, but I was a bit lazy to write a test in C to do it programmatically. But anyways, that would be a handy feature to have. Now on linux, I feel that the interface is kinda unfinished. That is, in addition to the issue I mentioned above, common keybindings like ctrl+a doesn't select all the entries in the wavepane, etc. Can't recall all the issues I faced. But yeah, it is a good project after all. I'd love to help but only after I find some time after this semester ends.

tbybell, 2024-12-29:
Development version of gtkwave or LTS?

The scroll wheel issue might be your window manager.  The wheel is context sensitive based on where your mouse pointer is at.  I'm not at my PC to test if you have to click in the window/pane first. If you're in the wave pane, it should scroll left/right, but in the signal one directly to the left of the wave pane, the wheel should make scrolling go up and down.  There should be no real need to have scrollbar focus--especially with this business of hidden scrollbars on some window managers.

How do you want to extract values?  At the marker time for selected signals?  Results go to the cut/paste clipboard?  A user asked for such a feature earlier in the month.  It's easy enough of a feature add.  I was considering such a thing for my day job in order to cut down on notes I take by hand.

Note that you can do this with a Tcl script, but it's more work.

Side note is that it's not that the tool is not exactly...unfinished...as it's been in existence and development since 1998, but I haven't documented everything it does.  A comprehensive set of documentation would be quite large and not be read by most users anyway.  

A problem is that different window managers and OSs intercept keystrokes and events in different ways.  I haven't used Arch yet.  I'm still on RHEL 7 with ongoing corporate support and use fvwm as a window manager, mainly because I want clean, fast, and reliable.  Ctrl-A and such work fine for me.

I've used gtkwave for FSDB files instead of Verdi (too slow and clunky) to debug for the day job.  Your computer may very well have a processor in it that I debugged my unit RTL with gtkwave.  For most features I felt were missing, I have long since added them (or user code submissions have added them for me).  That said, if something appears missing, it might be, but ask first.  :-)

-Tony