gtkwave/gtkwave #399: release 3.4 ?

csantosb, 2025-01-07:
Hi, any hope for a 3.4 release ? I'd like to upgrade current package under [guix](https://git.savannah.gnu.org/cgit/guix.git/tree/gnu/packages/fpga.scm?h=master#n447)
Thanks,

tbybell, 2025-01-08:
What are you considering 3.4?  Is it what Ralf is working (which I think merits a bump to 4.0) on or feature adds to the existing 3.3 branch?

csantosb, 2025-01-10:
I misunderstood the devel model, sorry.

By the time being, a v3.3.122 tag on lts branch would be enough. 

Concerning master, 3.4 tag or 4.0, would be useful too for packaging.

tbybell, 2025-01-14:
3.3.122 in git/svn so far is only compiler warnings fixes that aren't showstoppers.  It's not a must have version at the moment:

3.3.122 08jan25 Buffer warning fix in fstVcdIDForFwrite.
		Warning fixes in vzt_read.c and fstapi.c.
		
I'll do a release probably in a month or so, or earlier if anything compelling turns up.

closed 2025-01-14 by csantosb