gtkwave/gtkwave #400: Small new install issues

aolofsson, 2025-01-09:
Hi,

Long time user of gtkwave (since 2008), thank you for this amazing project. Small issues I found on a recent install.

1. readme depedency was missing, needed to install following first:
sudo apt install libgtk-4-dev

2. Is the (arrow) on the signal browser new behavior? I seem to remember one could just click on the name. Muscle memory had me clicking on the generated hiearchy and it took me an embarrasing amount of time to figture out that I had to click on the array(>) on the left to expand.:-)

[image]

rfuest, 2025-01-09:
> 1. readme depedency was missing, needed to install following first:
>        sudo apt install libgtk-4-dev

The docs for building GTKWave on macOS were recently updated to include the new GTK4 dependency by a user, but I forgot to also update the Linux docs.

> * Is the (arrow) on the signal browser new behavior? I seem to remember one could just click on the name. Muscle memory had me clicking on the generated hiearchy and it took me an embarrasing amount of time to figture out that I had to click on the array(>) on the left to expand.:-)

As far as I know this isn't new behavior and all GTK2 and GTK3 versions of GTKWave rely on the default behavior of the toolkit, which uses the arrow to expand the nodes in the hierarchy tree. Maybe a GTK1 version of GTKWave behaved differently, which you might have used back in 2008.

aolofsson, 2025-01-09:
Thanks for the quick reply! All good.

closed 2025-01-11 by aolofsson