gtkwave/gtkwave #408: Segfault: gw_tree_builder_push_scope: assertion 'name != NULL && name[0] != '\0'' failed

shareefj, 2025-01-22:
Hi there,

I'm trying to build for Rocky 9.5 and have compiled the latest head on main.  Trying to open an FST file results in a segfaul.

```
$ gtkwave dump.fst

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

FSTLOAD | Processing 4403 facs.

** (gtkwave:34143): CRITICAL **: 13:32:47.554: gw_tree_builder_push_scope: assertion 'name != NULL && name[0] != '\0'' failed
Segmentation fault (core dumped)
```

There don't seem to be any recent tags so what is the latest stable release known to work?

rfuest, 2025-01-22:
Are you able to share the FST file that is causing that error?

> There don't seem to be any recent tags so what is the latest stable release known to work?

The [`lts` branch](https://github.com/gtkwave/gtkwave/tree/lts) contains the stable branch, but apparently the tags haven't been updated. You can also compile GTKWave from the latest source archive on SourceForge: https://sourceforge.net/projects/gtkwave/files/

TristanRobitaille, 2025-03-26:
Hi @rfuest, I have the same issue and can share `.fst`: https://drive.google.com/file/d/1Uz2QaC0dp2bzcUc46mkTj7wk0yt80ol6/view?usp=sharing.
I'm also running v3.4.0.

rfuest, 2025-03-27:
Thank you for the test file, I was able to reproduce the issue.

The problem is that the FST file contains a scope with an empty name at the root of the hierarchy. I'm a bit surprised that the FST library allows such scopes to be created. Which tool did you use to create this FST file?

While I would consider this an issue with the generating program, GTKWave 4 should handle this issue more gracefully. I don't like the idea of unnamed scopes in the hierarchy and would like to not support them unless I have to. GTKWave LTS does open this FST file and shows an unnamed scope in the SST, but doesn't consider this scope when variable name as generated (e.g. `divider_tb.clk` and not `.divider_tb.clk`). This means that GTKWave 4 should be able to just ignore the unnamed scope and still remain compatible with GTKWave LTS save files. Converting the FST file to VCD via `fst2vcd` also handles the FST file this way.

tbybell, 2025-03-27:
If the program interfacing to the FST writer is trying to have a top level that is both signals and modules without an overarching hierarchy encompassing them all, emission of that "" module as well as its paired upscope should be skipped during writing.  The save files versus any sim runs you currently have should be compatible.  I don't know if GTKWave 4 will handle the resulting FST file properly, but 3.x will.

I've seen such a thing where everything is not under a single module coming out of various simulators.

rfuest, 2025-03-27:
> If the program interfacing to the FST writer is trying to have a top level that is both signals and modules without an overarching hierarchy encompassing them all, emission of that "" module as well [...]

That's how it should work and GTKWave 4 handles a mix of signals and modules in the root scope the same as GTKWave LTS.

But I don't like the way GTKWave LTS handles these noname scopes in FST files. If GTKWave doesn't reject this file as invalid the signal names should at least be distinct. This FST file demonstrates this issue: [issue_408.zip](https://github.com/user-attachments/files/19489738/issue_408.zip). Adding the `"".s1` and `s1` signals works at first, but if you save a `.gtkw` and reload it the same signal will be used. There are also other issues caused by this, like `fst2vcd` creating an invalid VCD file (`$scope module  $end`).

My suggestion would be to rename empty scopes to something like `unnamed_scope` during import.

rfuest, 2025-03-28:
I've updated the development version of GTKWave to use the latest version of `libfst`, which includes [the changes Tony made](https://github.com/gtkwave/libfst/commit/56810cebae0cbada92f57baf0a9caa5d8b9a3afc) to address the issue with unnamed scopes.