gtkwave/gtkwave #410: Feature request: Allow changing byte endianess for display

Fighter19, 2025-02-07:
When inspecting values from a little endian machine,
additional workload is placed on the user,
to properly change the order to big endian,
during dump creation.

This is especially cumbersome, when trying to create dumps
containing both strings and integers.

In the integer case, the byte order needs to be flipped,
(from little endian to big endian)
to be properly visualized,
while in the string case, byte order needs to remain the same.

Most of the logic should already be applicable from the "Reverse Bit" function,
so implementing this, should be a low hanging fruit.

tbybell, 2025-02-07:
Please explain exactly where in your usage case that you are encountering a performance problem.  Your description doesn't tell me exactly what you see as a problem.  Thanks.

Fighter19, 2025-02-08:
I'm sorry, I don't understand.
I did not encounter any performance problem.
(Assuming we don't talk about productivity as performance).

In this particular case, let's assume I have a register,
that can contain an 8 character long string OR a 64-bit long integer.
I create the VCD dump by dumping the content of the content of the register,
by going through it's memory range byte by byte.
As the platform I'm dumping the register from is little endian,
it requires me to invert the bit and byte order, before writing it to the VCD.

However the byte order needs to be changed, again for strings.
Let's assume, I write "AB\0\0\0\0\0\0" into a register.
This would put it as 0x4142000000000000 into the register (regardless of architecture, in the same notation as GTKWave uses).
This would mean 0x4241 in Little Endian, so 16961, while in Big Endian, it would be an insanely large number.

If I want to read it as a string, then (currently), I'd need to write it like that into the VCD.
If I want to read it as an integer however, there's currently no way in GTKWave to change the endianess of the byte order.
(Only the bit order).
This is why I need to manually change endianess of the dump to match that of GTKWave in case of integers.

Which makes it hard to read the values while browsing.
Because in the case of integers, I need to convert them to big endian for proper display withing GtkWave,
but for strings, the byte order would have to be kept, as addressed through memory.

Fighter19, 2025-02-08:
Basically, while trying to read a string from the register of a little endian machine,
the string will always be stored in reverse in the register,
when storing integers in a way that can be properly interpreted by GTKWave.

I can only give more complicated answers from here on, I believe.

tbybell, 2025-02-08:
You're dumping it incorrectly.  You need to use either fstWriterEmitValueChange and provide 01xz data that is of the proper number characters where one character = one verilog 01xz MVL4 (multivalue 4-state) "bit" representation, or use a convenience function such as fstWriterEmitValueChange32 or fstWriterEmitValueChange64 to send integers into the writer API such that the writer will reformat them for you to the character string that it uses.

A 64 bit register will have ASCII decomposed into 0/1 bits.  Try it in your simulator to see what kind of VCD is generated.  It will be a 1s and 0s expanded version of ASCII.

On the reader side, you're going to get expanded bitstrings of 1s and 0s out that you need to convert back to a number.  It's done this way because this is to conform with how most simulators produce data.  strtoul() or your own routine should be able to process the returned string though you might have to NULL terminate the string returned by the reader callback function if you're using a library function such as strtoul().

You should look at vcd2fst.c see to see how the writer API is intended to be used.  There is a large for(;;) loop starting around line 1336 where the individual case statements reflect different data being written.  The 's' is for strings and 'b' for binary or 4-state data.

tbybell, 2025-02-08:
BTW, note that the 1s/0s returned certainly are a big-endian looking representation but they are intended to be used as an XDR (external data representation), not to be used directly.  Simulators historically have emitted multi-state data in this left->right direction for historical reasons, mostly due to how VCD files are written. Yes, it complicates reading of integer values, but they can be reconstructed fairly easily by code that functions similar to a shift register.

Fighter19, 2025-02-08:
This is how a register would look like with the value 42 stored in a 64-bit register,
then having the string "WORLD" written to it on a little endian machine.

While visualizing with GtkWave, it reads "DLROW" instead of "WORLD".

[little_endian.vcd](https://github.com/user-attachments/files/18716605/little_endian.txt)

tbybell, 2025-02-08:
No, it doesn't quite work that way.  Verilog shifts strings in from the right hand side.  It doesn't treat them like numbers.  Please see here: https://www.dilloneng.com/strings-in-verilog.html
I've provided an example.  You can cut and paste that VCD into a file and bring it up in gtkwave, or just compile/run the .v file.

```
615:/tmp/p> more t.v
module t;

initial
begin
integer i;
reg [63:0] str;

#10
str = "Hello";
i = 10;

$dumpfile("z.vcd");
$dumpvars(0, t);

#10
i = 42;

end

endmodule

616:/tmp/p> iverilog t.v -g2005-sv
617:/tmp/p> ./a.out 
VCD info: dumpfile z.vcd opened for output.
618:/tmp/p> cat z.vcd 
$date
	Fri Feb  7 22:50:49 2025
$end
$version
	Icarus Verilog
$end
$timescale
	1s
$end
$scope module t $end
$scope begin $unm_blk_1 $end
$var reg 64 ! str [63:0] $end
$var integer 32 " i [31:0] $end
$upscope $end
$upscope $end
$enddefinitions $end
#10
$dumpvars
b1010 "
b100100001100101011011000110110001101111 !
$end
#20
b101010 "
```

rfuest, 2025-02-08:
If you cannot use a real string in the VCD file and need to change the displayed endianness you can manually rearrange the bits in a vector. This is a bit annoying to set up, because you need to manually shuffle signals around, but it might work if you don't need to do this for many signals:

[image]

[little_endian.gtkw.txt](https://github.com/user-attachments/files/18719233/little_endian.gtkw.txt)

Fighter19, 2025-02-08:
Thanks a lot! I see how I can apply that knowledge to my VCD file.
For now, I've only used SystemC and GHDL for generation,
and dumping the registers using the functions provided by the trace tools there, lead to this.
I wasn't aware however, that other object types were possible.

The way @rfuest demonstrated, it's possible to generate a file in the way I need it.

Fighter19, 2025-02-08:
@tbybell Your example does not show how the string would look like
as if it was loaded into a register on a little-endian machine.
This is how it would be represented in a big-endian machine.

Verilog uses a big-endian representation, together with padding 0 in front of the value.

If a little-endian CPU would load "Hello" from memory into a register, it would look like:
```
str = 'h0000006f6c6c6548;
```
in the register.

While on a big-endian CPU it would look like:
```
str = 'h48656c6c6f000000;
```

NOT like it's shown in the example (and how Verilog formats strings):
```
str = 'h00000048656c6c6f;
```

I believe that you are more knowledge, than me in how dumps are structured and their history etc.
And to be fair, I don't understand half of what you're saying.
But I don't believe I'm saying anything wrong here.

closed 2025-02-08 by Fighter19

tbybell, 2025-02-08:
Byte ordering in memory is indeterminate.  In some cases, for example, Icarus, a simulator might store one bit per byte, but that Verilog "bit" is augmented with state such as 01xz and drive strength.  Those bits can exist anywhere in memory and don't need to be contiguous, just that when it comes time to set/get values on them or visualize them that they can be picked out of memory and reconstructed in order.  In other cases such as VCS, values are stored as integers where there are two parallel integers or bitplanes of data: one for 0/1 and the other one that pushes that 0/1 into the x/z range.  That's a simulator implementation detail and unless someone is writing VPI or DPI or whatever code, it's invisible to most users.

As far as how Verilog stores strings, it's big endian right aligned toward the LSB despite being read/rendered left to right from MSB to LSB.  If you defined a vector as str[0:63] instead of str[63:0], it would still be zero padded for short strings on the MSB side.  If a simulator were storing values like VCS does as integers, in memory it would be little endian for each chunk of the actual container of bits/bytes, yet also byte reversed aligned with the least significant byte.  Does it make sense?  Maybe not, but it's the convention Verilog uses for strings.

Part of the reason gtkwave doesn't have a byte reverse function is that it's generally not necessary and is best handled using an external filter...I say that partially because I didn't want to have to deal with a non-multiple of 8 number of bits.  I work on x86 designs for a living and I've never had to byte reverse anything because for convenience the data are already stored byte reversed in the cache SRAMs so that when they're brought up in waves they look correct.  :-)