gtkwave/gtkwave #411: Unexpected behavior with --saveonexit

zapta, 2025-02-08:
When I type ``gtkwave --help`` I am getting this flag help:

```
 -7, --saveonexit           prompt user to write save file at exit
```

However, when I try to use it, gtkwave doesn't prompts the user and saves the changes to the .gtkw file automatically. Is there a way to let the user decide if they want to save the changes to the .gtkw file?

```
gtkwave --saveonexit --rcvar "splash_disable on" --rcvar "do_initial_zoom_fit 1" _build/ledon_tb.vcd ledon_tb.gtkw 
```

tbybell, 2025-02-08:
To https://github.com/gtkwave/gtkwave.git
   7ab46d5..36b88ba  lts -> lts

I changed the behavior in LTS.  Please see if it works for you.  If this in master, it needs to be updated there too.  I don't have a meson env set up at the moment as I've been running an old centos box for compatibility with some old tools.

zapta, 2025-02-08:
Thanks @tbybell. Does gtkwave have nightly builds that I can try?

tbybell, 2025-02-08:
I don't know.  I just use github as revision control.  Ralf and others have been the ones who have set up all the extra stuff such as a QA build process and such since I turned gtkwave over to a gtkwave org account.  Which version are you using and on what distribution or OS?  

Note that I'm close to releasing 3.3.122 as it's been quite a while since the last release.  If the save on exit with what is in LTS works for you, I'll do the release, but as for you, that's a bit chicken and egg.

zapta, 2025-02-08:
I am using ``GTKWave Analyzer v3.4.0 (w)1999-2022 BSI`` on Mac OSX (apple silicon). It's downstream for the oss-cad-suite release.  I just wanted to run the latest to test and provide you feedback for the fix.

zapta, 2025-02-08:
.. I am not familiar with the organizational structure of gtkwave. Will your fix here will eventually included in the official releases?

tbybell, 2025-02-09:
It should be.  That behavior you discovered for this issue is a bug.

3.4.0 should be the new version Ralf is working on.  The changes I made will have to get moved over to that branch.  They can't be directly integrated as the changes in codebases are extensive.  Builds are being run somewhere: I just haven't explored what is going on with the main branch lately.  Ralf or someone else should be able to comment.

zapta, 2025-02-10:
Thanks for the quick fix. I will look for it to be propagated to our project (https://github.com/FPGAwars/apio) via oss-cad-suite release.

zapta, 2025-05-04:
Hi @rfuest, @tbybell mentioned earlier your name in the context of upstream integration.  Anything else need to happen for this  change to propagate to the release?

zapta, 2025-05-05:
Thanks @rfuest, oss-cad-suite indeed uses v3.4.0 but it's good if fixes do move in the pipeline toward the end users.