gtkwave/gtkwave #412: launch script to access menu commands

nipscernlab, 2025-02-12:
Hello,

It seems there is no Tcl support anymore.
Is there any way to access the gtkwave commands by script?

rfuest, 2025-02-12:
Tcl support in the development version of GTKWave was removed to make refactoring of the code base easier. A replacement is planned, but not really usable for much in it's current state.

If you need scripting I would recommend to stick with the LTS release of GTKWave for now.

nipscernlab, 2025-02-12:
Ok, thank you for the explanation.
I just liked very much the new graphic engine, specially the possibility of using dark theme.
Is there any way to use tcl scripts with this new fency IDE?

rfuest, 2025-02-12:
You can use dark mode with the GTK 3 version of GTKWave LTS 3.3. Try to add the `--dark` argument to the command line if dark mode isn't activated automatically.

nipscernlab, 2025-02-12:
That s right, thankyou!
I was using the version comming together with the icarus verilog.
No idea that you have this version with gtk3.

By the way, just for my understanding: are you planning to change the tcl to other language, or you really want to remove any kind of automation?
I hoppe you keep automations by script and sockets for comunication with external software. This is very usefull for simulation environments, in my opinion.

Thank you again

rfuest, 2025-02-12:
> By the way, just for my understanding: are you planning to change the tcl to other language, or you really want to remove any kind of automation?

No, there is no plan to remove scripting entirely. I decided to remove Tcl scripting, because it was hard to keep it working during the refactoring. GTKWave 4 will have a more modular design and will initially support Python scripting. Some parts of this are already working, but there are still a lot of important parts missing.

If there is enough demand for Tcl scripting as a second scripting language this can be added back in later, but with a different API. It won't be possible to keep Tcl scripts compatible between GTKWave 3 and 4, because at the moment Tcl scripts need to access menu items by name and any change to the menu structure would break user scripts.

tbybell, 2025-02-12:
Some of the existing Tcl scripting in GTKWave 3  is a bit hackish and came out through evolution from user code submissions, not planning.  A problem is that Tcl and gtk have their own, independent event loops (which weren't designed to play well with each other--why aren't you using Tk?) so arbitrary decisions that got it to work are just that: decisions that got it to work.

I would think a thin translation layer between a scripting language and a gtk-friendly RPC mechanism might be the cleanest modern way to interface a scripting language to the viewer, but there are plenty of ways it can be done.  I think Ralf's strategy of pulling it out to add something cleaner back in later is the way to go.

Moreover, I don't know if modern EDA tools have moved beyond Tcl or if they use something else such as Python or Ruby.  At work, I see a lot of Python or Ruby coming from the new hires with respect to driving tools.  I see zero Tcl except for legacy scripts or stuff I bang out myself out of familiarity.

nipscernlab, 2025-02-12:
Understood. I fully agree.
For me, you can close the ticket.
Looking forward to use version 4.
thank you once more

closed 2025-02-12 by tbybell