gtkwave/gtkwave #413: Ubuntu24.0 doesn't support glibc2.27 causing symbol lookup error

ABSplus, 2025-02-14:
Is it possible to waiting for a new version of GTKWAVE pack or compile locally to handle this?
I try to start gtkwave from terminal, while got this:
```shell
gtkwave: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE
```
after that I search for some solution, and got the most of answer to the reason is about the version of glibc too high in system. They said I need to use some patch trick to install an older version but it seems so sophisticated that might causing more system-unstable problem.

ABSplus, 2025-02-14:
to further information, gtkwave had run normally in UB20.X, error came after I update my system from 20 to 24

rfuest, 2025-02-14:
Which version of GTKWave are you trying to run? The official `gtkwave` package from the Ubuntu repository or something else?

rfuest, 2025-02-14:
Also, are you trying to run GTKWave from a regular terminal or something installed via Snap? The `/snap` prefix in the error message suggests that Snap is somehow involved, but GTKWave isn't available as a Snap.

ABSplus, 2025-02-14:
The system I using is orginal Ub so I think the terminal and the package are originally coming from offcial repo., but it really reminds me...
started from UB24, system will ask and automatically change package manager from apt to snap, so probably when I install any package in terminal like ' sudo apt install gtkwave '  it will install a binary pack from snap... I never met the similar problem before
I will check gtkw version later hours, really thanks for your quick response!

ABSplus, 2025-02-15:
The problem solved after I restarted the virtual machine without any update, while no idea what make this error. 
All I known is that I was installing some driver update in real machine and installing verilator in virtual one at the same time, and unstable USB connection from external driver where I store my VM might also cause such problem.
btw **The version of GTKWAVE is 3.3.116**.
 sorry for wasting your precious time

closed 2025-02-15 by ABSplus