gtkwave/gtkwave #414: How not to show SST?

nipscernlab, 2025-02-16:
Hello,

Is there any tcl command or initialization function which allow us to hide the SST at the GTKWave lunching?
I am using tcl scripts to populate the wave window, so I want to hide the SST to have more space for the waves.

tbybell, 2025-02-16:
gtkwave --rcvar 'hide_sst on' [rest of arguments]

nipscernlab, 2025-02-17:
This is the error output trying to follow your suggestion:

C:\Sapho\sapho-p>gtkwave --rcvar 'hide_sst on'

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

RCVAR   | ''hide_sst' NOT FOUND
Error opening  .vcd file 'on''.
Why: No such file or directory

tbybell, 2025-02-17:
You're using Windows.  See if double quotes instead of single ones works in that command line shell.

rfuest, 2025-02-17:
I might have removed that option from 3.4 at some point, which could explain your error.

tbybell, 2025-02-17:
```
533:/home/bybell/gtkwave/code/gtkwave3-gtk3/examples> hex output
00000000: 52 43 56 41 52 20 7c 20 27 27 68 69 64 65 5f 73 RCVAR | ''hide_s
00000010: 73 74 27 20 4e 4f 54 20 46 4f 55 4e 44 0a 45 72 st' NOT FOUND.Er
00000020: 72 6f 72 20 6f 70 65 6e 69 6e 67 20 2e 76 63 64 ror opening .vcd
00000030: 20 66 69 6c 65 20 27 6f 6e 27 27 2e 0a -- -- --  file 'on''..
```

Ralf, the framing of the extra single quote for the rcvar vs the supposed dumpname filename makes me wonder about handling from the command line.

rfuest, 2025-02-17:
You are right, that there is something wrong with the quoting. But I just checked and I did remove the rcvar (https://github.com/gtkwave/gtkwave/commit/2a34bf50b03450f3d09be46983e20650e2d8e5d5) and it wouldn't work even with correct quoting. I can't remember why I did remove the option.

nipscernlab, 2025-02-17:
double quotes works for me.
thanks!

tbybell, 2025-02-17:
If you want to use it later, I believe it should pop up in an additional window when activated from the menu as Search->Signal Search Tree.

nipscernlab, 2025-02-17:
Right, I found it in the menu.
Thanks!

By the way, what else I can control trhough this --rcvar comand?
Is there in the documentation?

rfuest, 2025-02-17:
> By the way, what else I can control trhough this --rcvar comand?
> Is there in the documentation?

Everything you can set in the `gtkwaverc` file. See the `gtkwaverc` man page or the online version at https://gtkwave.github.io/gtkwave/man/gtkwaverc.5.html for a list of settings. Note that the online version is for GTKWave 3.3 and some settings are removed in development builds of GTKWave.