gtkwave/gtkwave #415: BitsToReal is ignored for analog vptr traces

rfuest, 2025-02-17:
I'm currently refactoring the code in `baseconvert.c` and noticed a minor issue in GTKWave 3.3. `convert_real` doesn't handle the `TR_REAL` flag, which causes the analog display for vptr traces to be incorrect:

[image]

VCD and GTKW files to reproduce the issue: https://gist.github.com/rfuest/9c4a1b064acc9e011b25dd0666d1b416

tbybell, 2025-02-17:
It's not that it's ignored...the 64b double vs 32b float case is missing.

At the bottom of convert_real

```
        if(!fail)
                {
                retval = val;
                }
```

needs to be something like

```
        if(!fail)
                {
                if(nbits == 64)
                        {
                        retval = val;
                        }
                else
                        {
                        float f;
                        uint32_t val_32 = val;
                        memcpy(&f, &val_32, sizeof(float)); /* otherwise strict-aliasing rules problem if retval = *(double *)&val; */
                        retval = (double)f;
                        }
                }
```

If that looks like the case to you, I'll change it in the LTS version.  I have to test it some more, but that looks like it.

rfuest, 2025-02-17:
I don't think that's correct, because at the end of `convert_real` the bits should be treated as an unsigned integer and not a floating point number. Take a look at the corresponding code from `convert_real_vec`:
https://github.com/gtkwave/gtkwave/blob/36b88bac313173b516ffe4174f4ddef9b8cfa5cd/gtkwave3-gtk3/src/baseconvert.c#L1984-L2018
This case is only active when the  `TR_REAL` flag is set.