gtkwave/gtkwave #418: Support opening multiple files

gezalore, 2025-03-04:
I would like to propose an enhancement where GTKWave can open multiple dump files (produced from the same simulation), and display them in a single view. That is, present it as if you opened a single file that contains the union of the dumps from the multiple files.

This is related to, but different from #286, which is more of an enhancement to FST. This request is more generic.

The 2 use cases for this are:

1. Being able to display the "split" dump files generated by Verilator, which is already implemented in this Verilator PR: verilator/verilator#5806, but requires viewer support. This has notable simulation speed benefit (please see the linked Verilator PR). #286 intends to cover use case with slightly different specifics.

2. Displaying dumps generated from a more generic cosimulation environment.

---

For case 1. the files contain the same time intervals and hierarchy, but each file contains a subset of the signals. Here is an example using VCD header syntax:

Assume this is the original hierarchy with all signals, which is what Verilator would produce today:
```
$scope module top $end
  $var wire 1 @1 sig_a $end
  $var wire 1 @2 sig_b $end
  $scope module sub $end
    $var wire 1 @3 sig_x $end
    $var wire 1 @4 sig_y $end
  $upscope $end
$upscope $end
```

Then when dumping on 2 threads, you will get 2 files, that might look like this, which is what Verilator will produce with verilator/verilator#5806::

```
$scope module top $end
  $var wire 1 @1 sig_a $end
  $scope module sub $end
    $var wire 1 @2 sig_y $end
  $upscope $end
$upscope $end
```

```
$scope module top $end
  $var wire 1 @1 sig_b $end
  $scope module sub $end
    $var wire 1 @2 sig_x $end
  $upscope $end
$upscope $end
```

The hierarchy of scopes will match by name and structure, and each file contains a disjoint subset of the signals that together combine to the original set of signals.

---

For case 2. the hierarchies might not completely overlap, and the time intervals in the files might be different. Even the file formats could be different (e.g.: VCD from System-C + split FSTs from Verilator).

---

From my perspective, I don't mind if you prefer to do #286 instead, which supports the Verilator specific use case 1. However you can try to support the more generic request in this issue which covers both 1 and 2.

Thank you for your time considering this.

rfuest, 2025-03-04:
I've planned to support this in GTKWave 4 and did some initial testing of how this could be added. My current idea is to add a new file type which contains all necessary metadata to merge multiple dump files into one view.

The file could look something like this:
```
{
  "files:": [
     { 
       "filename": "1.fst",
       "root": "file1",
     },
     {
       "filename": "2.fst",
       "root": "file2",
     },
  ],
}
```

The `root` parameter would make it possible to set the position in the hierarchy at which point the subtree in that file gets "mounted" in the global hierarchy. This would make it possible to either keep the hierarchies separate (by setting different `root`s, case 2) or to merge the hierarchies of different files (by setting the same `root`, case 1).

gezalore, 2025-03-04:
Thank you, that would certainly work and enable even more general use cases. The small overhead of emitting the extra metadata file is trivial for case 1. For case 2, worst case the user has to write one by hand, thought their cosim launch scripts could easily do that too.

For case 1, I think we need to support `root` begin set to ` ` (blank), to make sure no additional prefix is added above the root scope in the files.

I'm just curious, do you have a timeline for GTKWave 4? Not going to hold you to it of course.

gezalore, 2025-03-04:
There is also the use case of 'case 1 under case 2', so if you go with the metafile approach please consider some of the listed files being other metafiles themselves.

rfuest, 2025-03-06:
> Thank you, that would certainly work and enable even more general use cases. The small overhead of emitting the extra metadata file is trivial for case 1. For case 2, worst case the user has to write one by hand, thought their cosim launch scripts could easily do that too.

At some point in the future the user should be able to setup case 2 inside the UI without having to manually edit the metadata file. But as a proof of concept manually editing the file should be OK.

> For case 1, I think we need to support `root` begin set to ` ` (blank), to make sure no additional prefix is added above the root scope in the files.

That's right. My plan is to make `root` optional instead of using a blank string. This and some other details still need to be considered for the final version of the file format. The new save file format for GTKWave 4 will also be JSON based and I want to make sure all new file types use a similar structure.

> I'm just curious, do you have a timeline for GTKWave 4? Not going to hold you to it of course.

No, not really, but don't expect it any time soon. But a preview of this feature might land in GTKWave before the GTK 4 port. Support for loading multiple dump files at once has been requested multiple times and I'm getting curious to finally see some tangible results from all my refactoring effort. Most of the refactoring of the dump file loaders is done and I don't think any big UI changes are necessary, because the loader for the metafiles will merge the hierarchies before passing them on to the UI. But this will only be a basic version and not do anything fancy like multi threaded loading of the individual dump files.

> There is also the use case of 'case 1 under case 2', so if you go with the metafile approach please consider some of the listed files being other metafiles themselves.

That's a good point I hadn't thought about yet. But I think that should work out of the box, because the metafile loader provides the same interface as the loaders for the other dump file formats. This means that for the parent metafile loader and child metafile should look the same as any other file format. I'll try to remember to add an integration test for this case.

gezalore, 2025-03-13:
I'm wondering if you would be happy if we agreed on a spec for the metafile format now, even if you don't implement loading it in GTKWave until later. I think it would be beneficial to the community if this was reasonably standard, so e.g. Verilator can produce it directly, and it can be loaded in any viewer like GTKWave, Surfer, or whatever tool that might want to consume waveform dumps. If we have an agreed upon format, then I can also land the Verilator patch that produces these, so you can create test dumps easily when you need to. We can always stick a version number in there for extensibility. What do you think?

rfuest, 2025-03-13:
When I had the idea for this format I had only considered it to be a GTKWave specific file format and not something that would be used more broadly. But I would be OK with making it a more universally usable format.

Would you prefer to use JSON for the format or something else? For a simple file list some might consider JSON to be overkill. While I'm not a huge JSON fan myself it has the advantage that there are lots of libraries and other tooling to handle JSON files. This was the reason I decided to go with JSON for the new save file format in GTKWave 4.

## JSON format draft

Here is a first draft of a JSON schema for the file format, to serve as a base for further discussions:

```json
{
  "type": "object",
  "properties": {
    "version": {
      "type": "integer",
      "minimum": 1
    },
    "kind": {
      "type": "string",
      "enum": [
        "merged",
        "separate"
      ]
    },
    "files": {
      "type": "array",
      "items": {
        "$ref": "#/$defs/file"
      }
    }
  },
  "required": [
    "version",
    "kind",
    "files"
  ],
  "$defs": {
    "file": {
      "type": "object",
      "properties": {
        "path": {
          "type": "string"
        },
        "alias": {
          "type": "string"
        }
      },
      "required": [
        "path"
      ]
    }
  }
}
```

When `kind` is set to `merged` all files listed in the `files` array are loaded in the same root scope and the hierarchies are merged. When `kind` is set to `separate` the files are loaded as siblings and not merged. By default the roots of the subtrees for each file are named after the filename, but this can be overridden by setting the `alias` property.

The `path`s to the files would be relative to the metadata file by default. We would need to decide how to handle the different path separators in Windows and Unix-like OSes.

## Examples

### Merged

```json
{
    "version": 1,
    "kind": "merged",
    "files": [
        { "path": "file1.vcd" },
        { "path": "file2.vcd" }
    ]
}
```

Generated hierarchy:
```
top
    module_from_file1
    module_from_file2
```

### Separate

```json
{
    "version": 1,
    "kind": "separate",
    "files": [
        { "path": "file1.vcd" },
        { "path": "file2.vcd", "alias": "alias" }
    ]
}
```

Generated hierarchy:
```
file1.vcd
    top
        module_from_file1
alias
    top
        module_from_file2
```

## Possible extensions to the format

Some random ideas that might or might not be useful:

- metadata
  e.g. the program that generated the file
- optional file hashes 
  could be used to verify that a set of `merged` files is belonging to the same simulation run, but is expensive for large files

tbybell, 2025-03-13:
This is an example of what nWave uses as far as information in the file.  Not a whole lot of rocket science appears to be applicable to it.  I don't know if they have FileType other than stitch.  I would suppose so, because why specify it?

```
627:/run/media/bybell/PATRIOT> more ./memory_stick/school/utils/hdl/ver/fsdb/verilog.vf
@FSDB rc file Version 1.0
[VRTL_FILE_HEADER]
# !! DON'T EDIT [VRTL_FILE_HEADER] SESSION !!
Version = 1
[VRTL_FILE_SOURCE]
FileType = stitch
File1 = verilog_0000.fsdb
File2 = verilog_0001.fsdb
```

Anything works.  What Ralf proposes with that schema of his looks fine.  In short, it's just a "how to get there from here" indexing scheme.

Note that gtkwave can handle inadvertently duplicated definitions and it will hide the duplicates.  I've seen cases often enough during my day job where someone called the fsdb dumpvars function multiple times, so I made the viewer able to hide duplicates.  It's not optimal as memory is wasted internally if duplicate signals are emitted, but it allows viewing of sloppily dumped hierarchies.  That said, you can navigate into the same hierarchy multiple times, specify $var declarations for a level of hierarchy, and the signals for that level of hierarchy will still find themselves merged into their proper locations.

rfuest, 2025-03-14:
> Do we actually need `kind` 'merged' vs 'separate'?
> ...
> Note: when loading, you also need to be aware that one file might contain the actual hierarchy you are merging into, so if you want to keep the proper scope types, you probably need to load them with shortest `root` first (or combine the scope types), e.g. if 'file1.vcd' here contains root scope 'bar', then the contents of 'file2.vcd' need to be merged into that:

This was my reason for suggesting the simpler solution with two kinds of merge. The `separate` kind would add top level nodes of a new "Dump file" scope type and the `merged` kind would just check that the scope types in both files match and give a warning if they don't.

I'm not against adding arbitrary "mount points", but it might make the burden of implementing the format in other tools a bit higher and we need to make sure we properly document the expected behavior.

For reference, the VRTL format Tony mentioned does also support this feature, but I couldn't find any documentation whether they only allow existing prescopes. In the example I found the second and third file are added in a part of the hierarchy that has been created by the first file.
```
@FSDB rc file Version 1.0
[VRTL_FILE_HEADER]
# !! DON'T EDIT [VRTL_FILE_HEADER] SESSION !!
Version = 1
[VRTL_FILE_SOURCE]
FileType = split
File1 = ./1.fsdb
File2 = ./2.fsdb
File3 = ./3.fsdb
[VRTL_FILE_ADD_PRESCOPE]
ScopeSeparator = "/"
File2 = /top
File3 = /top/sub_sys1
```

> With `root`, (note we should accept a dotted name, so need to escape the file name dot):

My plan for the GTKWave save format is to leverage the flexibility of the JSON format to not have to deal with different hierarchy separators and escaping of separators. All references to hierarchy location can take two different value types, a string or an array of string:
```json5
{
   // In a regular string all dots are separators.
   // This is primarily added to make editing a file by hand easier and covers the vast majority of paths.
   "path": "mod.submod",

   // Instead of relying on a separator an array of strings can be uses. This is equivalent to the row above:
   "path": ["mod", "submod"],

   // This representation can easily handle all kinds of weird names and no special escaping is necessary.
   // (except for quotes, which is abstracted away by the JSON library)
   "path": ["mod", "spaces and .s are allowed"],

   // One minor issue with this would be that it isn't obvious for new users that these aren't the same:
   "path": "mod.submod",     // two hierarchy levels ("mod" -> "submod")
   "path": [ "mod.submod" ], // on hierarchy level ("mod.submod")
}
```

> For the consistency check, an alternative to hashes is to include a unique number, that you pick at random at simulation startup, in both the metafile, and child files (e.g in a VCD/FST comment entry). Like hashes this can catch consistency errors, but not corruption in transit, or swapping around files form the same simulation.

I had thought about unique id's, but had assumed that it would need a change in the child dump file to insert the id. The idea of using a comment for this purpose would work.

Swapping around of files from the same simulation could still be detected, because each child file could have it's own random id.

> What would you like to call this? Even if internally it's just JSON, a unique file extension would probably make it easier to handle e.g. in file explorers and "open file" dialogs.

Naming things is hard and I'm open to good suggestions. When I had planned to implement something like this as a GTKWave specific format I had thought about `gw_set` or `gw_fileset`, but that wouldn't work for a more generic format. Including `set` or `virtual` (like the format Tony mentioned) as part of the might be a good idea.

tbybell, 2025-03-14:
No matter what we do with this, users are going to abuse it in some capacity.  I don't know if we really need to generate checksums.  If we do, they should be optional because I could imagine some users creating an interesting usage case of a Frankenstein's monster of child files from unrelated sim jobs that none of us ever could imagine.

When I get stuff out of the emulator and it generates dozens of child files, I only care about what I'm specifically debugging.  If I'm looking to see why CPU core 7 is hung, I don't care about 0-6 and I really don't want to wait for it to get transferred across NFS or whatever on the datacenter machines if I'm not going to need it.

I'm not a fan of the whole flattened hierarchy delimeter thing (a general complaint, not about what Ralf discusses).  Flattened signal names have their place in industry due to historical reasons, but the existence of escaped identifiers makes them a mess to handle reliably.  I do like Ralf's bracketing of identifiers in comma separated quotes, mainly because it's not ambiguous and it's not too much of an added tax on top of what already exists and it can still be represented using a single line if a user wants to see it that way.

```
   // One minor issue with this would be that it isn't obvious for new users that these aren't the same:
   "path": "mod.submod",     // two hierarchy levels ("mod" -> "submod")
   "path": [ "mod.submod" ], // on hierarchy level ("mod.submod")
```

Users will get used to it.  The only thing it really needs is an escape mechanism just to ensure that the quote character can be in a name.  It's not something I'd ever generate, but the option exists for it for whatever tools.

gezalore, 2025-03-18:
I agree with Tony that people will find a way to break this no matter what. Also, If you allow child files to be metafiles, which we want to, then I think you can already achieve the "arbitrary mount points" by mixing `separate` to add prefixes, then `merge` to combine them, it's just harder to do.

I don't think we need to go into great length about "specifying" the behaviour when merging scope types. The scopes are just a partitioning (like namespaces), of the static variables in the program (design). Some things care about the "type" of a scope, like a wave viewer, some might not. I would think an implementation note of "be ware of this problem and do what makes sense in your tool" should suffice.

I also agree the checksums/IDs should be optional and a warning only. You will want to be able to take shortcuts like what Tony mentioned.

I'm happy with using lists for specifying hierarchy path elements. I think it would be ok even if that was the only option, and don't allow flattened names as "string with dots". The tools that generate or consume these files shouldn't really care, and I think it's ok to expect a user who wants to hand edit these to understand how a JSON list works. The tool UI/API can provide "friendlier" ways to specify these if necessary.

> Naming things is hard and I'm open to good suggestions.

This finally might be a use case for ChatGPT, here are some ideas it came up with after some prodding:

- Wave Aggregation DEscriptor (.wad / .wade)
- TRace Aggregate DEscriptor (.trade)
- TRace Index File/Format (.trif)
- Multiple Aggregate Signal Traces (.mast)
  
The last one has a nice nod to FST while being a single easy to pronounce syllable, but maybe too close to "AST", also "Signal" is a bit narrower concept than all the stuff that can appear in a trace file. I'm of course not attached to any of these, just for inspiration.

gezalore, 2025-03-19:
> including "do whatever you want" in a file format description might not be the best idea either.

You will notice that is not what I suggested ;) What I was trying to express here is that I don't think we need to specify the behaviour of consuming tools, they can decide for themselves, there is usually only a few choices that "make sense", plus we can't "make them" do what we want if they disagree anyway, it's a community effort after all. I think highlighting the tricky bits should be enough, at least in version 1, and we can trust the tool devs. Having said that, I'm not against writing anything down, just trying to explain my thoughts here.

> A more important issue is how duplicate signals should be handled. I would suggest to make this an error by default and add a flag 

Agreed. You can consider an enum instead of the flag `"duplicate_signals" : "useFirst" | "useLast" | "makeUnique" | "error"`, though I can't really think of a use case where anything but "error" is useful (outside of tool debugging), so probably not worth the complexity of having many options in the beginning, so allow/error should be Ok.

> Let the bike shedding begin ;)

Haha, I was hoping to avoid that.

I'm not a fan of the word "dump" with its dictionary meaning, but I would be happy with anything that is easier to pronounce than [zmrzlina](https://en.wiktionary.org/wiki/zmrzlina). How about `.vist`, based on your last suggestion. Google results show no real conflict.

---

Tangential, but while I was thinking about this, I realized you could actually go much further with this format. For example, you could have a whole hierarchy and signal definition section, that is mapped back onto the child files (that is, scopes and signals are defined in the json, values are stored in the child files). This would allow for resolving the ordering ambiguity when merging split files. (See example in the original post. You don't know how to interleave the signals in the same scope. Some tools display them in source declaration order.) It would also allow for adding "derived signals" (as discussed here https://github.com/gtkwave/libfst/discussions/7#discussioncomment-12388526) to anything, which can have significant performance and size benefits. Definitely not for the first version, just throwing it out there.

rfuest, 2025-03-19:
> You will notice that is not what I suggested ;) What I was trying to express here is that I don't think we need to specify the behaviour of consuming tools, they can decide for themselves, there is [...]

While it might not seem like it we seem to basically agree on this. I don't want to make this any formal specification and just add a short informal description of how the tricky bits could be handled. As someone who had to deal with plenty of incomplete and ambiguous file format descriptions I would appreciate some guidance on any issue that implementers of the formats are likely to encounter. But we can revisit this after the rest of the document is written to see if it is necessary in the end.

> Agreed. You can consider an enum instead of the flag `"duplicate_signals" : "useFirst" | "useLast" | "makeUnique" | "error"`, ...

Is `useFirst` and `useLast` necessary, or could the same achieved by reordering the file list, if one of these would be the default? 

> ... though I can't really think of a use case where anything but "error" is useful (outside of tool debugging), so probably not worth the complexity of having many options in the beginning, so allow/error should be Ok.

`error` was mainly intended for manually created files and not tool generated files. In most cases it should be pretty clear to the user that they messed up somehow if the hierarchy doesn't look like expected, but there might be cases where it isn't.

Do you have a use case where `makeUnique` would be useful? I can't think of any situation where this would be beneficial and it would make merging more complex and harder to get consistent between tools.

> I'm not a fan of the word "dump" with its dictionary meaning, but I would be happy with anything that is easier to pronounce than [zmrzlina](https://en.wiktionary.org/wiki/zmrzlina). ...

"dump" might have that meaning, but everyone who has used Verilog is familiar with the term and I doubt that many HDL designers have an issue with the term.

> ... How about `.vist`, based on your last suggestion. Google results show no real conflict.

Not my favorite, but let's just use that one.

tbybell, 2025-03-19:
> I'm not a fan of the word "dump" with its dictionary meaning, but I would be happy with anything that is easier to pronounce than [zmrzlina](https://en.wiktionary.org/wiki/zmrzlina). How about `.vist`, based on your last suggestion. Google results show no real conflict.

This has really turned into an xkcd comic.  :-)

".vist" is fine, and for all I know, it'll become a noun.  It has the one syllable brevity aspect going for it, so if I have to say to a co-worker, "I need a vist for that," it rolls off the tongue quickly.
 
> Tangential, but while I was thinking about this, I realized you could actually go much further with this format. For example, you could have a whole hierarchy and signal definition section, that is mapped back onto the child files (that is, scopes and signals are defined in the json, values are stored in the child files). This would allow for resolving the ordering ambiguity when merging split files. (See example in the original post. You don't know how to interleave the signals in the same scope. Some tools display them in source declaration order.) It would also allow for adding "derived signals" (as discussed here [gtkwave/libfst#7 (comment)](https://github.com/gtkwave/libfst/discussions/7#discussioncomment-12388526)) to anything, which can have significant performance and size benefits. Definitely not for the first version, just throwing it out there.

Scope creep (pun intended) seems to indicate that we might be edging into KDB territory (see slide 55):
http://media.ee.ntu.edu.tw/personal/pcwu/dclab/dclab_02.pdf

I really have to look at the KDB interface code as well as Verdi docs.  I don't know what we'd use to generate such a thing.  Verilator XML is quite representative of the design and these split FST files could be coming from its compiled code simulator.  The only other tool I can think of that would be suitable for generating something KDB-like would be a custom plugin for vrq.  I don't see that Hummel has updated it in a while though.  

For now, "vist rhymes with list," and basic functionality is all we really need.  I fully expect this is going to be a moving target for a while.

Note that GTKWave itself isn't really hung up too much on signal ordering (except for index direction on bitblasted nets) as everything goes through per-hierarchy alphanumeric sorting in each dumpfile loader, mainly because it would be impossible to find signals in large modules if that didn't occur.

gezalore, 2025-03-19:
I think we are in agreement.

---

> Is useFirst and useLast necessary

Probably not needed, let's not worry about it.

> Do you have a use case where makeUnique would be useful?

I was thinking of tool debugging, so you can see what is actually going, but you can always just mount the files at different roots manually, so we probably don't need this either.

---

> Scope creep (pun intended) seems to indicate that we might be edging into KDB territory 

Wasn't quite thinking going that far, just enhancing traces, but I agree it's out of *scope* for this iterations.

> I don't know what we'd use to generate such a thing.

Just FYI, slang is somewhat popular for this kind of thing: https://github.com/MikePopoloski/slang