gtkwave/gtkwave #419: Add a CLI option to parse a VCD file and exit

YuniTsukiyama, 2025-03-04:
It could be useful to have a CLI option to check if a VCD file is valid and then exit.

As gtkwave currently reports for missing display before parsing options / opening files, I fear this may need some rework to run in CLI only.

What do you think ?

tbybell, 2025-04-06:
If X11/Wayland display variables are set up, you can check for parser pass or failure already.  See the return code of 0 vs 255 when -x/--exit is used:
```
561:/tmp/q> gtkwave -x des.vcd

GTKWave Analyzer v3.3.122 (w)1999-2024 BSI

[0] start time.
[704] end time.
Exiting early because of --exit request.
562:/tmp/q> echo $?
0
563:/tmp/q> gtkwave -x des_broken.vcd

GTKWave Analyzer v3.3.122 (w)1999-2024 BSI

Near byte 58, VCD search table NULL..is this a VCD file?
Near byte 58, Unknown VCD identifier: 'ushsu'
Near byte 68, Unknown VCD identifier: 'sis'
Near byte 73, Unknown VCD identifier: 's'
Near byte 82, Unknown VCD identifier: 'shshu'
$VAR encountered after $ENDDEFINITIONS near byte 161.  VCD is malformed, exiting.
564:/tmp/q> echo $?
255
```
However, yes, given how gtk is initialized before args are checked, you can't currently do this in a headless session.  As gtkwave is a GUI-based tool, I can't really justify moving chunks of the init around at risk of breaking something else.  Note that based on some quick experiments I just ran, it "should" be doable, but I don't want to risk breaking some obscure usage case or sequence of operations that currently works.

If the intent is for something like an automated submission script to ensure undergrads submit completed assignment materials that aren't broken, I'd say just run vcd2fst against the VCD file, then ensure the resulting FST file isn't broken by running fst2vcd with an output file of /dev/null or /dev/zero.  Exit codes 0 vs 255 should perform as expected to signal pass/fail.

rfuest, 2025-04-06:
The development version of GTKWave isn't GUI only anymore. While there is no CLI tool included to check files for validity, it is possible to do this via a simple Python script (or an equivalent C program):
```python
import sys
import gi

gi.require_version('Gw', '1')
from gi.repository import Gw

loader = Gw.VcdLoader.new()

try:
    loader.load(sys.argv[1])
except Exception as e:
    print("Error", e, file=sys.stderr)
```

But the API isn't stable yet and will likely change in the future. Better warnings about how parsing failed are also on the way, but not included in the upstream development branch yet.

tbybell, 2025-04-06:
That's good to know that there's a component/interface approach in work.

Excluding the occasional truncated file causing a short read with unpredictable behavior, the usual source of errors I've encountered that are reported by users are when individuals attempt to make their own VCD writer and mess up something like VCD ID generation or something similar, or they emit stray control characters such as 0x00 into the file due to off by one string length errors, etc.  Better warnings will definitely help with such cases.

BTW, with respect to VCD IDs, I think the most interesting thing with them is the trick VCS uses where the identifiers are a bijective numeration and the number system for IDs lacks a zero symbol.  No zero means no ambiguity on leading zeros in an ID.  The loaders in gtkwave can completely avoid hash tables for identifiers if VCD IDs follow that VCS-like formatting.  https://en.wikipedia.org/wiki/Bijective_numeration