gtkwave/gtkwave #424: Legacy VCD loader is broken in GTKWave LTS

rfuest, 2025-04-08:
The legacy VCD loader is probably not used by anyone, but I noticed that this line refers to the wrong buffer which causes a segfault:
https://github.com/gtkwave/gtkwave/blob/6d759dbd322ce3a26eb9b53533863c59811d658c/gtkwave3-gtk3/src/vcd.c#L343

tbybell, 2025-04-09:
Introduced by me at 3.3.120, so it was fairly recent, at least as far as version numbers go:
`Fix case of missing newline at EOF for VCD loaders.`
Cut and paste directly from the recoder into the legacy loader = dead legacy loader.  It only happened in the -gtk3 version so it was a totally sloppy cut-and-paste.

I'll check in the fix to LTS along with the other fixes that have been stacking up.

tbybell, 2025-04-09:
To https://github.com/gtkwave/gtkwave.git
   6d759db..1fec704  lts -> lts

closed 2025-04-09 by tbybell