gtkwave/gtkwave #429: Distinguish between scalars and vectors of width 1

rfuest, 2025-04-27:
The VCD and FST loaders currently handle a variable with the name `var[0:0]` the same as one with the name `var[0]`. I believe that it would be better to handle these differently to preserve more of the minimal type information these formats provide.

GHDL is a real world example of a tool that uses the `[0:0]` suffix when it dumps something like a `bit_vector(0 downto 0)`. Using `(0 downto 0)` might not be common in hand written code, but it is more likely in code generated by generics.

I'm not sure if this change would break existing savefiles. If this is the case it might be better to limit this change to GTKWave 4, which could handle this difference as part of the conversion of GTKW savefiles to the new format. Or do you think that this change isn't necessary at all?

tbybell, 2025-04-27:
If it's found in the signal declaration in the dumpfile as `[0:0]`, the `[0:0]` probably should be preserved because it's how the signal was declared as a vector (which then later allows 1..n bit part selects on it), and if the viewer is used in conjunction with other tools, there might be potential for something to mess up correlating of information between tools (e.g., lost signals on a drag and drop).

I never preserved the colon notation in any of the existing loaders because any signals such as x[0] were more than likely part of a larger train of consecutive bit declarations from a "bitblasted" vector in it was obvious it was coming from a multi-bit vector and could be coalesced.  My gut feeling is that keeping the `[0:0]` in the loader instead of converting to `[0]` would break existing savefiles because we'd need a part select on an existing vector rather than straight one to one match on signal name.  I can look at it to see if it's possible, but I don't see it as a must have for LTS.  I'll leave it up to your discretion for GTKWave 4.

Note that the can of worms that opens is that when I run Icarus (this is against an old version), I see `[0:0]` for variable declarations in Verilog get removed, but `[1:1]` is present for the $var declaration in the VCD.  That might be tied to the simulator's treatment of scalars vs vectors.

I attempted to go a bit beyond this to abuse a single dimension in a variable declaration.  This is, well, interesting with the -1 that sneaked in there and the wrong width...I need to get the latest to see if it still does this, but you likely discovered a whole class of obscure, related bugs in existing tools.

```
526:/tmp/l> cat t.v
module top;

reg [0] x;

initial
begin
$dumpfile("xxx.vcd");
$dumpvars(0, top);
#0 x = 0;
#1 x = 1;
end

endmodule
527:/tmp/l> iverilog t.v -g2005-sv
528:/tmp/l> ./a.out 
VCD info: dumpfile xxx.vcd opened for output.
529:/tmp/l> cat xxx.vcd 
$date
	Sun Apr 27 13:37:06 2025
$end
$version
	Icarus Verilog
$end
$timescale
	1s
$end
$scope module top $end
$var reg 2 ! x [0:-1] $end
$upscope $end
$enddefinitions $end
#0
$dumpvars
bx !
$end
#1
530:/tmp/l> 
```

rfuest, 2025-04-27:
I've tried you test in a newer version of iverilog and it fails to compile:
```
iverilog t.v -g2005-sv
t.v:3: error: Dimension size must be greater than zero.
t.v:3       : This size expression violates the rule: 'sd0
1 error(s) during elaboration.
```

The reason I'm finding a lot of these edge cases at the moment is that I'm planning a fairly major rewrite of the way nodes, symbols, and the symbol tree work in GTKWave 4. I want to make sure to have a good test coverage before I start, to make it less likely to break anything undetected.

One idea is to have separate node types for scalar and vector waveforms, which will make it easy to distinguish between scalars and vectors of width 1. This is safer than relying on `lsi == msi` or `lsi == -1 && msi == -1` and gets rid of inconsistencies between the loaders (VCD uses `lsi = msi = 0` for scalars, while FST uses `lsi = msi = -1`). In the end nodes should only contain the information necessary to decode the waveform, which means that a vector node would only store the width and not the bounds or a name. The additional information, like the name and the bounds used in the dump file, are provided by the symbol. This should also make it possible to share a common node between aliases without the [black magic](https://github.com/gtkwave/gtkwave/blob/0e2789170cebab9f1941111979b48dc201651ace/gtkwave3-gtk3/src/fst.c#L1593-L1595) that is currently used.

tbybell, 2025-04-27:
OK, I'm glad to hear the only problem with Icarus is that I need to update my Icarus.  :-)

For the loaders, I agree with what you write in the final paragraph above.  The original code for GTKWave slurped in the entire VCD file on init, and when later on I added support for IBM MVLSIM AETs, I started looking at the dumpfile as a database rather than something to be read all at once.  For those aliases:

`analyzer.h:      struct fac *mvlfac; /* for use with mvlsim aets */`

MVLSIM would've been earlier than 2004 which is about the time when IBM's AET2 appeared.  That is some seriously old legacy code.  Anything you come up with will be an improvement: the "black magic" code was more-or-less arbitrary bridge code between the legacy VCD loader and later database loaders that should've been temporary, but it turns out that it wasn't.

tbybell, 2025-04-28:
I messed with this some last night.  Beyond the save file aspect which I'm not too concerned about, a part that I see that's potentially confusing would be vector expansion.  For a signal like `x[0:0]`, with respect how other ranged signals work, for consistency, it seems that one should be able to double click expand then pull out `x[0]` (even though it's the same thing) from the GUI as a signal extracted from that `[0:0]` bit range.  Interestingly enough, in the GUI for nWave (Debussy/Verdi), you're stuck with `x[0:0]` and can't directly extract `x[0]`, so the precedent nWave sets is that ranged signal that is one bit wide will always be a ranged signal.  I didn't look to see if that was an artifact of the GUI and if the expression building/parsing still would let you pull out bit zero without the range.  There was only so deep I wanted to go down this rabbit hole.

For FST, duplicating nWave's behavior seems easy enough to do: add another argument indicating an explicit range was encountered and have it flowing out of `extractNextVar()` alongside the lsb and msb.  Use that to override the `(f->len>1)` length check for the if block that calls `sprintf_2_sdd()`.  VCD would be a little bit more involved because it would have to be done during `vcd_build_symbols()` time instead of near immediately, but it's the same thing.

For LTS, I'm going to skip this one unless there's some ranged vs unranged ambiguity somewhere.  

Note that for Verilog for my day job, many components such as dflops and the like are parameterized, so if you had output q on a 2 bit signal, it would be q[1:0], but if the parameter specified the width instead as 1 bit, the range would not be dumped by VCS even though explicitly in the I/O list it's defined as something along the lines of q[WIDTH-1:0].  I'll have to dig out my book on the VPI to see if a scalar effectively is whatever[0:0].

rfuest, 2025-04-30:
I've changed the behaviour of my development branch to handle 1 bit vectors as vectors, which means the user will need to expand it to see the actual scalar waveform. In the future it might be a good idea to add a configuration option to let the user decide if these signals should be kept as vectors or automatically converted to a scalar trace when they are added as a new trace.

There is one more case which trips up LTS a bit because it treats `-1` as a special value. `signal[-1:-1]` isn't displayed as `signal[-1]`, but just as `signal`. But this is probably not something that is often encountered in real VCD files (or at all).