gtkwave/gtkwave #432: Autocoalescing of scalar vars with duplicate names in VCD files

rfuest, 2025-05-01:
GTKWave LTS autocoalesces scalar vars with duplicate names. Like `var1` and `var2` in this example, which get merged into two 2 bit vectors:
```
$var wire 1 a var1 $end
$var wire 1 b var1 $end
$var wire 1 b var2 $end
$var wire 1 a var2 $end
$enddefinitions $end
#0
0a
1b
#1
1a
0b
#2
```
Is this the intended behaviour to support some tool that outputs bit blasted vectors without index or is this a bug? In my opinion this is error prone, because the bit order in the vector is dependent on the order the variables are declared in the VCD file (`var1` and `var2` are reversed). If this is a bug, I think that GTKWave should instead warn the user and ignore the duplicate declarations.

I'm also considering to remove the `$DUP` prefix generation for all duplicate names in the VCD loader and instead also warn the user. Or is there a good reason too keep this feature?

tbybell, 2025-05-01:
Autocoalesce of bitblasted vectors broke in 3.3.122 from the fix for #428, so it looks like I'll be doing a 3.3.123 this weekend.

For the `if(num_seen)` add for that bugfix, it needs instead to be `if(num_seen && (v->size > 1))`.

6cd3033..d32b741  lts -> lts

For what you're doing with `var1`/`var2`, the VCD loader is thoroughly confused and you found a bug in those loaders.  Those shouldn't be autocoalesced.  

d32b741..92a0efa  lts -> lts

Note that there is no sanity checking on the order of bits in autocoalesce to ensure they're in consecutive sequence: it's assumed that whatever wrote the VCD file knew what it was doing, so the `v->lsi` of this symbol should be +1s or -1s of the previous symbol's `v->lsi` in a continuing run.  The != introduced in the change directly above is quick and dirty and isn't truly robust, but it's probably good enough.  I don't want to introduce more bugs in LTS.  :-)

With respect to the `$DUP` prefix generation, there's no real need to keep those `$DUP` symbols.  If you convert to FST, you'll notice those prefixed symbols don't exist, or more correctly, any duplicates are filtered from the signal display widgets.  I encountered rampant problems with duplicate symbols in FSDB files due to misuse of $fsdbDumpVars, so the extload/FSDB loader hid duplicates with the FST loader doing the same.

The original motivation for the `$DUP` prefix was to help debug broken VCD writers as there were a couple of cases where I was having to deal with people who were working on their own VCD writers, and that was the easiest way to debug where they emitted the same hierarchy more than once.  I would say it's obsolete and can be removed.  A warning message is definitely preferable.

BTW, the following demonstrates a run of signals that shows how autocoalesce works from a real simulator.  Subscripts are always expected and dumping is in order from MSB to LSB.  This one is old.  I haven't tried against Questa to see if modern simulators still do this.
```
$date
        Mon Jul 26 12:19:13 1999
$end
$version
        ModelSim Version 5.2a
$end
```
...
```
$var wire 1 t data_scrmble $end
$var wire 1 u s [11] $end
$var wire 1 v s [10] $end
$var wire 1 w s [9] $end
$var wire 1 x s [8] $end
$var wire 1 y s [7] $end
$var wire 1 z s [6] $end
$var wire 1 { s [5] $end
$var wire 1 | s [4] $end
$var wire 1 } s [3] $end
$var wire 1 ~ s [2] $end
$var wire 1 !! s [1] $end
$var wire 1 "! s [0] $end
$var reg 11 #! ds [10:0] $end
```

tbybell, 2025-05-01:
Today, I broke out a version of Questa (10.7d_2-QCF) that is a couple of years old.  I didn't see any evidence of bit blasting in some nontrivial models from my day job that I put through it, so it might no longer be an issue with modern Verilog simulators.  That would be a bonus, however I believe you'd still need autocoalesce functionality in some capacity for GHW based on how info is dumped into the file, which I suppose partially boils down to VHDL's type system.  The nvc simulator can dump the kind of FST with vectors that we're used to.  Outside of GHDL, autocoalesce might not have too much real use outside of homebrew simulators.

Moving the coalescing elsewhere makes sense.  The way it's currently done was out of cut and paste laziness where I never expected GTKWave to have half a dozen loaders and be in use this long.  I would have refactored the code long ago if I knew I'd be supporting it this long.

tbybell, 2025-05-02:
I looked at the `$DUP` code in the VCD loaders some more.  

```
if(symfind(str, NULL))
{
// nothing, was formerly $DUP generation
}
else /* else added, no longer fallthrough */
{
s=symadd(str,hashdirty?hash(str):GLOBALS->hashcache);
...
```
It looks like logic conceptually similar in flow to the above might work where the $DUP code gets removed as it then suppresses the duplicate symbol from being built later on in vcd_build_symbols().  It seems like it's as simple as that.

` * fix for duplicate nets	19dec00ajb`

I found the changelog entry in my old verilog compiler where GTKWave was initially being developed and bundled with it...
```
22/12/00        Fixed bug in vcd.c that would cause the viewer to crash
                        under certain conditions if the hierarchy would
                        have multiple dumps of the same exact signal.
                        The names are now resolved by prepending $DUPx
                        to the hierarchy (where x=0, 1, 2, etc).
```

There you have it.  If the viewer doesn't crash now, we don't need it.

rfuest, 2025-05-02:
> Today, I broke out a version of Questa (10.7d_2-QCF) that is a couple of years old. I didn't see any evidence of bit blasting in some nontrivial models from my day job that I put through it, so it [...]

I want to keep the functionality, but it's good to know that this isn't too common anymore. The GHW loader will need some more attention and I think it should be possible to at least merge some vectors directly in the loader. Otherwise it will be hard to ever successfully load GHW files like the one from https://github.com/gtkwave/gtkwave/issues/280, which contains 34 4096-bit and 24 10000-bit wide bitblasted vectors, without using massive amounts of RAM.