gtkwave/gtkwave #433: How to add signals by comand line?

nipscernlab, 2025-05-03:
Hello,

Is there anyway to add signals from command line withou using the SST from the GUI?

regards!

tbybell, 2025-05-03:
There is no direct command line option for that.  If you want to add signals without manipulating the GUI, you'd have to do it either from save files or by issuing Tcl commands to the wish terminal.

-Tony

nipscernlab, 2025-05-03:
Thank you for the answer!
How to invoke tcl commands in the git top branch?

rfuest, 2025-05-04:
As previously discussed in #412 the development branch, that will become GTKWave 4, doesn't support TCL scripting. The first stable release of GTKWave 4 will contain support for Python scripting as a replacement, but it's not yet possible to add traces via Python yet.

If you provide more details about what you are trying to achieve we might find a solution that works for your use case.

nipscernlab, 2025-05-04:
[GTKWave.zip](https://github.com/user-attachments/files/20028393/GTKWave.zip)

I am developping a C to assembler compiler.
So, the Assembler creates a vcd file with signals and variables found in the C code.
I need a script to read the vcd file, looking for some signals from pattners.
Please, run the following command (I use cmd from windows):

gtkwave --rcvar "hide_sst on" --dark proc_fft_tb.vcd --script=gtk_proc_init.tcl

the gtkwave version: GTKWave Analyzer v3.4.0 (w)1999-2022 BSI.
Here is a picture what you suppose to see:

[image]

tbybell, 2025-05-04:
If you're using a Tcl file to import signals, make alias names, etc., it stands to reason you could just as well have it emit the equivalent data "Write Save File" would do, then pick that up at the command line instead of executing the Tcl script directly from gtkwave.  That is, you're duplicating the viewer state after it would have executed the script.

I believe the save file format for the dev branch will be changing if it hasn't already, so this will be a bit of a moving target, but it seems like it would work as an interim solution in a viewer environment that does not provide scripting capability (yet).

-Tony

rfuest, 2025-05-05:
Thanks for the script, I'll keep that use case in mind for GTKWave 4's future scripting support. Tony's idea of generating a GTKW save file from your tool is probably the best solution at the moment. Another hacky workaround would be to use a secondary GTKWave 3 installation just to run your script and save it as a GTKW file that could then be opened in a newer GTKWave version.

I don't think there is any documentation about the GTKW file format, but you should be able to get the basics from looking at a save file. The least intuitive aspect of the format are the flags, which are the lines starting with `@`. These encode options like the display format or whether a filter is enabled. The values are declared here: https://github.com/gtkwave/gtkwave/blob/3a67eabcc296265d59acb32cdab90531acc5e07a/gtkwave3-gtk3/src/analyzer.h#L527-L599 But it's easier to just copy the values from an existing file. One thing to keep in mind is that the flags are stateful and are applied to all following traces until a new `@` line overrides them. For an automatically generated file I would recommend to just add a `@` line in front of each trace line.

Just out of curiosity, where did you install GTKWave from? It looks like a pretty old build of the development branch.

> I believe the save file format for the dev branch will be changing if it hasn't already, so this will be a bit of a moving target, but it seems like it would work as an interim solution in a viewer environment that does not provide scripting capability (yet).

The save file format hasn't changed yet. While GTKWave 4 will use a different save file format natively it will still be able to import GTKW files from older versions.

nipscernlab, 2025-05-05:
Thanks for your comments.
I like very much the idea of write the gtkw file directly from m'y tool!
I took a look and it seems to be readable.
The only thing not trivial is the @ values.
I will try to read the document about it and come Back if needed.

About your question. I am using a gtkwave version coming with the icarus verilog. But my idea is try to use the top branch of the github

Cheers
Luciano

nipscernlab, 2025-05-12:
Hello,

Following the idea of writing the gtkw file directly, I found this line here:

*-21.742342 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1

What does it means?

Luciano