gtkwave/gtkwave #435: How to set a clock signal?

SparkyNZ, 2025-05-08:
I am really struggling here - how can I tell GTKWave to use my cpuClk signal as the clock? Isn't there a way to change the time ruler to clock cycles instead?

[image]

tbybell, 2025-05-08:
Select only cpuClk.
Select menu option Search->Pattern Search 1
Move transition type box for cpuClk to "Rising Edge"
Click "Mark"
Click "Exit"

It will save to your save file if you make one.

closed 2025-05-08 by tbybell

tbybell, 2025-05-08:
No, you'll have to dump a cycle count signal of your own in sim in something like an integer signal.  I've seen such a thing as a part of some testbenches because, for example, for long sim runs if there is a $dumpvars off/on to save on disk space, visualization tools can't get the right cycle count from the waves alone.

SparkyNZ, 2025-05-08:
Yeah that make sense and is easy enough to do. Thanks for the quick
response.