gtkwave/gtkwave #436: Building on Windows

SparkyNZ, 2025-05-18:
Hi. I've cloned the latest Github repo (https://github.com/gtkwave/gtkwave.git). I have Msys64 (Mingw64) installed. How do I build on Windows?

I don't have a 'configure' script/file in the gtkwave folder or the src folder. I don't have a makefile either.

rfuest, 2025-05-18:
The development version has moved from using GNU Autotools to Meson as a build system. The Windows build instructions haven't been updated yet, but most of the Linux docs also apply to building it in Msys64: https://gtkwave.github.io/gtkwave/install/unix_linux_4.html

Most of the GTKWave 3.3 dependencies in this list are still relevant https://gtkwave.github.io/gtkwave/install/win.html#id1, but you will also need to install these:
- `mingw-w64-x86_64-gobject-introspection`
- `mingw-w64-x86_64-gtk4`
- `mingw-w64-x86_64-desktop-file-utils
- `flex`
- `meson`

I haven't tried building the development branch on Windows for a while and it isn't part of the CI build yet, so there might be additional issues or dependencies I forgot to list. Let me know if you encounter any issues and I'll take a closer look.

sadguitarius, 2025-10-21:
@marceloosterhuis Thanks for this! I'm able to get the build to work for the most part, but it's stuck on the gobject-introspection generation. I'm not too familiar with Meson builds, so I'm having trouble figuring out where the problem actually is. Is this something you've run into? Here's the error I'm getting.

```
FAILED: [code=1] lib/libgtkwave/src/Gw-1.gir
"C:/msys64/mingw64/bin/meson" "--internal" "exe" "--unpickle" "C:/msys64/home/sadguitarius/src/source/gtkwave/build/meson-private/meson_exe_g-ir-scanner.exe_f0af95d9a96210629d88d1cd48a98db7f5f774a8.dat"
while executing ['C:/msys64/mingw64/bin/g-ir-scanner.exe', '--quiet', '--no-libtool', '--namespace=Gw', '--nsversion=1', '--warn-all', '--output', 'lib/libgtkwave/src/Gw-1.gir', '--c-include=gtkwave.h', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/lib/libgtkwave/src', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libgtkwave/src', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/lib/libghw/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libghw/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/subprojects/libfst/src/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/subprojects/libfst/src/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/lib/libjrb/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libjrb/.', '--filelist=C:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libgtkwave/src/libgtkwave.dll.p/Gw_1_gir_filelist', '--include=GObject-2.0', '--symbol-prefix=gw', '--identifier-prefix=Gw', '--pkg-export=gtkwave-1', '--cflags-begin', '-D_GNU_SOURCE', '-DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_72', '-DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_72', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/lib/libghw/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libghw/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/subprojects/libfst/src/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/subprojects/libfst/src/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/lib/libjrb/.', '-IC:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libjrb/.', '-IC:/msys64/mingw64/include/glib-2.0', '-IC:/msys64/mingw64/lib/glib-2.0/include', '-IC:/msys64/mingw64/include', '-IC:/msys64/mingw64/include/gobject-introspection-1.0', '--cflags-end', '--add-include-path=C:/msys64/mingw64/share/gir-1.0', '-LC:/msys64/home/sadguitarius/src/source/gtkwave/build/subprojects/libfst/src', '--extra-library=fst', '-LC:/msys64/home/sadguitarius/src/source/gtkwave/build/lib/libgtkwave/src', '--library', 'gtkwave', '-LC:/msys64/home/sadguitarius/src/source/gtkwave/build/subprojects/libfst/src', '-LC:/msys64/mingw64/lib', '--extra-library=glib-2.0', '--extra-library=intl', '--extra-library=gobject-2.0', '--extra-library=z', '--extra-library=bz2', '--extra-library=girepository-1.0', '--sources-top-dirs', 'C:/msys64/home/sadguitarius/src/source/gtkwave/', '--sources-top-dirs', 'C:/msys64/home/sadguitarius/src/source/gtkwave/build/']
--- stdout ---

--- stderr ---
Traceback (most recent call last):
  File "C:\msys64\mingw64\bin\g-ir-scanner-script.py", line 135, in <module>
    sys.exit(scanner_main(args))
             ^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/gobject-introspection/giscanner/scannermain.py", line 673, in scanner_main
    shlibs = create_binary(transformer, options, args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/gobject-introspection/giscanner/scannermain.py", line 483, in create_binary
    binary = compile_introspection_binary(options,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/gobject-introspection/giscanner/dumper.py", line 304, in compile_introspection_binary
    return dc.run()
           ^^^^^^^^
  File "C:/msys64/mingw64/lib/gobject-introspection/giscanner/dumper.py", line 171, in run
    self._link(bin_path, introspection_obj)
  File "C:/msys64/mingw64/lib/gobject-introspection/giscanner/dumper.py", line 292, in _link
    subprocess.check_call(args)
  File "C:/msys64/mingw64/lib/python3.12/subprocess.py", line 408, in check_call
    retcode = call(*popenargs, **kwargs)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.12/subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:/msys64/mingw64/lib/python3.12/subprocess.py", line 1462, in _execute_child
    args = list2cmdline(args)
           ^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.12/subprocess.py", line 608, in list2cmdline
    for arg in map(os.fsdecode, seq):
               ^^^^^^^^^^^^^^^^^^^^^
  File "<frozen os>", line 859, in fsdecode
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

rfuest, 2025-10-21:
I haven't looked into this, but you can try to build without introspection enabled by adding a parameter to the `meson setup` call: `meson setup build --prefix=/opt -Dintrospection=false`

sadguitarius, 2025-10-22:
Thank you but unfortunately that didn't seem to help either.